espressif/esp-idf #15075: Help with command line debugging on esp32-c3-wroom

gamename, 2024-12-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The command `idf.py gdbtui` should start debugging for my app.

### What is the actual behavior?

I get this output:
```GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=aarch64-apple-darwin21.1 --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
add symbol table from file "/Users/tennis/src/C2DS-egg/build/bootloader/bootloader.elf"
/Users/tennis/src/C2DS-egg/build/gdbinit/gdbinit:4: Error in sourced command file:
/Users/tennis/src/C2DS-egg/build/gdbinit/connect:4: Error in sourced command file:
:3333: Operation timed out.
(gdb)```

### Steps to reproduce.

1. build the app using `idf.py build flash`
2. run `idf.py gdbtui`

### Build or installation Logs.

```plain
Everything works, but here it is for completeness:

idf build flash monitor
Executing action: all (aliases: build)
Running ninja in directory /Users/tennis/src/C2DS-egg/build
Executing "ninja all"...
[0/1] Re-running CMake...CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
[... 472 lines cut ...]
I (5272) core_mqtt_agent_manager: Connecting to the broker.
I (5532) coreMQTT: MQTT connection established with the broker.
I (5532) core_mqtt_agent_manager: Session present: 1

I (5532) core_mqtt_agent_manager: MQTT connection established.
I (5542) core_mqtt_agent_manager: Received MQTT Agent event: 0
I (5552) core_mqtt_agent_manager: coreMQTT-Agent connected.
I (5552) mqtt_subscriptions: coreMQTT-Agent connected.
I (5562) core_mqtt_agent_manager: Posted MQTT Agent event: 0
I (5562) mqtt_subscriptions: Task "LedColorTask" sending subscribe request to coreMQTT-Agent for topic filter: farm/coop/led/color with id 1
I (5762) SubManager: Adding subscription for topic farm/coop/led/color
I (5762) mqtt_subscriptions: Subscribe 1 for topic filter farm/coop/led/color succeeded for task "LedColorTask".
I (5832) coreMQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (5832) coreMQTT: State record updated. New state=MQTTPubAckSend.
I (5842) mqtt_subscriptions: Task "LedColorTask" received: {"LED": "LED_SOLID_GREEN"}
I (5852) mqtt_subscriptions: Received message: {"LED": "LED_SOLID_GREEN"}

Done
```

### More Information.

I am completely confused about how to get GDB working for this environment.  Supposedly, the esp32-c3-wroom chip does not require openocd or an external jtag. True?

Either way, I cannot get GDB to work in the simpliest way I can configure.

igrr, 2024-12-22:
> Supposedly, the esp32-c3-wroom chip does not require openocd or an external jtag. True?

That's true. For this to work, you need to connect to the chip using it's built-in USB port, not via a USB-UART converter such as CP2102. This depends on your development board.

Additionally, you are running `idf.py gdbtui` which expects that OpenOCD is already running somewhere. Usually if you also want to launch OpenOCD you'd use `idf.py openocd gdbtui`.

I would recommend to run `idf.py openocd` and see if it starts up without errors. If you get errors that the device isn't found, likely you aren't connected to ESP32-C3 over the built-in USB port.

Some links for reference:

- https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/using-debugger.html#jtag-debugging-with-idf-py

gamename, 2024-12-22:
@igrr Thanks for the quick reply.

> I would recommend to run `idf.py openocd` first and see if it starts up without errors. If you get errors that the device isn't found, likely you aren't connected to ESP32-C3 over the built-in USB port. 

So, `openocd` _is_ required for esp32-c3-wroom?  

Anyway, here is the output:
```log
idf openocd
Executing action: openocd
Note: OpenOCD cfg not found (via env variable OPENOCD_COMMANDS nor as a --openocd-commands argument)
OpenOCD arguments default to: "-f board/esp32c3-builtin.cfg"
OpenOCD started as a background task 22208
Executing action: post_debug
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!
/Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error:
at file "/Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
Error: [esp32c3] Unsupported DTM version: -1
Error: [esp32c3] Could not identify target type.
"openocd" exited with 1
HINT: Please check the wire connection to debugging device or access rights to a serial port.
```

This is very confusing.  There is only *one* micro-usb connector on the esp32.  There is no alternative.  It is the only USB connection.  If that is the case, how can a separate USB be possible?

gamename, 2024-12-22:
> * https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html

This another point of confusion.  The article talks about connecting pins and USB signals.  Here is a picture of my esp32-c3-wroom:
[image]

There is micro-USB connection and no facility for breaking out the individual leads.  The article doesn't mention anything about it. What is it trying to convey?

igrr, 2024-12-22:
On this specific development board, microUSB port is connected to a CP2102 USB-UART bridge.

Internal USB port of ESP32-C3 is not connected to a physical USB connector. However you can connect a USB breakout cable such as this one https://www.adafruit.com/product/4448 to the pins on the side of the board.

Connection can be done according to https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html#configure-hardware

I wrote about a couple of alternative options earlier in https://github.com/espressif/openocd-esp32/issues/343#issuecomment-2424735344

gamename, 2024-12-22:
@igrr 
> I wrote about a couple of alternative options earlier in [espressif/openocd-esp32#343 (comment)](https://github.com/espressif/openocd-esp32/issues/343#issuecomment-2424735344)

Hmm.  I have an esp32-s3-wroom-01. Does this mean that I can move my app over to the s3, plug into the "USB" port, and debug should "just work"?

igrr, 2024-12-22:
> I have an esp32-s3-wroom-01

ESP32-S3-WROOM-01 is the name of a module. I am not sure which development board with this module you have. If you have an [ESP32-S3-DevKitC-1](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/index.html) then it should work as you have described.

gamename, 2024-12-22:
@igrr 

Excellent!

It is an ESP32-S3-DevKitC-1.  for _monitoring_, do I use the USB or UART port?

gamename, 2024-12-22:
@igrr 

Re the esp32-c3-DevKitC-02... Do I need to use [this](https://www.adafruit.com/product/4448) for debugging and a connection via the micro-USB port for flash and monitoring?

I tried monitoring via the special adapter (above) and it didn't work.

igrr, 2024-12-22:
Could you please describe what exactly didn't work? (No /dev/cu... port found, or port found but no output, or some other behavior?)
That specific adapter is not required, a USB cable cut in half with a pin header soldered should also work.

gamename, 2024-12-22:
@igrr 

> Could you please describe what exactly didn't work? (No /dev/cu... port found, or port found but no output, or some other behavior?) That specific adapter is not required, a USB cable cut in half with a pin header soldered should also work.

Ah, good point. I didn't say how it didn't work.  Apologies. It simply hung when I tried `idf.py monitor`.  No other diagnostics AFAIK.

As for the adapter, I made one that simply breaks out all the connections and I matched the pin connections per the description [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html#configure-hardware)

gamename, 2024-12-23:
@igrr 
> Could you please describe what exactly didn't work? (No /dev/cu... port found, or port found but no output, or some other behavior?) That specific adapter is not required, a USB cable cut in half with a pin header soldered should also work.

Could I simply use the adapter cable instead of the micro-USB connector for normal loading/monitoring as well as debugging?  I have a esp32-c3-DevKitC-02.
Thanks

igrr, 2024-12-25:
Yes, that should work. You might need to check that the console output is configured to go to USB_SERIAL_JTAG.

I'd recommend flashing a hello_world example first to verify the setup and then moving on to your real application.

gamename, 2024-12-25:
@igrr 
> Yes, that should work. You might need to check that the console output is configured to go to USB_SERIAL_JTAG.
> 
> I'd recommend flashing a hello_world example first to verify the setup and then moving on to your real application.

Thank you, sir.  I'll close this out.

closed 2024-12-25 by gamename