espressif/esp-idf #15076: Internal DAC and ADC support for Classic Bluetooth Handsfree Example

theavijitsarkar, 2024-12-22:
### Is your feature request related to a problem?

I am trying to turn an esp32 into a bluetooth handsfree device and use a 3.5mm trrs headphone with mic. I dont want to use an external dac or adc. I would just connect the pin 25 26 to the tip and ring, and the ring2 to the adc (35).
I have been able to play audio via internal dac after configuring the a2dp sink example, but there is no way to configure the hfp_hf example. Kindly add this config or help us achieve this.

### Describe the solution you'd like.

A simple config to choose internal dac, and the pins for tip ring ring sleeve.

esp-qing, 2024-12-25:
Hi,  there are two types of audio datapath in the hfp_hf example: PCM and HCI. In PCM datapath, the internal DAC cannot be configured due to some limitations. But in vHCI datapath, the internal DAC can be configured (currently no configuration). So which datapath are you using?

theavijitsarkar, 2024-12-26:
@esp-qing thanks for your response, as per your suggestion I have configured it to use HCI data path, and made all DAC related configs. Still no sound from the headphone (I do get audio playback via headphone and internal dac with the a2dp example). Any guidance would be really helpful.

Here is the log output 

```
I (1330) main_task: Returned from app_main()
hfp_hf> I (14030) BT_HF: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
W (17990) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
I (18000) BT_HF: event: 16
W (18030) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
I (18130) BT_HF: event: 21
W (18270) BT_APPL: new conn_srvc id:27, app_id:1
I (18270) BT_HF: APP HFP event: CONNECTION_STATE_EVT
I (18270) BT_HF: --connection state connected, peer feats 0x0, chld_feats 0x0
I (18390) BT_HF: APP HFP event: CALL_IND_EVT
I (18390) BT_HF: --Call indicator call in progress
I (18390) BT_HF: APP HFP event: CALL_SETUP_IND_EVT
I (18410) BT_HF: --Call setup indicator NONE
I (18410) BT_HF: APP HFP event: NETWORK_STATE_EVT
I (18420) BT_HF: --NETWORK STATE available
I (18420) BT_HF: APP HFP event: SIGNAL_STRENGTH_IND_EVT
I (18430) BT_HF: -- signal strength: 3
I (18430) BT_HF: APP HFP event: ROAMING_STATUS_IND_EVT
I (18430) BT_HF: --ROAMING: inactive
I (18440) BT_HF: APP HFP event: BATTERY_LEVEL_IND_EVT
I (18450) BT_HF: --battery level 4
I (18450) BT_HF: APP HFP event: CALL_HELD_IND_EVT
I (18450) BT_HF: --Call held indicator NONE held
I (18470) BT_HF: APP HFP event: CONNECTION_STATE_EVT
I (18470) BT_HF: --connection state slc_connected, peer feats 0xfef, chld_feats 0x3f
I (18480) BT_HF: APP HFP event: INBAND_RING_TONE_EVT
I (18480) BT_HF: --inband ring state Provided
I (18690) BT_HF: APP HFP event: AUDIO_STATE_EVT
I (18690) BT_HF: --audio state connected_msbc
W (25890) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(50 150)
W (25890) BT_HCI: hcif mode change: hdl 0x80, mode 0, intv 0, status 0x20
E (25890) BT_APPL: bta_dm_pm_btm_status hci_status=32
I (25910) BT_HF: ESP_BT_GAP_MODE_CHG_EVT mode:0
W (33310) BT_HCI: hcif disc complete: hdl 0x180, rsn 0x13
I (33320) BT_HF: APP HFP event: CALL_SETUP_IND_EVT
I (33320) BT_HF: --Call setup indicator OUTGOING_DIALING
I (33320) BT_HF: APP HFP event: AUDIO_STATE_EVT
I (33330) BT_HF: --audio state disconnected
I (33430) BT_HF: APP HFP event: CALL_IND_EVT
I (33430) BT_HF: --Call indicator NO call in progress
I (33440) BT_HF: APP HFP event: CALL_SETUP_IND_EVT
I (33450) BT_HF: --Call setup indicator NONE
I (33590) BT_HF: APP HFP event: AUDIO_STATE_EVT
I (33590) BT_HF: --audio state connected_msbc
I (36240) BT_HF: APP HFP event: CALL_SETUP_IND_EVT
I (36240) BT_HF: --Call setup indicator OUTGOING_ALERTING
I (39050) BT_HF: APP HFP event: CALL_IND_EVT
I (39050) BT_HF: --Call indicator call in progress
I (39050) BT_HF: APP HFP event: CALL_SETUP_IND_EVT
I (39050) BT_HF: --Call setup indicator NONE
W (46250) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(400 800)
W (46250) BT_HCI: hcif mode change: hdl 0x80, mode 0, intv 0, status 0x20
E (46250) BT_APPL: bta_dm_pm_btm_status hci_status=32
I (46270) BT_HF: ESP_BT_GAP_MODE_CHG_EVT mode:0
W (46410) BT_HCI: hcif disc complete: hdl 0x180, rsn 0x13
I (46410) BT_HF: APP HFP event: AUDIO_STATE_EVT
I (46410) BT_HF: --audio state disconnected
I (46530) BT_HF: APP HFP event: CALL_IND_EVT
I (46530) BT_HF: --Call indicator NO call in progress
```
[image]
[image]

closed 2025-05-13 by Alvin1Zhang