espressif/esp-idf #15077: The program crashes for an unknown reason. Suspect: esp_timer_impl_lac.c

xixxi, 2024-12-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3.1-dirty

### Espressif SoC revision.

ESP32 chip revision: v1.0

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

Particle system run as usual...

### What is the actual behavior?

after half second crash

### Steps to reproduce.

run program. 
click 3 times. 
particle system run. 
some particles emitted. 
Crash.

### Debug Logs.

```plain
load:0x40080404,len:4060                                                                                                                                                                                                                   entry 0x400806b8                                                                                                                                                                                                                           I (30) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader                                                                                                                                                                                     I (30) boot: compile time Dec 22 2024 08:28:07                                                                                                                                                                                             I (30) boot: Multicore bootloader                                                                                                                                                                                                          I (35) boot: chip revision: v1.0                                                                                                                                                                                                           I (39) boot.esp32: SPI Speed      : 40MHz                                                                                                                                                                                                  I (44) boot.esp32: SPI Mode       : DIO                                                                                                                                                                                                    I (48) boot.esp32: SPI Flash Size : 4MB                                                                                                                                                                                                    I (53) boot: Enabling RNG early entropy source...                                                                                                                                                                                          I (58) boot: Partition Table:                                                                                                                                                                                                              I (62) boot: ## Label            Usage          Type ST Offset   Length                                                                                                                                                                    I (69) boot:  0 nvs              WiFi data        01 02 0000a000 00006000                                                                                                                                                                  I (76) boot:  1 phy_init         RF data          01 01 00010000 00001000                                                                                                                                                                  I (84) boot:  2 factory          factory app      00 00 00020000 00200000                                                                                                                                                                  I (91) boot:  3 storage          Unknown data     01 82 00220000 00100000                                                                                                                                                                  I (99) boot: End of partition table                                                                                                                                                                                                        I (103) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=3715ch (225628) map                                                                                                                                                       I (168) esp_image: segment 1: paddr=00057184 vaddr=3ffb0000 size=03f90h ( 16272) load                                                                                                                                                      I (173) esp_image: segment 2: paddr=0005b11c vaddr=40080000 size=04efch ( 20220) load                                                                                                                                                      I (180) esp_image: segment 3: paddr=00060020 vaddr=400d0020 size=c676ch (812908) map                                                                                                                                                       I (385) esp_image: segment 4: paddr=00126794 vaddr=40084efc size=11c20h ( 72736) load                                                                                                                                                      I (418) boot: Loaded app from partition at offset 0x20000                                                                                                                                                                                  I (418) boot: Disabling RNG early entropy source...                                                                                                                                                                                        I (430) cpu_start: Multicore app                                                                                                                                                                                                           I (438) cpu_start: Pro cpu start user code                                                                                                                                                                                                 I (438) cpu_start: cpu freq: 240000000 Hz                                                                                                                                                                                                  I (438) app_init: Application information:                                                                                                                                                                                                 I (441) app_init: Project name:     PromoDisplayPrj                                                                                                                                                                                        I (447) app_init: App version:      1                                                                                                                                                                                                      I (451) app_init: Compile time:     Dec 22 2024 08:27:58                                                                                                                                                                                   I (457) app_init: ELF file SHA256:  a55b8d066251da6c...                                                                                                                                                                  [... cut ...] leSystem::run(long) at C:/MyDevEnv/Projs/PromoDisplay/CodeESP32/components/particles/src/ParticleSystem.hpp:144                                                                                                           (inlined by) displayTask(void*) at C:/MyDevEnv/Projs/PromoDisplay/CodeESP32/main/display/displayTask.cpp:2925
```

### More Information.

There is plenty of memory, and the same code runs perfectly on a PC. During execution, no specific calls are made except for rendering on HUB75.

ESP-Marius, 2024-12-23:
Crash seems to be coming from ` 0x400da488: ParticleEmitter::emit(long) at C:/MyDevEnv/Projs/PromoDisplay/CodeESP32/components/particles/src/ParticleEmitter.hpp:853 ` accessing an invalid address `0xfffda894`. 

Does not seem related to `esp_timer_impl_lac` from what I can see

xixxi, 2024-12-23:
> Crash seems to be coming from `0x400da488: ParticleEmitter::emit(long) at C:/MyDevEnv/Projs/PromoDisplay/CodeESP32/components/particles/src/ParticleEmitter.hpp:853` accessing an invalid address `0xfffda894`.
> 
> Does not seem related to `esp_timer_impl_lac` from what I can see

I think I found the problem. It's stack and heap overlap.

ESP-Marius, 2024-12-25:
Marking this as closed then as it seems to be not an issue with esp-idf.

closed 2024-12-25 by ESP-Marius