espressif/esp-idf #15079: ESP32-S3: Wi-Fi and USB (CDC ACM Host) Incompatibility – USB not work after

antunesls, 2024-12-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Connect to wifi as station (client) and comunicate as a USB host , using serial connection.

### What is the actual behavior?

When start the wifi connection the USB stops to work, not even detect new devices.

### Steps to reproduce.

Build the project attached

[cdc_acm_project.zip](https://github.com/user-attachments/files/18223192/cdc_acm_project.zip)

### Build or installation Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037759b
0x4037759b: _xt_handle_exc at C:/Users/lukss/esp/master/esp-idf/components/xtensa/xtensa_vectors.S:772

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x1790
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd10
load:0x403cb700,len:0x2d48
entry 0x403c8928
I (31) boot: ESP-IDF v5.4-dev-2194-gd7ca8b94c8 2nd stage bootloader
I (31) boot: compile time Dec 22 2024 17:37:05
I (32) boot: Multicore bootloader
I (36) boot: chip revision: v0.2
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 2MB
W (54) boot.esp32s3: PRO CPU has been reset by WDT.
W (60) boot.esp32s3: APP CPU has been reset by WDT.
I (65) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (81) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (89) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (96) boot:  2 factory          factory app      00 00 00010000 00100000
I (104) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=24280h (148096) map
I (143) esp_image: segment 1: paddr=000342a8 vaddr=3fc99800 size=049a8h ( 18856) load
I (147) esp_image: segment 2: paddr=00038c58 vaddr=40374000 size=073c0h ( 29632) load
I (155) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=8ce44h (577092) map
I (261) esp_image: segment 4: paddr=000cce6c vaddr=4037b3c0 size=0e3fch ( 58364) load
I (283) boot: Loaded app from partition at offset 0x10000
I (283) boot: Disabling RNG early entropy source...
I (295) cpu_start: Multicore app
I (304) cpu_start: Pro cpu start user code
I (304) cpu_start: cpu freq: 160000000 Hz
I (304) app_init: Application information:
I (307) app_init: Project name:     UPS_ESP32_tinySrv
I (313) app_init: App version:      cf30de4-dirty
I (318) app_init: Compile time:     Dec 22 2024 17:35:47
I (324) app_init: ELF file SHA256:  46a44532f...
I (330) app_init: ESP-IDF:          v5.4-dev-2194-gd7ca8b94c8
I (336) efuse_init: Min chip rev:     v0.0
I (341) efuse_init: Max chip rev:     v0.99
I (346) efuse_init: Chip rev:         v0.2
I (351) heap_init: Initializing. RAM available for dynamic allocation:
I (358) heap_init: At 3FCA2008 len 00047708 (285 KiB): RAM
I (364) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (370) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (376) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (383) spi_flash: detected chip: winbond
I (387) spi_flash: flash io: dio
W (391) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (404) sleep: Configure to isolate all GPIO pins in sleep state
I (411) sleep: Enable automatic switching of GPIO sleep configuration
I (418) main_task: Started on CPU0
I (438) main_task: Calling app_main()
I (458) UPS-Srv: ESP_WIFI_MODE_STA
I (458) pp: pp rom version: e7ae62f
I (458) net80211: net80211 rom version: e7ae62f
I (468) wifi:wifi driver task: 3fcac3ec, prio:23, stack:6656, core=0
I (478) wifi:wifi firmware version: cbdeb9883
I (478) wifi:wifi certification version: v7.0
I (478) wifi:config NVS flash: enabled
I (478) wifi:config nano formatting: disabled
I (478) wifi:Init data frame dynamic rx buffer num: 32
I (488) wifi:Init static rx mgmt buffer num: 5
I (488) wifi:Init management short buffer num: 32
I (488) wifi:Init dynamic tx buffer num: 32
I (498) wifi:Init static tx FG buffer num: 2
I (498) wifi:Init static rx buffer size: 1600
I (508) wifi:Init static rx buffer num: 10
I (508) wifi:Init dynamic rx buffer num: 32
I (508) wifi_init: rx ba win: 6
I (518) wifi_init: accept mbox: 6
I (518) wifi_init: tcpip mbox: 32
I (528) wifi_init: udp mbox: 6
I (528) wifi_init: tcp mbox: 6
I (528) wifi_init: tcp tx win: 5760
I (538) wifi_init: tcp rx win: 5760
I (538) wifi_init: tcp mss: 1440
I (548) wifi_init: WiFi IRAM OP enabled
I (548) wifi_init: WiFi RX IRAM OP enabled
I (558) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (598) wifi:mode : sta (80:65:99:df:57:30)
I (598) wifi:enable tsf
I (598) UPS-Srv: wifi_init_sta finished.
I (608) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1, snd_ch_cfg:0x0
I (608) wifi:state: init -> auth (0xb0)
I (638) wifi:state: auth -> assoc (0x0)
I (698) wifi:state: assoc -> run (0x10)
I (748) wifi:connected with iot-antunesls-sala, aid = 5, channel 1, 40U, bssid = a6:05:d6:a1:4e:20
I (748) wifi:security: WPA2-PSK, phy: bgn, rssi: -45
I (758) wifi:pm start, type: 1

I (758) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (758) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000       
I (778) wifi:<ba-add>idx:0 (ifx:0, a6:05:d6:a1:4e:20), tid:0, ssn:0, winSize:64
I (828) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1768) esp_netif_handlers: sta ip: 10.100.99.114, mask: 255.255.255.0, gw: 10.100.99.1
I (1768) UPS-Srv: got ip:10.100.99.114
I (1768) UPS-Srv: connected to ap SSID:iot-antunesls-sala password:10203099
I (1778) UPS-Srv: Conectando Wifi
I (1778) UPS-Srv: Instalando USB Host
I (1818) UPS-Srv: Instalando driver CDC-ACM
I (1818) UPS-Srv: Procurando dispositivos
I (2818) UPS-Srv: Procurando dispositivos
I (3818) UPS-Srv: Procurando dispositivos
```

tore-espressif, 2025-01-03:
@antunesls  Could you please check that you have this option enabled in menuconfig?

[image]

antunesls, 2025-01-08:
Enabling this option resolved the issue, and the board is now operating as expected. Thank you !

closed 2025-01-08 by antunesls