espressif/esp-idf #15082: octal_psram: PSRAM ID read error: 0x0000000a, PSRAM chip not found or not supported, or wrong PSRAM line mode

lmabbe-lm, 2024-12-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF version : 5.3.2
brand: esp32-s3-wroom-1-N16R8.
我想用 psram 和 flash，menuconfig配置如下
[image]
oct 和 quad 两种模式都试过，都不行。

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b0c
--- 0x40375b0c: esp_restart_noos at /Users/lmabbe/esp/v5.3.2/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x19fc
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe8c
load:0x403cb700,len:0x3164
entry 0x403c8940
I (31) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (31) boot: compile time Dec 23 2024 17:51:28
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot: efuse block revision: v1.3
I (43) qio_mode: Enabling default flash chip QIO
I (49) boot.esp32s3: Boot SPI Speed : 80MHz
I (53) boot.esp32s3: SPI Mode       : QIO
I (58) boot.esp32s3: SPI Flash Size : 16MB
I (63) boot: Enabling RNG early entropy source...
I (68) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (79) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  2 factory          factory app      00 00 00010000 00100000
I (102) boot: End of partition table
I (106) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a920h ( 43296) map
I (121) esp_image: segment 1: paddr=0001a948 vaddr=3fc93600 size=02f20h ( 12064) load
I (125) esp_image: segment 2: paddr=0001d870 vaddr=40374000 size=027a8h ( 10152) load
I (133) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=182f0h ( 99056) map
I (154) esp_image: segment 4: paddr=00038318 vaddr=403767a8 size=0cda4h ( 52644) load
I (171) boot: Loaded app from partition at offset 0x10000
I (171) boot: Disabling RNG early entropy source...
E (183) octal_psram: PSRAM ID read error: 0x0000000a, PSRAM chip not found or not supported, or wrong PSRAM line mode
E cpu_start: Failed to init external RAM!

abort() was called at PC 0x4037551d on core 0
--- 0x4037551d: call_start_cpu0 at /Users/lmabbe/esp/v5.3.2/esp-idf/components/esp_system/port/cpu_start.c:598

Backtrace: 0x40375bd2:0x3fceb220 0x4037ae29:0x3fceb240 0x403813ed:0x3fceb260 0x4037551d:0x3fceb2d0 0x403ccb38:0x3fceb340 0x403ccf59:0x3fceb380 0x403c89a9:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
--- 0x40375bd2: panic_abort at /Users/lmabbe/esp/v5.3.2/esp-idf/components/esp_system/panic.c:463
0x4037ae29: esp_system_abort at /Users/lmabbe/esp/v5.3.2/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x403813ed: abort at /Users/lmabbe/esp/v5.3.2/esp-idf/components/newlib/abort.c:38
0x4037551d: call_start_cpu0 at /Users/lmabbe/esp/v5.3.2/esp-idf/components/esp_system/port/cpu_start.c:598
0x40045c01: ets_run_flash_bootloader in ROM
0x40043ab6: main in ROM
0x40034c45: .stack_ok in ROM

这个是启动日志

nopnop2002, 2024-12-25:
>brand: esp32-s3-wroom-1-N16R8.

It could be ESP32-S3-WROOM-1U-N16R8.

ESP32-S3-WROOM-1U-N16R8 is in Quad Mode.

lmabbe-lm, 2024-12-26:
[image]

**The problem remains, esp32s3 is stuck in infinite restarts。**

**I am using the hello_word template and only modified psram and flash size**

**The startup log is as follows**
-------------------------------------------------------------------

ELF file SHA256: 

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a4c
--- 0x40375a4c: esp_restart_noos at /Users/lmabbe/esp/v5.3.2/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x19fc
load:0x403c8700,len:0x4
load:0x403c8704,len:0xe8c
load:0x403cb700,len:0x3164
entry 0x403c8940
I (31) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (31) boot: compile time Dec 23 2024 17:51:28
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot: efuse block revision: v1.3
I (43) qio_mode: Enabling default flash chip QIO
I (49) boot.esp32s3: Boot SPI Speed : 80MHz
I (53) boot.esp32s3: SPI Mode       : QIO
I (58) boot.esp32s3: SPI Flash Size : 16MB
I (63) boot: Enabling RNG early entropy source...
I (68) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (79) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  2 factory          factory app      00 00 00010000 00100000
I (102) boot: End of partition table
I (106) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a920h ( 43296) map
I (121) esp_image: segment 1: paddr=0001a948 vaddr=3fc92a00 size=02b14h ( 11028) load
I (125) esp_image: segment 2: paddr=0001d464 vaddr=40374000 size=02bb4h ( 11188) load
I (133) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=182f4h ( 99060) map
I (154) esp_image: segment 4: paddr=0003831c vaddr=40376bb4 size=0bd48h ( 48456) load
I (170) boot: Loaded app from partition at offset 0x10000
I (170) boot: Disabling RNG early entropy source...
E (182) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E cpu_start: Failed to init external RAM!

abort() was called at PC 0x4037545d on core 0
--- 0x4037545d: call_start_cpu0 at /Users/lmabbe/esp/v5.3.2/esp-idf/components/esp_system/port/cpu_start.c:598

Backtrace: 0x40375b12:0x3fceb220 0x4037a1f1:0x3fceb240 0x403807b5:0x3fceb260 0x4037545d:0x3fceb2d0 0x403ccb38:0x3fceb340 0x403ccf59:0x3fceb380 0x403c89a9:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710
--- 0x40375b12: panic_abort at /Users/lmabbe/esp/v5.3.2/esp-idf/components/esp_system/panic.c:463
0x4037a1f1: esp_system_abort at /Users/lmabbe/esp/v5.3.2/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x403807b5: abort at /Users/lmabbe/esp/v5.3.2/esp-idf/components/newlib/abort.c:38
0x4037545d: call_start_cpu0 at /Users/lmabbe/esp/v5.3.2/esp-idf/components/esp_system/port/cpu_start.c:598
0x40045c01: ets_run_flash_bootloader in ROM
0x40043ab6: main in ROM
0x40034c45: .stack_ok in ROM

> > 品牌：esp32-s3-wroom-1-N16R8
> 
> 该模型确实是八进制 PSRAM，但请尝试四进制 PSRAM 以确保无误。
> 
> 若开启[未找到时忽略 PSRAM]，则即使未找到 PSRAM，系统也不会重置。