espressif/esp-idf #15083: ESP-IDF5.3.2 + lvgl+ Arduino ESP32 as ESP-IDF Component      cause error :esp_lcd_new_panel_io_i2c

gellary, 2024-12-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

i start with lvgl everything work fine , until i add "ESP-IDF: Add Arduino ESP32 as ESP-IDF Component"
cmakelist.txt is set properly, main.c change to main.cpp extern "C" is put ,but eventually this error came out.

`C:/Users/user/esp/v5.3.2/esp-idf/components/esp_lcd/include/esp_lcd_io_i2c.h: At global scope:
C:/Users/user/esp/v5.3.2/esp-idf/components/esp_lcd/include/esp_lcd_io_i2c.h:102:25: error: conflicting declaration of C function 'esp_err_t esp_lcd_new_panel_io_i2c(i2c_master_bus_handle_t, const esp_lcd_panel_io_i2c_config_t*, esp_lcd_panel_io_t**)'
  102 | static inline esp_err_t esp_lcd_new_panel_io_i2c(i2c_master_bus_handle_t bus, const esp_lcd_panel_io_i2c_config_t *io_config, esp_lcd_panel_io_handle_t *ret_io)`
  
  pls help ~

suda-morris, 2024-12-23:
This is how the i2c lcd header file was used in a .cpp source file, no build error was found. https://github.com/espressif/esp-idf/blob/master/tools/test_apps/system/cxx_build_test/main/test_i2c_lcd.cpp

gellary, 2024-12-23:
@suda-morris can i upload my code ,maybe i do something wrong !

gellary, 2024-12-23:
@suda-morris 
main.cpp
```
extern "C"{
#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"

#include "lvgl.h"
#include "lv_port_disp.h"
#include "lv_port_indev.h"
#include "lv_port_fs.h"
#include "lv_demos.h"
#include "bsp_board.h" //1.方案板参数配置初始化必须包含的头文件
#include "lvgl_init.h" //2.方案板LVGL接口初始化必须包含的头文件
#include "gui_app/ui.h"
}

extern "C"{
static const char *TAG = "main";
}

extern "C" void app_main(void)
{
    // 1.方案板参数配置初始化
    sys_int();
    // 2.方案板LVGL接口初始化
    lvgl_init();

    ui_init();
    // 3.LVGL应用层,用户程序
    // lv_demo_widgets();
    // lv_demo_benchmark();
    // lv_demo_music();
    // lv_demo_stress(); //LVGL应用层,用户程序
    //lv_port_sem_give();
}

```

CMakeLists.txt
```
idf_component_register(
    SRCS
    "main.cpp"
    "gui_app/ui_helpers.c"
    "gui_app/ui.c"  # 如果有多个文件，请逐一列出
    "gui_app/components/ui_comp_button2.c"
    "gui_app/components/ui_comp_hook.c"
    "gui_app/components/ui_comp.c"

    INCLUDE_DIRS
    "include"
    "gui_app"
    "gui_app/screens"
    "gui_app/components"
)
```

original file from https://gitee.com/hemukeji/esp32-s3-5inch-rgb-001-sdk
i use file : example > 1.lvgl_demo 
and add ESP-IDF: Add Arduino ESP32 as ESP-IDF Component
then boom error come out

gellary, 2024-12-23:
This is the whole folder https://drive.google.com/file/d/1pH49klrENvSGmdykOWFfMqFUORg64b0V/view?usp=sharing

location 
shooting_case >examples>2.shooting_case

suda-morris, 2024-12-24:
please check which file includes the `esp_lcd_i2c.h` and how it include it. Please don't include it in the following way:

```c
#ifdef __cplusplus
extern "C" {
#endif

#include "esp_lcd_i2c.h"

#ifdef __cplusplus
}
#endif
```

gellary, 2024-12-24:
@suda-morris i do what you say and found     lv_port_disp.h > esp_lcd_panel_io.h >esp_lcd_io_i2c.h 

so i put lv_port_disp.h and esp_lcd_panel_io.h outside from extren "C"
program works smooth ~

@suda-morris Thank you so much, you save my life

closed 2024-12-24 by gellary