espressif/esp-idf #15085: Strange ledc_start_fade behaviour (more time than request)

StefanoTesla, 2024-12-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

running this code:

`
  ledc_channel_t ledcChannel = static_cast<ledc_channel_t>(0);
  esp_err_t fade_in_status = ledc_set_fade_with_time(LEDC_LOW_SPEED_MODE, ledcChannel, dutyValue, time);
  if (fade_in_status == ESP_OK){
    ledc_fade_start(LEDC_LOW_SPEED_MODE, ledcChannel, LEDC_FADE_NO_WAIT);
`

I got a strange behaviour, for example if I pass 5000 in the time, esp32 took approx 9 seconds to make the fade, If I use something like 2000 it works like a sharm...

I got an esp32-d v3.0 espressif version 6.9.0

of course at the beginning I run `ledc_fade_func_install(0);`

songruo, 2024-12-24:
Hi @StefanoTesla, our `ledc_set_fade_with_time` API does have the possibility of an inaccurate fade time, up to 50%, due to the the rounding of the scale/cycle parameters calculation internally (as a single linear fade). If you want a very accurate timing, you need to do some calculation by yourself to split the entire fade into multiple linear fades to avoid the rounding error.

StefanoTesla, 2024-12-27:
This make sense to me but in my case time encrease of 90%, I don't need precision but if I ask for 5seconds I wouldn't like to wait 9 second...

songruo, 2025-01-02:
Sorry, indeed up to 100% error.

A most extreme example:
For a 1kHz PWM, fade from 8000 to 0, in 4.001 sec

total_cycles = 4001
scale = duty_delta / total_cycles = (8000 - 0) / 4001 = 1 (round down)
step_num = duty_delta / scale = 8000 steps
8000 steps require 8 seconds to complete

So what you can do to improve the fade timing for this case can be:
step 1. fade from 8000 to 2, in 3.999 sec (scale = 2)
step 2. fade from 2 to 0, in 0.002 sec (scale = 1)

We will add some notes in the programming guide or API reference to clarify the fade behavior.

closed 2025-02-13 by espressif-bot