espressif/esp-idf #15086: Missing type declaration in i2c.html sample code

wellssam100, 2024-12-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In `/esp-idf/en/v5.3.2/esp32/api-reference/peripherals/i2c.html` there is sample code for using the Slave Read, written [here](https://docs.espressif.com/projects/esp-idf/en/v5.3.2/esp32/api-reference/peripherals/i2c.html#i2c-slave-read).<br>
On line 25, the following is written: `s_receive_queue = xQueueCreate(1, sizeof(i2c_slave_rx_done_event_data_t));`<br>Which should updated to be: `QueueHandle_t s_receive_queue = xQueueCreate(1, sizeof(i2c_slave_rx_done_event_data_t));` <br><br>Another example is given in `components/freertos/FreeRTOS-Kernel/include/freertos/queue.h` line 126, which uses the `QueueHandle_t` type for declaration. <br>
I believe that this code was written with some variables declared outside of the scope of the code and this is an artifact of that.

mythbuster5, 2024-12-24:
Thanks for your report, and we will fix it. However, we have a more useful i2c slave since v5.4 called i2c_slave_v2, and we encourage you to use that one. For this old version of i2c slave, we will not treat it as a high periority fix. So I'm going to close this one and find proper time to fix this doc on 5.3, thank you!

mythbuster5, 2025-01-13:
Problem solved .

closed 2025-01-13 by mythbuster5