espressif/esp-idf #15088: WIFI STA and AP Connection Issue (SDK 5.3.2)

varram-bdh, 2024-12-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.3.2

### Espressif SoC revision.

ESP32-WROOM-32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROOM-32

### Power Supply used.

External 3.3V

### What is the expected behavior?

There is an issue with connecting in STATION mode to a specific AP.

**Currently, there are no problems with other APs except for this particular one. While the AP itself might be suspected, all other devices I own, except for the ESP32, connect to the AP without any issues.**
I don't think there is an issue with the AP itself.

Additionally, there are no problems when using programs developed with ESP32 SDK 3.x. However, the issue occurs with SDK 4.x and SDK 5.x.

The security protocol is WPA2-PSK with AES encryption.

### What is the actual behavior?

Even when using the STA example program provided in SDK 5.3.2, the same issue occurs.
Path: v5.3.2\esp-idf\examples\wifi\getting_started\station

Additionally, the connection issue persists even when using the AT COMMAND method.

### Steps to reproduce.

1. The issue occurs in my application
2. Using the STATION example from SDK 5.3.2 - the same issue occurs.
3. Using the AT COMMAND method with the BIN file - the same issue occurs.
...

### Debug Logs.

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
[... 660 lines cut ...]
D (11150) wifi:disable connect timer
D (11150) wifi:clear scan ap list
D (11150) event: running post WIFI_EVENT:5 with handler 0x400def2c and context 0x3ffbdd0c on loop 0x3ffbcb54
D (11160) esp_netif_handlers: esp_netif action disconnected with netif0x3ffbda7c from event_id=5
D (11170) esp_netif_lwip: check: remote, if=0x3ffbda7c fn=0x400dce54
D (11180) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffbda7c
D (11180) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffbda7c
D (11190) esp_netif_lwip: if0x3ffbda7c start ip lost tmr: already started
D (11200) esp_netif_lwip: check: local, if=0x3ffbda7c fn=0x400dd164
D (11200) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbda7c
D (11210) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (11220) event: running post WIFI_EVENT:5 with handler 0x400d6be0 and context 0x3ffc5e3c on loop 0x3ffbcb54
I (11230) wifi station: connect to the AP fail
I (11230) wifi station: Failed to connect to SSID:SK_3724_2.4G, password:!!1cbdbad346
I (11240) main_task: Returned from app_main()
D (122910) esp_netif_lwip: esp_netif_ip_lost_timer esp_netif:0x3ffbda7c
D (122910) esp_netif_lwip: check: local, if=0x3ffbda7c fn=0x400dd164
D (122910) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffbda7c
D (122910) esp_netif_lwip: if0x3ffbda7c ip lost tmr: raise ip lost event
D (122920) event: no handlers have been registered for event IP_EVENT:1 posted to loop 0x3ffbcb54
```

### More Information.

I also think it might be a timing issue. Occasionally, the connection succeeds during repeated attempts. This occurs when the log level is set to DEBUG.

Xiehanxin, 2024-12-25:
hi @varram-bdh could you share your router setting page? I want to confirm th settings, and could you tell me you specific router brand?

varram-bdh, 2024-12-26:
hi @Xiehanxin The router currently causing the issue is provided by SK Telecom in Korea.
Model: GW-ME611R
Manufacturer/Country of Origin: Mercury Co., Ltd./China

I will update the page soon.

varram-bdh, 2024-12-26:
As an additional update, I found that the event handler is generating an ID of 43. This is not documented anywhere.

varram-bdh, 2024-12-26:
[image]
[image]
[image]
[image]

varram-bdh, 2025-01-08:
@Xiehanxin  hi.
Is there anything else that needs to be checked?

zhangyanjiaoesp, 2025-02-19:
@varram-bdh 
Please use the new wifi libs to replace the libs in `idf/componts/esp_wifi/lib/esp32`, have test again, thanks.

[32-wifi-lib.zip](https://github.com/user-attachments/files/18863795/32-wifi-lib.zip)
wifi firmware version: 2676547

AxelLin, 2025-03-05:
@varram-bdh Any update regarding https://github.com/espressif/esp-idf/issues/15088#issuecomment-2667933999

varram-bdh, 2025-03-10:
@AxelLin @zhangyanjiaoesp 
Sorry for keeping you waiting.

We had previously reported the issue to the telecom provider regarding their router, and after their analysis, they made the necessary modifications on their end. Currently, all routers have undergone an OTA update, and the issue has been resolved.

We were unable to test the lib file you provided, as the routers have already been upgraded.

Additionally, I would like to ask whether the encryption method WPA-PSK is not supported.
In SDK 5.3.2, the connection is currently failing, and this issue is common across all routers.

Below is the log when connecting to a WPA-PSK router.
The program used is station_example_main.c.
I set ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD to WPA-PSK, but the issue persists.
The same issue also occurs when set to OPEN.

I (31) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (31) boot: compile time Dec 30 2024 10:03:45
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v3.0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=18688h ( 99976) map
I (139) esp_image: segment 1: paddr=000286b0 vaddr=3ffb0000 size=03ee0h ( 16096) load
I (145) esp_image: segment 2: paddr=0002c598 vaddr=40080000 size=03a80h ( 14976) load
I (151) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=7908ch (495756) map
I (323) esp_image: segment 4: paddr=000a90b4 vaddr=40083a80 size=13b94h ( 80788) load
I (367) boot: Loaded app from partition at offset 0x10000
I (367) boot: Disabling RNG early entropy source...
I (379) cpu_start: Multicore app
I (388) cpu_start: Pro cpu start user code
I (388) cpu_start: cpu freq: 160000000 Hz
I (388) app_init: Application information:
I (391) app_init: Project name:     wifi_station
I (396) app_init: App version:      v5.3.2-dirty
I (401) app_init: Compile time:     Mar 10 2025 13:30:03
I (407) app_init: ELF file SHA256:  42b51dfa7...
I (412) app_init: ESP-IDF:          v5.3.2-dirty
I (418) efuse_init: Min chip rev:     v0.0
I (423) efuse_init: Max chip rev:     v3.99
I (428) efuse_init: Chip rev:         v3.0
I (433) heap_init: Initializing. RAM available for dynamic allocation:
I (440) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (446) heap_init: At 3FFB8110 len 00027EF0 (159 KiB): DRAM
I (452) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (458) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (465) heap_init: At 40097614 len 000089EC (34 KiB): IRAM
I (472) spi_flash: detected chip: generic
I (475) spi_flash: flash io: dio
W (479) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (493) main_task: Started on CPU0
I (503) main_task: Calling app_main()
I (523) wifi station: ESP_WIFI_MODE_STA
I (533) wifi:wifi driver task: 3ffc00f8, prio:23, stack:6656, core=0
I (543) wifi:wifi firmware version: 2676547e0b
I (543) wifi:wifi certification version: v7.0
I (543) wifi:config NVS flash: enabled
I (543) wifi:config nano formating: disabled
I (543) wifi:Init data frame dynamic rx buffer num: 32
I (553) wifi:Init static rx mgmt buffer num: 5
I (553) wifi:Init management short buffer num: 32
I (563) wifi:Init dynamic tx buffer num: 32
I (563) wifi:Init static rx buffer size: 1600
I (573) wifi:Init static rx buffer num: 10
I (573) wifi:Init dynamic rx buffer num: 32
I (573) wifi_init: rx ba win: 6
I (583) wifi_init: accept mbox: 6
I (583) wifi_init: tcpip mbox: 32
I (583) wifi_init: udp mbox: 6
I (593) wifi_init: tcp mbox: 6
I (593) wifi_init: tcp tx win: 5760
I (603) wifi_init: tcp rx win: 5760
I (603) wifi_init: tcp mss: 1440
I (603) wifi_init: WiFi IRAM OP enabled
I (613) wifi_init: WiFi RX IRAM OP enabled
W (613) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
I (623) phy_init: phy_version 4840,02e0d70,Sep  2 2024,19:39:07
I (703) wifi:mode : sta (94:e6:86:98:05:34)
I (703) wifi:enable tsf
E (713) wifi station: WIFI_EVENT 43
I (713) wifi station: wifi_init_sta finished.
E (713) wifi station: WIFI_EVENT 2
I (713) wifi station: WIFI_EVENT_STA_START
E (3133) wifi station: WIFI_EVENT 5
I (3133) wifi station: WIFI_EVENT_STA_DISCONNECTED
I (3133) wifi station: ---------------------
I (3143) wifi station: 201 : CANT FIND AP

I (3143) wifi station: retry to connect to the AP
I (3153) wifi station: connect to the AP fail
E (5563) wifi station: WIFI_EVENT 5
I (5563) wifi station: WIFI_EVENT_STA_DISCONNECTED
I (5563) wifi station: ---------------------
I (5563) wifi station: 201 : CANT FIND AP

I (5573) wifi station: retry to connect to the AP
I (5573) wifi station: connect to the AP fail
E (7983) wifi station: WIFI_EVENT 5
I (7983) wifi station: WIFI_EVENT_STA_DISCONNECTED
I (7983) wifi station: ---------------------
I (7983) wifi station: 201 : CANT FIND AP

mhdong, 2025-03-10:
hi @varram-bdh 
Authmode threshold resets to WPA2 as default if password matches WPA2 standards (password len => 8). If you want to connect the device to deprecated WEP/WPA networks, Please set the threshold value to WIFI_AUTH_WEP/WIFI_AUTH_WPA_PSK and set the password with length and format matching to WIFI_AUTH_WEP/WIFI_AUTH_WPA_PSK standards.

D (4372) wifi:Send disconnect event, reason=211
WIFI_REASON_NO_AP_FOUND_IN_AUTHMODE_THRESHOLD  = 211,    /**< No AP found in auth mode threshold */

varram-bdh, 2025-03-10:
@mhdong 
Thank you for your response.
When performing a product OTA update, customers using the existing AP experience connection difficulties.

Is this the expected behavior?

In SDK v4, the connection worked normally.

mhdong, 2025-03-10:
hi @varram-bdh 
If there are no special configurations, it should not affect the connection. However, if the connection becomes significantly more difficult after an OTA update, please provide logs and packet captures so that we can analyze the cause of the connection issues.

varram-bdh, 2025-03-10:
hi @mhdong 
It was stated that the password must be less than 8 characters to connect using WPA-PSK, but in recent routers, whether by mistake or not, it is still possible to set a password longer than 8 characters in WPA-PSK mode.

Therefore, I tested this using the example program and provided the logs above.

The logs above are from a WPA-PSK connection where the password length is 13 characters.
This issue occurs after performing an OTA update from an SDK 4 version program to an SDK 5 applied program, where the device fails to connect to the AP.
At this time, the router is set to WPA-PSK, and changing it to WPA2 resolves the issue.
However, some customers are unable to change the AP settings, so the issue remains.

mhdong, 2025-03-11:
hi @varram-bdh 
Sorry, my explanation yesterday was not accurate. We reset the authmode threshold to WAP2 if the user not explictly set it(auth mode threshold == open) and password matches WPA2 standards (password len => 8).
Yes, it is possible to set a password longer than 8 characters in WPA-PSK mode. 
If you want to connect to a WPA router, your auth mode threshold should be set to WPA or lower. However, if you haven't set it or have set it to open, and the password length is 8 or more, the auth mode threshold will be automatically set to WPA2, which will prevent connection to a WPA router.
Our modification aims to ensure that if the device is connecting to **deprecated** security modes like WPA or WEP, the user is explicitly aware of it and making a deliberate choice.
If you want to connect to a router with WPA mode, simply set the auth mode threshold to WPA. This allows the device to connect to networks with encryption levels equal to or higher than WPA.
Please reference :            https://github.com/espressif/esp-idf/blob/d5b84196206a0bde16bd43f4cdf10cdd4ebee9b0/examples/wifi/getting_started/station/main/station_example_main.c#L129

varram-bdh, 2025-03-11:
hi @mhdong 
Thank you for your quick help.

I have confirmed that the connection was established with a WPA setting other than OPEN.
It seems that the OPEN setting was the issue.
I appreciate your assistance.

closed 2025-03-12 by varram-bdh