espressif/esp-idf #15089: Disabling optimization ("-O0") for debugging causes stack overflow

gamename, 2024-12-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

While debugging, lots of vars had been "optimized away". So, I turned off optimization ("-O0") via menuconfig and recompiled.

I expected the firmware to run and allow me to debug via vscode.  

### What is the actual behavior?

A stack overflow crash during boot.  Could not debug.

### Steps to reproduce.

1. Turn off optimization `(Compiler Options -> Optimization Level -> Debug without optimization (-O0))`

2. `idf.py build flash monitor`

### Debug Logs.

```plain
Executing action: all (aliases: build)
Running ninja in directory /Users/tennis/src/C2DS-egg/build
Executing "ninja all"...
[0/1] Re-running CMake...CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- Component directory /Users/tennis/src/C2DS-egg/components/esp-aws-iot/libraries/common does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/tennis/src/C2DS-egg/components/esp-aws-iot does not contain a CMakeLists.txt file. No component will be added
-- Component directory /Users/tennis/src/C2DS-egg/components/mp3 does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32s3
NOTICE: Processing 4 dependencies:
NOTICE: [1/4] espressif/cbor (0.6.0~1)
[... 469 lines cut ...]
I (3641) wifi:state: assoc -> run (0x10)
I (3761) wifi:connected with disconnected_iot, aid = 47, channel 3, BW20, bssid = 76:fe:ce:47:5c:f6
I (3761) wifi:security: WPA2-PSK, phy: bgn, rssi: -55
I (3761) wifi:pm start, type: 1

I (3761) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3771) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (3801) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (04:33:01.352) core_mqtt_agent_manager: WiFi connected.
I (04:33:01.353) app_wifi: Connected with IP Address:192.168.0.125
I (04:33:01.356) esp_netif_handlers: sta ip: 192.168.0.125, mask: 255.255.255.0, gw: 192.168.0.1
I (4791) wifi:<ba-add>idx:0 (ifx:0, 76:fe:ce:47:5c:f6), tid:0, ssn:5, winSize:64
I (04:33:01.372) main_task: Returned from app_main()
I (4831) wifi:<ba-add>idx:1 (ifx:0, 76:fe:ce:47:5c:f6), tid:2, ssn:1, winSize:64

***ERROR*** A stack overflow in task CoreMqttAgentConnectionTask has been detected.

Backtrace: 0x4037612e:0x3fcc9f40 0x40384081:0x3fcc9f70 0x403857e2:0x3fcc9fa0 0x4038824f:0x3fcca0a0 0x403859dc:0x3fcca0e0 0x403859d2:0x0000002f |<-CORRUPTED
--- 0x4037612e: panic_abort at /Users/tennis/esp/esp-idf-v5.3.1/components/esp_system/panic.c:463
0x40384081: esp_system_abort at /Users/tennis/esp/esp-idf-v5.3.1/components/esp_system/port/esp_system_chip.c:92
0x403857e2: vApplicationStackOverflowHook at /Users/tennis/esp/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:554
0x4038824f: vTaskSwitchContext at /Users/tennis/esp/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/tasks.c:3701 (discriminator 7)
0x403859dc: _frxt_dispatch at /Users/tennis/esp/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x403859d2: _frxt_int_exit at /Users/tennis/esp/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246
```

igrr, 2024-12-26:
> ***ERROR*** A stack overflow in task CoreMqttAgentConnectionTask has been detected.

"CoreMqttAgentConnectionTask" doesn't look like a task created by ESP-IDF built-in components? Is it a task you have created? If yes, you need to increase the stack size, as the error indicates.

gamename, 2024-12-26:
@igrr 
> > _**ERROR**_ A stack overflow in task CoreMqttAgentConnectionTask has been detected.
> 
> "CoreMqttAgentConnectionTask" doesn't look like a task created by ESP-IDF built-in components? Is it a task you have created? If yes, you need to increase the stack size, as the error indicates.

Ah, ok. I missed that. Thanks!

closed 2024-12-26 by gamename