espressif/esp-idf #15090: esp_rom_delay_us crash  and log show ets_delay_us in ROM

MATACE, 2024-12-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-S

### Power Supply used.

USB

### What is the expected behavior?

esp_rom_delay_us can be normal microsecond delay

### What is the actual behavior?

0x40041a79: ets_delay_us in ROM
0x42008f15: app_main at F:/CodeProject/ESP32/IDF_LVGL/main/main.c:99
0x4201c497: main_task at F:/SsdSofawareInstall/ESP32_IDF/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/app_startup.c:208
0x4037af79: vPortTaskWrapper at F:/SsdSofawareInstall/ESP32_IDF/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

E (50312) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:E (50312) task_wdt:  - IDLE0 (CPU 0)
E (50312) task_wdt: Tasks currently running:
E (50312) task_wdt: CPU 0: main
E (50312) task_wdt: CPU 1: IDLE1
E (50312) task_wdt: Print CPU 0 (current core) backtrace

 esp_rom_delay_us crash: ets_delay_us in ROM

### Steps to reproduce.

code like this:

static int delay_ms[3] = {10, 20, 30};

void delay_test(void *args)
{
    int i = 0;
    
    while (1)
    {
        esp_rom_delay_us(delay_ms[i] * 1000);
        i++;
        if (i == 3) {
            i = 0;
        }
    }
}

void app_main(void)
{
.....
.....
xTaskCreate(delay_test, "delay_test", 4096, NULL, 5, NULL);
.....
}

### Debug Logs.

```plain
E (40312) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:E (40312) task_wdt:  - IDLE0 (CPU 0)
E (40312) task_wdt: Tasks currently running:
E (40312) task_wdt: CPU 0: main
E (40312) task_wdt: CPU 1: IDLE1
E (40312) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4200B707:0x3FC94440 0x4200BB24:0x3FC94460 0x4037732D:0x3FC94490 0x40041A79:0x3FC99410 0x42008F15:0x3FC99430 0x4201C497:0x3FC99450 0x4037AF79:0x3FC99480
0x4200b707: task_wdt_timeout_handling at F:/SsdSofawareInstall/ESP32_IDF/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/task_wdt/task_wdt.c:434
0x4200bb24: task_wdt_isr at F:/SsdSofawareInstall/ESP32_IDF/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/task_wdt/task_wdt.c:507
0x4037732d: _xt_lowint1 at F:/SsdSofawareInstall/ESP32_IDF/Espressif/frameworks/esp-idf-v5.3.1/components/xtensa/xtensa_vectors.S:1240
0x40041a79: ets_delay_us in ROM
0x42008f15: app_main at F:/CodeProject/ESP32/IDF_LVGL/main/main.c:99
0x4201c497: main_task at F:/SsdSofawareInstall/ESP32_IDF/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/app_startup.c:208
0x4037af79: vPortTaskWrapper at F:/SsdSofawareInstall/ESP32_IDF/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

suda-morris, 2024-12-25:
Because you created a task that never yield the CPU (the task is stuck in an infinite loop).

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/wdts.html#task-watchdog-timer-twdt

ESP-Marius, 2024-12-25:
As already explained by @suda-morris, this is expected behavior, if you busy-wait (which is what `esp_rom_delay_us` does) then the WDT will indeed trigger as it is intended to do.

If you want delays that actually yield the task and let the other tasks (and IDLE) run then please take a look at [vTaskDelay](https://www.freertos.org/Documentation/02-Kernel/04-API-references/02-Task-control/01-vTaskDelay)

closed 2024-12-25 by ESP-Marius