espressif/esp-idf #15093: Getting "undefined reference to `app_main`" in link step

gamename, 2024-12-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I just added a new component (ESP-ADF) and was expecting it to compile as usual.

### What is the actual behavior?

Got a strange "undefined reference to `app_main`" in the link step.

### Steps to reproduce.

1. Add `esp-adf` to the components subdir (per the instructions [here](https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html#get-started-setup-esp-idf)
2. recompile 

### Build or installation Logs.

```plain
Executing action: all (aliases: build)
Running ninja in directory /Users/tennis/src/C2DS-egg/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /Users/tennis/src/C2DS-egg/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.1/components/soc/esp32s3/ld/esp32s3.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.wdt.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.version.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.1/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.1/components/bootloader/subproject/main/ld/esp32s3/bootloader.ld
-- Adding linker script /Users/tennis/esp/esp-idf-v5.3.1/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /Users/tennis/esp/esp-idf-v5.3.1/components/bootloader /Users/tennis/esp/esp-idf-v5.3.1/components/bootloader_support /Users/tennis/esp/esp-idf-v5.3.1/components/efuse /Users/tennis/esp/esp-idf-v5.3.1/components/esp_app_format /Users/tennis/esp/esp-idf-v5.3.1/components/esp_bootloader_format /Users/tennis/esp/esp-idf-v5.3.1/components/esp_common /Users/tennis/esp/esp-idf-v5.3.1/components/esp_hw_support /Users/tennis/esp/esp-idf-v5.3.1/components/esp_rom /Users/tennis/esp/esp-idf-v5.3.1/components/esp_system /Users/tennis/esp/esp-idf-v5.3.1/components/esptool_py /Users/tennis/esp/esp-idf-v5.3.1/components/freertos /Users/tennis/esp/esp-idf-v5.3.1/components/hal /Users/tennis/esp/esp-idf-v5.3.1/components/log /Users/tennis/esp/esp-idf-v5.3.1/components/bootloader/subproject/main /Users/tennis/esp/esp-idf-v5.3.1/components/bootloader/subproject/components/micro-ecc /Users/tennis/esp/esp-idf-v5.3.1/components/newlib /Users/tennis/esp/esp-idf-v5.3.1/components/partition_table /Users/tennis/esp/esp-idf-v5.3.1/components/soc /Users/tennis/esp/esp-idf-v5.3.1/components/spi_flash /Users/tennis/esp/esp-idf-v5.3.1/components/xtensa
-- Configuring done (3.6s)
-- Generating done (0.1s)
-- Build files have been written to: /Users/tennis/src/C2DS-egg/build/bootloader
[112/114] Generating binary image from built executable
esptool.py v4.8.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /Users/tennis/src/C2DS-egg/build/bootloader/bootloader.bin
[113/114] cd /Users/tennis/src/C2DS-egg/build/bootloader/esp-idf/esptool_py && /Users/tennis/.espress...ck_sizes.py --offset 0xb000 bootloader 0x0 /Users/tennis/src/C2DS-egg/build/bootloader/bootloader.bin
Bootloader binary size 0x4900 bytes. 0x6700 bytes (59%) free.
[676/857] Building C object esp-idf/esp-adf-libs/CMakeFiles/__idf_esp-adf-libs.dir/media_lib_sal/port/media_lib_os_freertos.c.obj
In file included from /Users/tennis/src/C2DS-egg/components/esp-adf/components/esp-adf-libs/media_lib_sal/port/media_lib_os_freertos.c:43:
[... 77 lines cut ...]
    gecl-rgb-led-manager
    gecl-time-sync-manager
    json
    esp_gdbstub
)

idf_component_register(
    SRCS
        ${MAIN_SRCS}
        ${COMPONENT_SRC}
    INCLUDE_DIRS
        ${MAIN_INCLUDE_DIRS}
        ${COMPONENT_INCLUDE_DIRS}
    REQUIRES
        ${MAIN_REQUIRES}
)

# OTA demo
if( CONFIG_GRI_ENABLE_OTA_DEMO OR CONFIG_GRI_RUN_QUALIFICATION_TEST )
    target_add_binary_data(${COMPONENT_TARGET} "certs/aws_codesign.crt" TEXT)
endif()

# Root Certificate
target_add_binary_data(${COMPONENT_TARGET} "certs/root_cert_auth.crt" TEXT)
```

dobairoland, 2024-12-26:
> Add esp-adf to the components subdir (per the instructions [here](https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html#get-started-setup-esp-idf)

Could you please be more specific where did you find this instruction? I don't know too well ESP-ADF. The ESP-ADF examples use a completely different approach to you. See for example https://github.com/espressif/esp-adf/tree/ab43323e70f3f0aed25073117b1337b06675dc96/examples/get-started/play_mp3_control

My guess is that you need here `main` as well: 

> set(COMPONENTS "esp-adf-libs" "audio_pipeline" "audio_sal")

gamename, 2024-12-26:
> My guess is that you need here `main` as well:
> 
> > set(COMPONENTS "esp-adf-libs" "audio_pipeline" "audio_sal")

That was it! Thanks!

closed 2024-12-26 by gamename