espressif/esp-idf #15094: ESP32S3使用ADC1过一段时间测量电压偏高

WzxKB, 2024-12-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.3.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

ESP32S3使用IO7，ADC1，channel5，采集一氧化碳的传感器ADC，实际采集电压应该是几百毫伏

### What is the actual behavior?

ESP32S3使用IO7，ADC1，channel5，采集一氧化碳的传感器ADC，使用了ESPNOW，不过连不连接wifi，都会再一段时间后，采集的电压偏高，一开始采集的是200毫伏，过一段时间后采集的电压就稳定变为2伏多，使用万用表测量电压确实如此。断开ADC脚和一氧化碳adc采集脚的连接，发现eso32S3电压采集脚有1伏多的电压

### Steps to reproduce.

1、ESP32S3 ADC初始化
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/timers.h"
#include "onboard_sensor.h"
#include "soc/soc_caps.h"
#include "esp_log.h"
#include "hal/adc_types.h"
#include "esp_adc/adc_oneshot.h"
#include "esp_adc/adc_cali.h"
#include "esp_adc/adc_cali_scheme.h"
#include "my_adc.h"
#include "my_sync.h"
#include "my_nvs.h"

const static char *TAG = "my_adc";

static adc_oneshot_unit_handle_t adc1_handle;
static bool do_calibration1_chan0 = false;
static adc_cali_handle_t adc1_cali_chan0_handle = NULL;

#define CM_ONBOARD_ADC1_CHAN ADC_CHANNEL_5
#define CM_ONBOARD_ADC1_ATTEN ADC_ATTEN_DB_12
#define CM_ONBOARD_ADC1_BITWIDTH ADC_BITWIDTH_12

static bool adc_calibration_init(adc_unit_t unit, adc_channel_t channel, adc_atten_t atten, adc_cali_handle_t *out_handle);

/// @brief 初始化ADC模块，包括配置ADC1及其通道，并进行校准初始化
/// @return ESP_OK: 成功; 其他: 错误代码
static esp_err_t mq_adc_init(void)
{
    esp_err_t ret = ESP_OK;
    //-------------ADC1 Init---------------//
    adc_oneshot_unit_init_cfg_t init_config1 = {
        .unit_id = ADC_UNIT_1,
        .ulp_mode = ADC_ULP_MODE_DISABLE,
    };
    ret = adc_oneshot_new_unit(&init_config1, &adc1_handle);
    if (ret != ESP_OK)
    {
        return ret;
    }

    //-------------ADC1 Config---------------//
    adc_oneshot_chan_cfg_t config = {
        .bitwidth = CM_ONBOARD_ADC1_BITWIDTH,
        .atten = CM_ONBOARD_ADC1_ATTEN,
    };
    ret = adc_oneshot_config_channel(adc1_handle, CM_ONBOARD_ADC1_CHAN, &config);
    if (ret != ESP_OK)
    {
        return ret;
    }

    //-------------ADC1 Calibration Init---------------//
    do_calibration1_chan0 = adc_calibration_init(ADC_UNIT_1, CM_ONBOARD_ADC1_CHAN, CM_ONBOARD_ADC1_ATTEN, &adc1_cali_chan0_handle);

    return ESP_OK;
}

/// @brief 从ADC读取值并进行校准
/// @param[out] mq_adc_value 读取的ADC值（校准后为电压值，未校准为原始值）
/// @return ESP_OK: 成功; 其他: 错误代码
esp_err_t mq_read(int *mq_adc_value)
{
    int adc_value = 0;
    int voltage;
    esp_err_t ret = adc_oneshot_read(adc1_handle, CM_ONBOARD_ADC1_CHAN, &adc_value);
    if (ret != ESP_OK)
    {
        *mq_adc_value = 0;
        return ret;
    }
    else
    {
        if (do_calibration1_chan0)
        {
            ret = adc_cali_raw_to_voltage(adc1_cali_chan0_handle, adc_value, &voltage);
            if (ret != ESP_OK)
            {
                *mq_adc_value = adc_value;
                ESP_LOGW(TAG, "校准系统损坏,未校准ADC数据");
                return ESP_OK;
            }
            else
            {
                *mq_adc_value = voltage;
                return ESP_OK;
            }
        }
        else
        {
            *mq_adc_value = adc_value;
            ESP_LOGW(TAG, "校准系统损坏,未校准ADC数据");
            return ESP_OK;
        }
    }
    return ESP_FAIL;
}

/// @brief 初始化ADC校准
/// @param[in] unit ADC单元
/// @param[in] channel ADC通道
/// @param[in] atten ADC衰减
/// @param[out] out_handle ADC校准句柄
/// @return true: 校准成功; false: 校准失败
static bool adc_calibration_init(adc_unit_t unit, adc_channel_t channel, adc_atten_t atten, adc_cali_handle_t *out_handle)
{
    adc_cali_handle_t handle = NULL;
    esp_err_t ret = ESP_FAIL;
    bool calibrated = false;
#if ADC_CALI_SCHEME_CURVE_FITTING_SUPPORTED
    if (!calibrated)
    {
        ESP_LOGI(TAG, "calibration scheme version is %s", "Curve Fitting");
        adc_cali_curve_fitting_config_t cali_config = {
            .unit_id = unit,
            .chan = channel,
            .atten = atten,
            .bitwidth = CM_ONBOARD_ADC1_BITWIDTH,
        };
        ret = adc_cali_create_scheme_curve_fitting(&cali_config, &handle);
        if (ret == ESP_OK)
        {
            calibrated = true;
        }
    }
#endif

#if ADC_CALI_SCHEME_LINE_FITTING_SUPPORTED
    if (!calibrated)
    {
        ESP_LOGI(TAG, "calibration scheme version is %s", "Line Fitting");
        adc_cali_line_fitting_config_t cali_config = {
            .unit_id = unit,
            .atten = atten,
            .bitwidth = CM_ONBOARD_ADC1_BITWIDTH,
        };
        ret = adc_cali_create_scheme_line_fitting(&cali_config, &handle);
        if (ret == ESP_OK)
        {
            calibrated = true;
        }
    }
#endif

    *out_handle = handle;
    if (ret == ESP_OK)
    {
        ESP_LOGI(TAG, "Calibration Success");
    }
    else if (ret == ESP_ERR_NOT_SUPPORTED || !calibrated)
    {
        ESP_LOGW(TAG, "eFuse not burnt, skip software calibration");
    }
    else
    {
        ESP_LOGE(TAG, "Invalid arg or no memory");
    }

    return calibrated;
}

/// @brief 初始化MQ（包括ADC）
/// @return ESP_OK: 成功; 其他: 错误代码
esp_err_t mq_init(void)
{
    esp_err_t ret = ESP_OK;
    ret = mq_adc_init();
    if (ret != ESP_OK)
    {
        return ret;
    }
    return ret;
}

### Build or installation Logs.

```plain
无
```

### More Information.

文档上找到的是ADC2和WIFI不能共用，但是这个是ADC1脚，所以很奇怪，在两个板子上都复现了

wanckl, 2024-12-27:
@WzxKB Hi,
> 使用万用表测量电压确实如此。断开ADC脚和一氧化碳adc采集脚的连接，发现eso32S3电压采集脚有1伏多的电压

ADC引脚为浮空输入脚，断开后测该脚没有意义，，可以测一下：
1. 不断开时候该脚是否电压变高
2. 断开后，传感器输出脚电压是否变高

😸

WzxKB, 2024-12-27:
> @WzxKB Hi,
> 
> > 使用万用表测量电压确实如此。断开ADC脚和一氧化碳adc采集脚的连接，发现eso32S3电压采集脚有1伏多的电压
> 
> ADC引脚为浮空输入脚，断开后测该脚没有意义，，可以测一下：
> 
> 1. 不断开时候该脚是否电压变高
> 2. 断开后，传感器输出脚电压是否变高
> 
> 😸
@wanckl  Hi,
你好，测量发现，不断开时候，ESP32S3上的ADC测量脚会升高，在开机使用一段时间后（几分钟），电压就会升高到2V多，ADC检测的值也变成2v多，断开后测量一氧化碳传感器脚，一直维持在稳定的200mv左右

wanckl, 2024-12-30:
@WzxKB emmm
可以帮助测一下 直接不开启wifi/不使用 ESP_NOW，甚至直接只开启ADC，一段时间后是否还有问题吗，以定位更准确的原因，感谢

Bruce297, 2024-12-30:
> ESP32S3使用IO7，ADC1，channel5，采集一氧化碳的传感器ADC

channel5是IO6

Alvin1Zhang, 2025-02-07:
@WzxKB 您好，麻烦可以提供更多反馈么？谢谢。

closed 2025-05-08 by Alvin1Zhang