espressif/esp-idf #15095: 请问　mcpwm_sync　样板演示的是同一个unit的三个timer的同步，是否可以让mcpwm的两个unit的所有timer完成同步呢？

nhchmg, 2024-12-26:
### Is your feature request related to a problem?

请问　mcpwm_sync　样板演示的是同一个unit的三个timer的同步，是否可以让mcpwm的两个unit的所有timer完成同步呢？　是否可以让两个unit的timer0都设为从同一个软件源同步？然后每个unit的timer1和timer2又从自己单元的timer0同步，是否可行？

suda-morris, 2024-12-26:
MCPWM0内部的timer肯定是不可以去同步MCPWM1内部的timer的，他们之间没有信号连接。但是借助外部的GPIO，应该是可以做到让MCPWM0和MCPWM1内部的timers被同一个GPIO的边沿信号给同步

nhchmg, 2024-12-27:
> MCPWM0内部的timer肯定是不可以去同步MCPWM1内部的timer的，他们之间没有信号连接。但是借助外部的GPIO，应该是可以做到让MCPWM0和MCPWM1内部的timers被同一个GPIO的边沿信号给同步

1.请问能否用软件同步这种不占用ＩＯ口的方法，让６个mcpwm得到同步？

suda-morris, 2025-01-02:
@nhchmg 不可以，软件做不到**同时**写MCPWM0和MCPWM1相同的寄存器（各自内部都有一个trigger寄存器）

closed 2025-01-06 by suda-morris