espressif/esp-idf #15097: Branching never executes one branch

safocl, 2024-12-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

In code
https://github.com/espressif/esp-idf/blob/b5ac4fbdf9e9fb320bb0a98ee4fbaa18f8566f37/components/wpa_supplicant/src/rsn_supp/wpa.c#L245
the **true** branch will never start -- **wpa_sm_get_bssid** function always return 0.

safocl, 2024-12-26:
https://github.com/espressif/esp-idf/blob/b5ac4fbdf9e9fb320bb0a98ee4fbaa18f8566f37/components/wpa_supplicant/src/rsn_supp/wpa.c#L197 this line means that the return is invalid.

safocl, 2024-12-26:
https://w1.fi/wpa_supplicant/devel/wpa__i_8h_source.html contain the implementation of **wpa_sm_get_bssid** function:
```
169 static inline int wpa_sm_get_bssid(struct [wpa_sm](https://w1.fi/wpa_supplicant/devel/structwpa__sm.html) *sm, u8 *bssid)
170 {
171         WPA_ASSERT(sm->ctx->get_bssid);
172         return sm->ctx->get_bssid(sm->ctx->ctx, bssid);
173 }
```
is this what should be in place of the existing code?

nopnop2002, 2024-12-26:
memcpy() function returns a pointer to dest.
This comment is incorrect.
https://github.com/espressif/esp-idf/blob/b5ac4fbdf9e9fb320bb0a98ee4fbaa18f8566f37/components/wpa_supplicant/src/rsn_supp/wpa.c#L197
wpa_sm_get_bssid() always returns 0.

closed 2025-01-23 by espressif-bot

safocl, 2025-01-23:
https://github.com/espressif/esp-idf/commit/22897aaf5bd1c7f215d3ef9c6d8cb1bf098f60f2 left dangerous code that now runs a non-existent (`#ifndef ESP_SUPPLICANT`).
https://github.com/espressif/esp-idf/blob/d30e4d9960afd0d1ed067fef741126af810f675e/components/wpa_supplicant/src/rsn_supp/wpa.c#L235

kapilkedawat, 2025-01-23:
@safocl `ESP_SUPPLICANT` is always defined for esp-idf supplicant, code was left intentionally.

safocl, 2025-01-23:
> [@safocl](https://github.com/safocl) `ESP_SUPPLICANT` is always defined for esp-idf supplicant, code was left intentionally.

ok