espressif/esp-idf #15098: Undefined behavior in wpa_eapol_key_send function

safocl, 2024-12-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://github.com/espressif/esp-idf/blob/b5ac4fbdf9e9fb320bb0a98ee4fbaa18f8566f37/components/wpa_supplicant/src/rsn_supp/wpa.c#L245 contain UB with use **wpa_sm_get_bssid(sm, sm->bssid)** ->
```
static inline int   wpa_sm_get_bssid(struct wpa_sm *sm, u8 *bssid)
{
    memcpy(bssid, sm->bssid, ETH_ALEN);
    return 0;
}
```
copy from **sm->bssid** to **bssid** that is **sm->bssid** ->
"**If the objects overlap (which is a violation of the [restrict](https://en.cppreference.com/w/c/language/restrict) contract)(since C99), the behavior is undefined.**" https://en.cppreference.com/w/c/string/byte/memcpy

KaeLL, 2025-01-13:
Nice catch.

closed 2025-01-23 by espressif-bot

safocl, 2025-01-23:
https://github.com/espressif/esp-idf/issues/15097#issuecomment-2609644167