espressif/esp-idf #15103: [ESP32] Linker error for esp_fill_random when using C++ <random> RNG

MayaPosch, 2024-12-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I expected the code to compile and link, as the only significant change was adding an RNG using the standard C++ header.

### What is the actual behavior?

Instead I get the linker error: `[..]components/newlib/random.c:31:(.text.getrandom+0x18): undefined reference to 'esp_fill_random'`

### Steps to reproduce.

1. Add the C++ `<random>` header.
2. Set up a new RNG:
```
std::random_device rd;
std::mt19937 gen(rd());
```
3. Try to compile code.
4. Observe the linker error: `[..]components/newlib/random.c:31:(.text.getrandom+0x18): undefined reference to 'esp_fill_random'`

### Build or installation Logs.

```plain
D:/Dev/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/newlib/libnewlib.a(random.c.obj): in function `getrandom':
D:/Dev/Espressif/frameworks/esp-idf-v5.2.2/components/newlib/random.c:31:(.text.getrandom+0x18): undefined reference to `esp_fill_random'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

### More Information.

I upgraded from ESP-IDF v5.2.2 to the current stable (v5.3.2, in the same folder), without changes to the error.

igrr, 2024-12-27:
Thanks for the report @MayaPosch! We have an issue open for this in the internal tracker, but haven't yet addressed it unfortunately. Sometimes this issue manifests in `::getrandom` itself not being found at link time. The root cause is related to how the dependencies between libstdc++ library, "newlib" and "esp_hw_support" components are set up.

The workaround for the time being is to force `esp_fill_random` to be considered for inclusion into the binary by the linker, for example by adding
```cmake
target_link_libraries(${COMPONENT_LIB} PUBLIC "-Wl,-u esp_fill_random")
```
to one of your components' CMakeLists.txt files.

MayaPosch, 2024-12-27:
Thank you, I can confirm that the workaround allows the code to link at least. I'll be running some tests to hopefully confirm that it fully fixes the issue for now.

Is there any projection for when this issue is expected to fixed in an upcoming ESP-IDF release?

igrr, 2024-12-27:
Will try to get this fixed before v5.3.3, however I don't yet have an estimate of the date the release will be ready.