espressif/esp-idf #15104: Cannot run i2s component tests

gamename, 2024-12-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

When presented with this menu item on the console:
```I (307) main_task: Calling app_main()
  ___ ____  ____    _____         _   
 |_ _|___ \/ ___|  |_   _|__  ___| |_ 
  | |  __) \___ \    | |/ _ \/ __| __|
  | | / __/ ___) |   | |  __/\__ \ |_ 
 |___|_____|____/    |_|\___||___/\__|

Press ENTER to see the list of tests.
```

I should be able to press ENTER and get a list of tests to run

### What is the actual behavior?

Was presented with this menu item on the console:
```I (307) main_task: Calling app_main()
  ___ ____  ____    _____         _   
 |_ _|___ \/ ___|  |_   _|__  ___| |_ 
  | |  __) \___ \    | |/ _ \/ __| __|
  | | / __/ ___) |   | |  __/\__ \ |_ 
 |___|_____|____/    |_|\___||___/\__|

Press ENTER to see the list of tests.
```

When I pressed enter, nothing happened

### Steps to reproduce.

1. Go to `<esp-idf>/components/esp_driver_i2s/test_apps/i2s`
2. Run `idf.py build flash monitor`

### Build or installation Logs.

```plain
idf build flash monitor
Executing action: all (aliases: build)
Running cmake in directory /Users/tennis/esp/esp-idf-v5.3.1/components/esp_driver_i2s/test_apps/i2s/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/tennis/.espressif/python_env/idf5.3_py3.13_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/tennis/esp/esp-idf-v5.3.1/components/esp_driver_i2s/test_apps/i2s"...
-- IDF_TARGET is not set, guessed 'esp32s3' from sdkconfig '/Users/tennis/esp/esp-idf-v5.3.1/components/esp_driver_i2s/test_apps/i2s/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.39.5 (Apple Git-154)")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/tennis/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
[... 199 lines cut ...]
I (225) efuse_init: Chip rev:         v0.1
I (229) heap_init: Initializing. RAM available for dynamic allocation:
I (237) heap_init: At 3FC95D28 len 000539E8 (334 KiB): RAM
I (243) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (249) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (255) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (262) spi_flash: detected chip: gd
I (265) spi_flash: flash io: dio
W (269) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (283) sleep: Configure to isolate all GPIO pins in sleep state
I (290) sleep: Enable automatic switching of GPIO sleep configuration
I (297) main_task: Started on CPU0
I (307) main_task: Calling app_main()
  ___ ____  ____    _____         _   
 |_ _|___ \/ ___|  |_   _|__  ___| |_ 
  | |  __) \___ \    | |/ _ \/ __| __|
  | | / __/ ___) |   | |  __/\__ \ |_ 
 |___|_____|____/    |_|\___||___/\__|

Press ENTER to see the list of tests.
```

### More Information.

I'm running this to try and isolate an i2s issue of my own.  By running this, I hope to get a better understanding of how i2s works.

igrr, 2024-12-27:
If you are monitoring through a USB_SERIAL_JTAG port, please make sure that the primary console output channel is set accordingly. By default, USB_SERIAL_JTAG is configured as a secondary channel, meaning that it only provides output. Therefore input from USB_SERIAL_JTAG is not handled and your keystrokes are ignored.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/stdio.html#secondary-output

gamename, 2024-12-27:
> If you are monitoring through a USB_SERIAL_JTAG port, please make sure that the primary console output channel is set accordingly. By default, USB_SERIAL_JTAG is configured as a secondary channel, [...]

@igrr That was the problem. Thanks! Closing

closed 2024-12-27 by gamename