espressif/esp-idf #15111: How do I use the internal AES encrypt and decrypt functions in ESP code?

andy-danieal, 2024-12-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[wpa_supplicant/src/crypto/aes-cbc.c](https://github.com/espressif/esp-idf/blob/master/components/wpa_supplicant/src/crypto/aes-cbc.c)
**Functions**
aes_128_cbc_encrypt(const u8 *key, const u8 *iv, u8 *data, size_t data_len);
aes_128_cbc_decrypt(const u8 *key, const u8 *iv, u8 *data, size_t data_len);

how to used that function without 16-bytes blocks restrictions.

kapilkedawat, 2025-01-01:
Hi @andy-danieal, may I ask what's the usecase here? This is internal WiFi API and supposed to be used by internal functions only. If you are looking for a public implementation for applications, may be you should look at `mbedtls_aes_crypt_cbc()`.

andy-danieal, 2025-01-02:
Thank you for your response, @kapilkedawat . I'm looking for an example of how to use the internal AES encrypt and decrypt functions in ESP code. Specifically, I'd like to encrypt and decrypt a uint8_t array of bytes. Do you have an example that I could use?

kapilkedawat, 2025-01-02:
you can refer this sample code for encryption. https://github.com/espressif/mbedtls/blob/mbedtls-3.6.2-idf/programs/test/benchmark.c#L720

closed 2025-03-28 by Alvin1Zhang