espressif/esp-idf #15112: wifi_prov_mgr: unable to provision ESP32-S3 devive in SoftAP mode  using the ESP SoftAP Provisioning app on Androiod

bogdanm, 2024-12-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The ESP SoftAP Provisioning app  on Android (https://play.google.com/store/apps/details?id=com.espressif.provsoftap&hl=en) is unable to provide the WiFi credentials to an ESP32-S3 device running the `wifi_prov_mgr` sample using SoftAP as the provisioning method.

What I've done:

- Changed the provisioning method in the sample configuration (`idf.pt menuconfig`) from BLE to SoftAP.
- Compile the `wifi_prov_mgr` sample for ESP32-S3 (`idf.py set-target esp32-s3` followed by `idf.py flash monitor`).
- Scanned the QR code in the terminal using the ESP SoftAP Provisioning app.

Expected result: the device is succesfully provisioned.

Actual result: after scanning the QR code displayed in the terminal, the ESP SoftAP Provisioning app asks "Connect to device?" (and it identifies the device correctly). After pressing "Yes", the app display the following error:

```
Unable to connect with device.
Do you want to connect the device manually?
```

Answering "Yes" to the question above shows that the phone is in fact connected to the device's AP. The provisioning process doesn't progress after this.

Device logs:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (27) boot: compile time Dec 30 2024 17:43:22
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00140000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=2b370h (177008) map
I (131) esp_image: segment 1: paddr=0003b398 vaddr=3fc96e00 size=04c80h ( 19584) load
I (136) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=a223ch (664124) map
I (256) esp_image: segment 3: paddr=000e2264 vaddr=3fc9ba80 size=002e8h (   744) load
I (257) esp_image: segment 4: paddr=000e2554 vaddr=40374000 size=12d80h ( 77184) load
I (287) boot: Loaded app from partition at offset 0x10000
I (287) boot: Disabling RNG early entropy source...
I (299) cpu_start: Multicore app
I (309) cpu_start: Pro cpu start user code
I (309) cpu_start: cpu freq: 160000000 Hz
I (309) app_init: Application information:
I (312) app_init: Project name:     wifi_prov_mgr
I (317) app_init: App version:      v5.3.1-dirty
I (322) app_init: Compile time:     Dec 30 2024 17:43:18
I (328) app_init: ELF file SHA256:  36fa0c091...
--- Warning: ELF file not found. You need to build & flash the project before running 'monitor', and the binary on the device must match the one in the build directory exactly. 
I (333) app_init: ESP-IDF:          v5.3.1-dirty
I (339) efuse_init: Min chip rev:     v0.0
I (344) efuse_init: Max chip rev:     v0.99 
I (348) efuse_init: Chip rev:         v0.2
I (353) heap_init: Initializing. RAM available for dynamic allocation:
I (361) heap_init: At 3FC9F978 len 00049D98 (295 KiB): RAM
I (367) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (373) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (379) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (386) spi_flash: detected chip: gd
I (389) spi_flash: flash io: dio
W (393) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (407) sleep: Configure to isolate all GPIO pins in sleep state
I (413) sleep: Enable automatic switching of GPIO sleep configuration
I (421) coexist: coex firmware version: 4482466
I (435) coexist: coexist rom version e7ae62f
I (436) main_task: Started on CPU0
I (446) main_task: Calling app_main()
I (486) pp: pp rom version: e7ae62f
I (486) net80211: net80211 rom version: e7ae62f
I (496) wifi:wifi driver task: 3fca9ff4, prio:23, stack:6656, core=0
I (506) wifi:wifi firmware version: ccaebfa
I (506) wifi:wifi certification version: v7.0
I (506) wifi:config NVS flash: enabled
I (506) wifi:config nano formating: disabled
I (516) wifi:Init data frame dynamic rx buffer num: 32
I (516) wifi:Init static rx mgmt buffer num: 5
I (526) wifi:Init management short buffer num: 32
I (526) wifi:Init dynamic tx buffer num: 32
I (536) wifi:Init static tx FG buffer num: 2
I (536) wifi:Init static rx buffer size: 1600
I (536) wifi:Init static rx buffer num: 10
I (546) wifi:Init dynamic rx buffer num: 32
I (546) wifi_init: rx ba win: 6
I (556) wifi_init: accept mbox: 6
I (556) wifi_init: tcpip mbox: 32
I (556) wifi_init: udp mbox: 6
I (566) wifi_init: tcp mbox: 6
I (566) wifi_init: tcp tx win: 5760
I (566) wifi_init: tcp rx win: 5760
I (576) wifi_init: tcp mss: 1440
I (576) wifi_init: WiFi IRAM OP enabled
I (586) wifi_init: WiFi RX IRAM OP enabled
I (606) app: Starting provisioning
I (606) app: Development mode: using hard coded salt
I (606) app: Development mode: using hard coded verifier
I (626) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (666) wifi:mode : sta (84:fc:e6:6d:c1:34)
I (666) wifi:enable tsf
I (676) wifi:mode : sta (84:fc:e6:6d:c1:34) + softAP (84:fc:e6:6d:c1:35)
I (676) wifi:Total power save buffer number: 16
I (676) wifi:Init max length of beacon: 752/752
I (676) wifi:Init max length of beacon: 752/752
I (686) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (686) wifi:Total power save buffer number: 16
I (696) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (706) wifi_prov_mgr: Provisioning started with service name : PROV_6DC134 
I (706) app: Provisioning started
I (716) app: Scan this QR code from the provisioning application for Provisioning.
I (726) QRCODE: Encoding below text with ECC LVL 0 & QR Code Version 10
I (726) QRCODE: {"ver":"v1","name":"PROV_6DC134","username":"wifiprov","pop":"abcd1234","transport":"softap"}
                                          
  █▀▀▀▀▀█ ▀▀▀█▄ ▀ ▄█▄▄██  ▄▄▀ ▀ █▀▀▀▀▀█   
  █ ███ █  ▀▄█ █▀ ▀▄▄  █▀ ▀█▄ █ █ ███ █   
  █ ▀▀▀ █  ▄▀█▀▄█ ██ ▀█▄ ▄ ▄▀ ▄ █ ▀▀▀ █   
  ▀▀▀▀▀▀▀ █▄▀ █▄█▄▀ █▄█ █ █▄█ █ ▀▀▀▀▀▀▀   
  █▀ ▀█▄▀ ▄█▄ ▀ ▀██▀███  █▀ ▀█▄▄▀ ▄ ▄ ▀   
   ▄█   ▀ █ ▀ ▀  █▄▄█ █ █▀▀ ▄███   ▀██▀   
  ▄▄▀ ▀▀▀█▄▄█ ▄▄▄▄ ▄█▀▀ ▀█▀▄ █ ▀▀█▄ ▄█▀   
   ▄▀█▄▀▀▄▀ ▄ ▄▀ ██ ▄██ █ ▄  ▀▄ ▀▄▄▀▄██   
  ▄███▄▄▀▄▄▄ █▀▀▄█▄▄█▄█▄▄ ▀▀ ██ █▄ ▀█▄▀   
  ▄██▀ █▀▄▄▀▀▄▀██▀█ █▀█▀ █ █ █ ▀█ ▀█  ▀   
   ▀█ ▀▄▀ █▄ ▀▄ ▄█ ▀ █  █▄  ▀▀▀████▄▄█▀   
  ▄▀██ ▀▀▀▀█▄█▄▀▀█▄█▄   ▄▀█▀▀▀ █▀▀▀▀█▄▀   
  ▄▀  ▀█▀▀▀▄ ▀▀  █▄  ▀▀ ▄██ ▄▀█ █▀▄  ▀▀   
  █▀█▄▄█▀▀▄▄██▀▀▀█ ▀██▄▀█▄█ ██▀█ █▀ █▄▀   
  ▀  ▀▀▀▀ ▄  ▄▄▄▀▄ ▄▄▄▀▄█ ▄▀ ██▀▀▀█▀█▀    
  █▀▀▀▀▀█  ▄ ▀▄ ▄▀▀▀▀▀█ ▄███▀ █ ▀ █▀ ▄█   
  █ ███ █ ███ ▀▀██ █▄▄█  ▄█▄██▀██▀▀█ █▄   
  █ ▀▀▀ █ ▄█  ▀ ███▄▄▄▀▀▄█  ▄▄█ █▀█▀▄▄█   
  ▀▀▀▀▀▀▀ ▀▀   ▀▀▀   ▀▀ ▀▀▀   ▀▀ ▀▀▀  ▀   
                                          

I (986) app: If QR code is not visible, copy paste the below URL in a browser.
https://espressif.github.io/esp-jumpstart/qrcode.html?data={"ver":"v1","name":"PROV_6DC134","username":"wifiprov","pop":"abcd1234","transport":"softap"}
I (8146) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<0,0>, prof:1, snd_ch_cfg:0x0
I (8156) wifi:station: 82:dc:ba:83:85:f5 join, AID=1, bgn, 20
I (8156) app: SoftAP transport: Connected!
I (8306) wifi:<ba-add>idx:2 (ifx:1, 82:dc:ba:83:85:f5), tid:0, ssn:0, winSize:64
I (8356) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
```

Note that this works in the default configuration that uses provisioning via BLE mode. However, that's not useful for me for various reasons.

ESP-IDF version: 5.3.1
ESP SoftAP Provisioning app version:
Android version: 2.2.1-8a5e434
Phone: Samsunt S20FE 5G
Hardware: [MakerFabs MaTouch ESP32-S3 Parallel IPS with Touch 2.8"](https://www.makerfabs.com/matouch-esp32-s3-parallel-ips-with-touch-2-8.html)

hrushikesh430, 2024-12-31:
Hi @bogdanm, I tried to reproduce your issue, but unable to reproduce it locally. All configurations and version that you mentioned I used, but still works fine on my side. May be you can check from your side if anything is going wrong or just send patch file, which will help me to reproduce.

bogdanm, 2024-12-31:
> Hi @bogdanm, I tried to reproduce your issue, but unable to reproduce it locally. All configurations and version that you mentioned I used, but still works fine on my side. May be you can check from your side if anything is going wrong or just send patch file, which will help me to reproduce.

Thanks for the quick reply, it was a hint that something could be wrong on the phone side, not on the device side. After a bit of testing, it turns out that the SoftAP provisioning app fails if the Tailscale client that is running on my phone is connected to my tailnet. The provisioning app doesn't crash, it just doesn't complete the provisioning process, as described in my original post. With Tailscale disconnected, everything works as expected. Quite a surprising result (at least for me), but after trying a few more times it appears that this is indeed the issue.

closed 2024-12-31 by bogdanm