espressif/esp-idf #15114: I2C Failure After Launching HTTP Client Task

kevinkfci, 2024-12-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Everything operates as designed and expected until a command is issued to activate the HTTP client task and download a file from the AWS cloud. I encountered a similar issue when testing OTA examples with the HTTP client.
Once the HTTP client is launched, it begins to interfere with the I2C portion of the code, producing the following error. Even after stopping the HTTP client task, the error messages persist on the console port and do not stop.
At this stage, I am confident that the HTTP client is the source of the problem. I have divided the code into smaller sections, integrated them into my project, and tested each part. The issue consistently occurs as soon as the HTTP client is called. I am not sure what I have missed or if there is any HTTP client patch I have overlooked. I have tried different task approaches, such as changing task priority, but none of them have helped.

I am using ESP-IDF version 4.4.5 and cannot use the latest code due to compatibility constraints. 
please help!!!

I have attached the following files for reference:
[bootuplog.txt](https://github.com/user-attachments/files/18277907/bootuplog.txt)
[sdkconfig.txt](https://github.com/user-attachments/files/18277909/sdkconfig.txt)
[errormessage.txt](https://github.com/user-attachments/files/18277908/errormessage.txt)
[http_mp3_pwm.zip](https://github.com/user-attachments/files/18277921/http_mp3_pwm.zip)

mythbuster5, 2024-12-31:
The error message is `i2c driver not installed`?

kevinkfci, 2024-12-31:
I suspect the HTTP client is overwriting a memory section used by the I2C driver. I have several tasks utilizing I2C, and they work fine as long as the HTTP client-related task is not running. However, when I trigger the HTTP client task, error messages begin appearing, regardless of whether a file is being downloaded from AWS Cloud or not.

mythbuster5, 2024-12-31:
I see your code but seems not very helpful. Maybe a minimum reproducable project (including app_main) interface would be helpful for debugging. Perhaps it's not an I2C issue seems.

kevinkfci, 2024-12-31:
I try to put something together based on a dev-board. currently, it is ruing on our board.

nileshkale123, 2025-01-03:
Hello @kevinkfci ,

Is this issue still occurring? If not, could you please share the steps you took to resolve it?

If the issue persists, feel free to reach out with a concise, reproducible example so we can assist you further in debugging.

kevinkfci, 2025-01-03:
yes, it is still there, I have not found any solution yet.

nileshkale123, 2025-01-21:
Hello @kevinkfci,
Please let us know the steps or minimal example to reproduce and debug the issue.
Thanks

nileshkale123, 2025-02-18:
Hello @kevinkfci ,

I am closing this issue as no activity for a month. Please feel free to reopen if required and provide above asked details for further debugging.

closed 2025-02-18 by nileshkale123