espressif/esp-idf #15116: I2C Master Write and Delayed Read

paalcald, 2024-12-31:
### Is your feature request related to a problem?

I have this recurring issue when using the new I2C API where many sensors require a time  to take the measurement in between the writing of the measurement command and the lecture of the measurement.

### Describe the solution you'd like.

Id like for the API to have an option to either introduce custom delays inside the hardware command chain for `i2c_operation_t` such as 
```c
i2c_operation_t i2c_ops[] = {
        {.hw_cmd = I2C_TRANS_START_COMMAND},
        {.hw_cmd = I2C_TRANS_WRITE_COMMAND(i2c_dev->ack_check_disable ? false : true), .data = (uint8_t *)write_buffer, .total_bytes = write_size},
        {.hw_cmd = I2C_TRANS_DELAY(delay_duration)}, 
        {.hw_cmd = I2C_TRANS_START_COMMAND},
        {.hw_cmd = I2C_TRANS_READ_COMMAND(ACK_VAL), .data = read_buffer, .total_bytes = read_size - 1},
        {.hw_cmd = I2C_TRANS_READ_COMMAND(NACK_VAL), .data = (read_buffer + read_size - 1), .total_bytes = 1},
        {.hw_cmd = I2C_TRANS_STOP_COMMAND},
    };
```
or have an `i2c_master_transmit_delay_receive` that allows me to wait a bit before making the read.

### Describe alternatives you've considered.

I have attempted to implement it myself to no avail, this is what I've attempted:

If I attempt to perform the read and write using first `i2c_master_transmit` sleep the task for a bit and then `i2c_master_receive`.
Nonetheless in `i2c_master_transmit`:
```c
i2c_operation_t i2c_ops[] = {
        {.hw_cmd = I2C_TRANS_START_COMMAND},
        {.hw_cmd = I2C_TRANS_WRITE_COMMAND(i2c_dev->ack_check_disable ? false : true), .data = (uint8_t *)write_buffer, .total_bytes = write_size},
        {.hw_cmd = I2C_TRANS_STOP_COMMAND},
    };
```
That last `TRANS_STOP_COMMAND` makes this approach not adequate, since it should not be there, therefore I attempted to just remove that line and implement a custom function `i2c_master_transmit_no_stop`

```c
esp_err_t i2c_master_transmit_no_stop(i2c_master_dev_handle_t i2c_dev, const uint8_t *write_buffer, size_t write_size, int xfer_timeout_ms)
{
    ESP_RETURN_ON_FALSE(i2c_dev != NULL, ESP_ERR_INVALID_ARG, TAG, "i2c handle not initialized");
    ESP_RETURN_ON_FALSE((write_buffer != NULL) && (write_size > 0), ESP_ERR_INVALID_ARG, TAG, "i2c transmit buffer or size invalid");

    i2c_operation_t i2c_ops[] = {
        {.hw_cmd = I2C_TRANS_START_COMMAND},
        {.hw_cmd = I2C_TRANS_WRITE_COMMAND(i2c_dev->ack_check_disable ? false : true), .data = (uint8_t *)write_buffer, .total_bytes = write_size},
    };

    if (i2c_dev->master_bus->async_trans == false) {
        ESP_RETURN_ON_ERROR(s_i2c_synchronous_transaction(i2c_dev, i2c_ops, DIM(i2c_ops), xfer_timeout_ms), TAG, "I2C transaction failed");
    } else {
        ESP_RETURN_ON_ERROR(s_i2c_asynchronous_transaction(i2c_dev, i2c_ops, DIM(i2c_ops), xfer_timeout_ms), TAG, "I2C transaction failed");
    }
    i2c_dev->master_bus->trans_done = true;
    xSemaphoreGive(i2c_dev->master_bus->bus_lock_mux);
    return ESP_OK;
}
```
This approach somehow encounters an unexpected NACK

```
E (1363) i2c.master: I2C transaction unexpected nack detected
E (1363) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
I (1363) gpio: GPIO[21]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (1373) gpio: GPIO[22]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
E (1383) i2c.master: i2c_master_transmit_no_stop(1073): I2C transaction failed
```

I've attempted to take a look at the inner workings of `static void s_i2c_send_commands(i2c_master_bus_handle_t i2c_master, TickType_t ticks_to_wait)` to try and trace where this error is comming from and from this if-else
```c
        if (i2c_operation->hw_cmd.op_code == I2C_LL_CMD_WRITE) {
            s_i2c_write_command(i2c_master, i2c_operation, &fifo_fill, &address_fill, NULL);
        } else if (i2c_operation->hw_cmd.op_code == I2C_LL_CMD_READ) {
            s_i2c_read_command(i2c_master, i2c_operation, &fifo_fill, NULL);
        } else {
            s_i2c_start_end_command(i2c_master, i2c_operation, &address_fill, NULL);
        }
```
it seems to me that any command chain has to be stop terminated from the logic inside `s_i2c_start_end_command(...)`, the only thing that occurs to me now is to add a new `op_code` to handle the delay and add logic inside `s_i2c_send_commands` to handle my new op_code but that seems to me like changing a lot of the esp idf source code.

### Additional context.

The sensor I'm using sends back an `ACK` as soon as it receives the lecture cmd but registers contents are undefined for the following `12ms` after those, you can retrieve the value from the register.

This is my first Issue ever, sorry if I did not conform to any guidelines, I tried.

mythbuster5, 2025-01-06:
Sorry, you said you use `new I2C API where many sensors require a time to take the measurement in between the writing of the measurement command and the lecture of the measurement.` May I know how do you do this in legacy driver?

mythbuster5, 2025-06-03:
Seems there is no more update, and we can't do this in both software and hardware level. This issue is going to be closed. Many thanks.

closed 2025-06-03 by mythbuster5