espressif/esp-idf #15118: windows11 24H2编译工程只使用到CPU四个核心

Ahypnis, 2025-01-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.1.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3

### Power Supply used.

External 5V

### What is the expected behavior?

我的CPU是AMD的3945HX，预期是半分钟能够编译完成，但是今早因为系统的部分bug升级了系统，电脑已经打开性能模式，晚上在编译工程的时候发现编译的时候只调度起来几个核心，其他核心都在摸鱼。

### What is the actual behavior?

我给几个人在安装ESPIDF环境的时候尝试安装过5.1.2 和5.3.1   发现编译速度都一样很慢，他们的系统是也是更新了新的补丁，发现编译一个helloworld都要4-5分钟，后台没有杀毒软件，也尝试把idf的SDK添加进win11的排除项，还是不能够解决问题，还有尝试过在注册表强行把大小核全开，在编译的时候打开任务管理器看核心的工作状况，即使全部都在工作，编译也是很慢。不是一个正常的速度。

### Steps to reproduce.

每次编译

dobairoland, 2025-01-03:
I also updated to Win 11 24H2 but I don't experience this issue. I'm building under 2 minutes without ccache after fullclean-ing the project, and under 1 minute with ccache after fullclean. I didn't have to do anything after the update and I'm running on a 4 years old laptop.

- It is possible that there is an issue with Win 11 24H2 and there is going to be fix soon.
- You can try to re-run the Windows tool installer. It does several performance related tune-up. Maybe one of them was reseted by the Windows update.
- You can check whether you use ccache. See: https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/tools/idf-py.html#id6

Ahypnis, 2025-01-03:
> 我还更新到 Win 11 24H2，但我没有遇到这个问题。在完全清理项目后，我在没有 ccache 的情况下构建了不到 2 分钟，在完全清理后使用 ccache 构建了不到 1 分钟。更新后我什么都没做，而且我在一台 4 年前的笔记本电脑上运行。
> 
> * Win 11 24H2 可能存在问题，很快就会得到修复。
> * 您可以尝试重新运行 Windows 工具安装程序。它会进行一些与性能相关的调整。也许其中一个被 Windows 更新重置了。
> * 您可以检查是否使用了 ccache。请参见： https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/tools/idf-py.html#id6

I solved this problem tonight. The reason is that the Microsoft PC Manager in Windows has a service called MSPCManagerService.exe. When this process runs in the background, it will severely limit the compilation speed of ESPIDF.

Ahypnis, 2025-01-03:
> I also updated to Win 11 24H2 but I don't experience this issue. I'm building under 2 minutes without ccache after fullclean-ing the project, and under 1 minute with ccache after fullclean. I didn't [...]

I solved this problem tonight. The reason is that the Microsoft PC Manager in Windows has a service called MSPCManagerService.exe. When this process runs in the background, it will severely limit the compilation speed of ESPIDF.

closed 2025-01-03 by dobairoland