espressif/esp-idf #15121: LVGL display with unexpected pixels

emesamwong, 2025-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My project is a Matter Project with LCM Display. I am using LVGL as the display component. I found that the display contains some unexpected pixels or is missing some pixels of the Object.
Here are some display examples:

The unexpected pixels:
[image]
Normal Display:
[image]

The missing pixels example:
[image]

When Pressed the button:
[image]
Note: the Green Line is created by lv_obj_set_style_border_width function with 1 pixel

Since Matter requires a lot of resources, I set the LVGL display buffer using PSRAM. 
Here is the SDK config of LVGL and PSRAM:
[image]

LVGL Init code:
[image]
buf1 size is: 240 * 240 * sizeof(lv_color_t)

**Any idea about the missing pixels and unexpected pixels? Please let me know if more information is needed.**

Details:
MCU: ESP32-S3-WROOM-1 (MCN16R8)
Development OS: Linux
Development Environment: VS Code
ESP-IDF version: Tag v5.2.2 (commit hash #3b8741b)
ESP-Matter: release/v1.3 (commit hash #b5a23374)
connectedhomeip: 	 commit hash #82748b9f
LVGL version: v8.3.11 (ESP Component Registry)
LCM: ST7789P3 (240*240 pixels)
Touch: CST816D

Kainarx, 2025-01-02:
Are you using the spi lcd driver? Can you check the raw data of the lvgl draw buffer? Or try to use the lvgl emulator to see the phenomena?

emesamwong, 2025-01-02:
For the LCD Display Driver part, I am using the i80_controller example. Here is the driver code
[image]
[image]

Kainarx, 2025-01-02:
I guess that the draw buffer itself is missing (or has extra) these pixels, could you please check the raw data?

emesamwong, 2025-01-02:
@Kainarx Thank you for your support. I am confused about the raw data checking part. 

[image]

I created 2 objects, The background object is 240*240 with a black color.  The parent object is 229*240 with a black color based on the background. The Green color (R: 0x00, G: 0xFF, B:0x00) outline is based on the parent. 

I added the Debug Log before executing "esp_lcd_panel_draw_bitmap"
[image]

Here is the result:
[image]

According to the Debug log, there seems mismatch in the display, I would expect the Green color value to be 0xFF and the x1 to be 31. Is that my Debug log incorrect? How could I print all the content of the draw_buff in console?

Kainarx, 2025-01-03:
Can't see what is printed after `offsety2` on your picture. I have another suggestion, use the [Simulator]( https://docs.lvgl.io/master/details/integration/ide/pc-simulator.html) to see the phenomenon, it will be easier to determine if the problem is with lvgl or the lcd driver

emesamwong, 2025-01-03:
The full Log statement:
[image]
 
For Simulator:
> I created 2 objects, The background object is 240240 with a black color. The parent object is 229240 with a black color based on the background. The Green color (R: 0x00, G: 0xFF, B:0x00) outline is based on the parent.

I used EEZ studio with LVGL 8.x version to create two Objects, which do not contain the missing pixels. Since My project contains Matter and I set the LVGL buffer to PSRAM, would those be reasons why the expected display occurred? 

Back to the LVGL buffer or flushing issue, how could I check the raw buffer data? should I also check the data on the ST7789P3 side?

suda-morris, 2025-01-03:
The symptom is likely related to the PSRAM cache. But I'm not sure how you allocate your LVGL draw buffer. 

BTW, Could you try [this example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/i80_controller) on the **master** branch, the example already support ST7789 and also PSRAM draw buffer. @emesamwong

emesamwong, 2025-01-03:
@suda-morris Here is the setting of the LVGL draw buffer in PSRAM
[image]
[image]
[image]
[image]

For the PSRAM consideration: 
I tried to move the draw buffer from PSRAM to DRAM, and I do not found Missing pixels, or unexpected pixels for now. Of course, the Matter part is not working since not enough DRAM for provisioning.

I will merge the i80_control example in master with my project one.

emesamwong, 2025-01-07:
I reviewed the i80_control example in master, and the PSRAM / draw buffer setup is similar to my current project. 
In addition, I enabled "CONFIG_LV_MEM_CUSTOM" in sdkconfig which could free a lot of D/IRAM and enough to move the LVGL buffer from PSRAM to Internal RAM. 

The LVGL component is downloaded in the esp component registry and the version is v8.3.11

Any idea about the "CONFIG_LV_MEM_CUSTOM" setting? The description could not provide enough information.

Kainarx, 2025-01-07:
If `CONFIG_LV_MEM_CUSTOM` is not set. The default memory manager of LVGL (lv_mem) pre-allocates a fixed-size memory pool (configured via LV_MEM_SIZE). This memory pool is used for all dynamic memory allocations in LVGL (such as objects, styles, animations, etc.). Even if you do not use the entire memory pool, this memory will still be reserved, leading to SRAM waste.

emesamwong, 2025-01-07:
@Kainarx If the CONFIG_LV_MEM_CUSTOM is enabled, could I claim that LVGL allocates a size memory based on the current Objects/Styles, etc?
For example, I created 3 Objects in PageOne, and then LVGL always allocates a size memory based on those Objects instead of pre-allocated

Kainarx, 2025-01-07:
Yes, more infomation about lvgl memory allocation can be found in the specific implementation of `lv_mem_init`

closed 2025-03-10 by Alvin1Zhang