espressif/esp-idf #15123: Port related issue

Manasabalakrishna, 2025-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[image]
port is used by otjer task 
ninja failed with exit code 1, output of the command is in the C:\Users\ManasaJB\hello_world\build\log\idf_py_stderr_output_22332 and C:\Users\Manasa

1.if i stop the terminal and again rerun the code its showing termianl closed 
2.if i run the code 2 times again out put showing port is not availbke
3.each time i got stuck like this even if i remove port and close ide it doesnot works  i need to compleetely turn off my pc then it willl work and again after 2 times it will gte stuck

steps took to resolve downgarde trhe version and confirm that port is not being used in any other task .

KonstantinKondrashov, 2025-01-03:
Hi @Manasabalakrishna!
It seems the COM port is not being released properly when you close the terminal. Always properly close the terminal or stop the idf monitor before closing the terminal or IDE.
- Before closing the terminal, ensure you stop the connection (e.g., Ctrl+] in idf.py monitor).
- If stuck, open Task Manager (Ctrl+Shift+Esc), find any process using the port (e.g., Python or terminal emulator), and end it.
- If the port still doesn’t work, unplug and replug the device.

If does not help please provide more information on what IDE you use, what esp-idf version, OS, etc.

Manasabalakrishna, 2025-01-03:
Thanks @KonstantinKondrashov 
****Each time when port got stuck** **
open Task Manager > Details> see for Python.exe 
end the python .exe task ,and this is working fine for me.

peterdragun, 2025-01-07:
Hi @Manasabalakrishna, can you please confirm that this does not happen if you close the IDF Monitor properly before closing the terminal (with `CTRL+]`), as Konstantin suggested in the previous reply?

Manasabalakrishna, 2025-01-07:
Hello, The issue was there even if when I give ctrl+] .Only when I end
python.exe it works.

dobairoland, 2025-01-20:
I'm closing this for inactivity. We can reopen it if there is more information to work with.

closed 2025-01-20 by dobairoland