espressif/esp-idf #15124: No free Interrupt Inputs for Matter OTA

emesamwong, 2025-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When trying to execute the Matter OTA test, the DUT would reset due to **'No free interrupt inputs for RSA interrupt (flags 0xE)"** 
I found the Error Log message location and printed the list of interrupts. When the error occurred, there were still 9 interrupts available. 

Here is the Debug Log:
[image]
[image]

What is the "interrupt input" representing? 

Any Solution for the unstable interrupt resource?

Manually using Interrupt:
UART*2 with Shared
I2C with Shared 
I2C with ESP_INTR_FLAG_LOWMED
GPIO
Timer0 
LEDC & LCD_CAM (seem installed by the esp _lcd driver, i80_Control example)

Environment: 
MCU: ESP32-S3-WROOM-1 (MCN16R8)
Development OS: Linux
Development Environment: VS Code
ESP-IDF version: Tag v5.2.2 (commit hash #3b8741b)
ESP-Matter: release/v1.3 (commit hash #b5a23374)
connectedhomeip: commit hash #82748b9f
LVGL version: v8.3.11 (ESP Component Registry)
LCM: ST7789P3 (240*240 pixels)
Touch: CST816D

chshu, 2025-01-03:
@emesamwong The interrupt inputs on Core 0 are indeed fully utilized. When a task on Core 0 attempts to initialize the RSA driver, it encounters this issue.

You can refer to: [troubleshooting-interrupt-allocation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/intr_alloc.html#troubleshooting-interrupt-allocation) for possible solutions.

A quick way to try out: use `xTaskCreatePinnedToCore()` to pin some tasks to Core1.

emesamwong, 2025-01-03:
@chshu Thank you for your suggestion.
I will try to free more Core 0 resources by using the xTaskCreatePinnedToCore() function. 
For the RSA driver, should it be installed by the Matter?

esp_intr_dump() function only list out the interrupt instead of interrupt inputs ([troubleshooting-interrupt-allocation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/intr_alloc.html#troubleshooting-interrupt-allocation))
How could I check the interrupt inputs resources?

chshu, 2025-01-03:
Matter itself doesn't use RSA, please check if any other tasks are using RSA.

> esp_intr_dump() function only list out the interrupt instead of interrupt inputs [...]

I'm not entirely sure I understand the question. The `esp_intr_dump()` function lists the interrupts (or interrupt inputs), 32x available per core. However, there are more interrupt sources than available interrupts, so they should be allocated properly in the application.

emesamwong, 2025-01-03:
I would try to find the task that installed the RSA interrupt.

I am confused about the Error Message "No **free interrupt inputs** for "X" interrupt" and there are free interrupts in the interrupt list. It means the interrupt list could not represent interrupt inputs. Hence, I would like to keep checking the interrupt sources (instead of the interrupt list) usage which leads to the Error Message occurring. 

For this statement, Any configuration should I set to enable it?
> However, there are more interrupt sources than available interrupts, so they should be allocated properly in the application.

chshu, 2025-01-03:
As mentioned in my initial comment: https://github.com/espressif/esp-idf/issues/15124#issuecomment-2568629115, there are no free interrupts on Core 0, which caused the failure to allocate an additional interrupt on Core 0. Therefore, we recommended switching some tasks to Core 1.

Interrupts are allocated on the core where the task is running.

Refer to a similar issue: https://github.com/espressif/esp-idf/issues/15124#issuecomment-2568629115

emesamwong, 2025-01-03:
@chshu Thank you for your support. I disabled the CONFIG_MBEDTLS_HARDWARE_MPI config, and tested 20 times OTA, no more Errors occurred.

closed 2025-02-26 by chshu