espressif/esp-idf #15126: DMA buffer size for I2S_TDM seems to be wrong

h-milz, 2025-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-1050-gb5ac4fbdf9

### Espressif SoC revision.

ESP32-C6FH4 (QFN32) (revision v0.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

DFRobot FireBeetle 2 ESP32-C6

### Power Supply used.

USB

### What is the expected behavior?

I am trying to set up TDM256 with currently 4 slots, data bit width 32 bit = 4 byte, sample rate 44100 Hz. I want to capture 60 samples (overall 960 bytes) before further processing and have 6 DMA buffers of 960 bytes each. The word clock GPIO shows a steady 44100 Hz clock on the oscilloscope, so that appears correct. (Further processing involves fetching each 60 sample frame and send them over UDP one by one.) 

I tried multiple different ways to set up a proper timing:  

1. use `i2s_channel_read()` in blocking mode in an endless loop, waiting for 960 bytes to appear. Theoretically, this should happen every 1.36 ms = 60/44100 s. 

3. use the word clock and trigger a hardware interrupt using a second pin, then count up to 60 in the ISR (which takes 60/44100 seconds = 1.36 ms) and set a flag when this is the case, or notify the processing task using `xTaskNotifyFromISR()`. The processing task waits for the flag (or the notification) to appear and then reads the accumulated data using `i2s_channel_read()`. Whatever I do, the call takes 4 times the 1.36 ms, although I would expect the call itself to take not more than a couple of microseconds because all it pretty much does is `memcpy()` data from the DMA buffer to my destination buffer. 

4. use the PCNT peripheral to perform the counting, and then I either set the flag as in no. 2 or notify the processing task using `xTaskNotifyFromISR().` This works fine, but the subsequent `i2s_channel_read `again takes ~ 4* 1.36 ms. 

5. use an `on_recv` callback routine that gets triggered (should get triggered) each time one of the DMA buffers is full. I notify a new processing task which takes the DMA buffer pointer and the data size. I can see that the 6 DMA buffers are used in a ring buffer like fashion, and the data size is always 960 bytes. The strange thing is that the `on_recv` callback gets called only every ~4 * 1.36 ms, i.e. 4 times slower than expected. 

The factor 4 that shows up each time indicates that I either have a very strange thinking error, there is a bug in the ESP-IDF (which I do not have the means to troubleshoot), or that there is actually a connection between the callback called only every 4 word clocks and the `i2s_channel_read()` call taking as much time as 4 word clocks. But why is that so, and why would the i2s examples in the ESP-IDF work as designed? Where is a potential difference? 

The configuration is as follows: 

```
#define I2S_NUM                 I2S_NUM_AUTO
#define NSAMPLES                60                      // the number of samples we want to send in a batch
#define NUM_SLOTS               4                       // TDM256, 8 slots per sample
#define SLOT_WIDTH              32                      // 
#define DMA_BUFFER_COUNT        12                       // Number of DMA buffers
#define DMA_BUFFER_SIZE         NSAMPLES * NUM_SLOTS * SLOT_WIDTH / 8  // Size of each DMA buffer
#define SAMPLE_RATE             44100                   // 44100

static i2s_chan_config_t chan_cfg = {        // I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_AUTO, I2S_ROLE_MASTER);
    .id = I2S_NUM_AUTO,
    .role = I2S_ROLE_MASTER,
    .dma_desc_num = DMA_BUFFER_COUNT,
    .dma_frame_num = NSAMPLES, 
    .auto_clear_after_cb = false, 
    .auto_clear_before_cb = false, 
    .allow_pd = false, 
    .intr_priority = 5, 
};       
static i2s_tdm_config_t rx_cfg = {
    .clk_cfg = {                                // I2S_STD_CLK_DEFAULT_CONFIG(SAMPLE_RATE),
        .sample_rate_hz = SAMPLE_RATE,
        .clk_src = I2S_CLK_SRC_DEFAULT,         // we will use I2S_CLK_SRC_EXTERNAL !
        .mclk_multiple = I2S_MCLK_MULTIPLE_512,  // Set MCLK multiple to 256
    },
    .slot_cfg = I2S_TDM_MSB_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_32BIT, I2S_SLOT_MODE_MONO,
                                                I2S_TDM_SLOT0 | I2S_TDM_SLOT1 | I2S_TDM_SLOT2 | I2S_TDM_SLOT3),
    .gpio_cfg = {
        .mclk = GPIO_NUM_3,
        .bclk = GPIO_NUM_4,
        .ws = GPIO_NUM_2,
        .dout = I2S_GPIO_UNUSED,
        .din = GPIO_NUM_5,
        .invert_flags = {
            .mclk_inv = false,
            .bclk_inv = false,
            .ws_inv = true,    // we use AK5538 mode 32/36 (start on rising WS edge)
        },
    },
};

```

I checked the source code in `i2s_common.c`, `i2s_tdm.c`and `i2s_private.h`. As it appears, the chain of calls is like this: 

- `i2s_channel_init_tdm_mode()` calls  `i2s_tdm_set_slot()`. 
- `i2s_tdm_set_slot()` calls `i2s_get_buf_size()` with the number of DMA buffers (number of frames in a buffer = 60) and the slot bit width (32).  The correct bufsize should be 60 * 4 samples / frame * 4 byte / sample = 960 byte.
- `i2s_alloc_dma_desc()` should allocate buffers of size 960, so that each frame fits exactly in one DMA buffer. 

### What is the actual behavior?

`i2s_get_buf_size()` seems to calculate 240 instead of 960 byte bufsize. This apparently leads to the `on_recv` callback not to be called every 1/44100 s, but only 4 times slower, making it impossible to retrieve and process the samples in real time. 

I consider this a bug because the actual behaviour is not in sync with the documentation. 

### Steps to reproduce.

Set the code up as above, then `idf.py app-flash monitor`. 

### Debug Logs.

```plain
D (288) i2s_common: rx channel is registered on I2S0 successfully
D (298) i2s_common: DMA malloc info: dma_desc_num = 6, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 240
D (308) i2s_tdm: Clock division info: [sclk] 160000000 Hz [mdiv] 7 [mclk] 22579200 Hz [bdiv] 4 [bclk] 5644800 Hz
D (318) i2s_common: MCLK is pinned to GPIO3 on I2S0
D (318) i2s_tdm: The rx channel on I2S0 has been initialized to TDM mode successfully
D (328) i2s_common: i2s rx channel enabled
```

h-milz, 2025-01-02:
I would like to re-specify. When I use `chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_AUTO, I2S_ROLE_MASTER);`, then I get 960 byte transfers but the callback interrupt rate is 4 times the 1.36 ms. The macro sets the number of buffers `dma_frame_num` to 240.  When I force the number of buffers to 60, the interrupt rate is correct but the buffers are only 240 bytes. I struggle to find in the i2s source code where this is set. The question is how I get 960 byte buffers (or later, 1920 byte buffers for 8 channels on C5) with the correct interrupt rate.

L-KAYA, 2025-01-03:
Hi @h-milz , thanks for reporting!

Have checked your configurations, it seems that you're using `I2S_SLOT_MODE_MONO` when enabled 4 slots, it means the 4 slots are all transmitting same data, so the DMA will only be a quarter, because there is no need to store the repeated data.

You may try to set `I2S_SLOT_MODE_STEREO` instead~

h-milz, 2025-01-03:
@L-KAYA  thank you for the heads up! I dug a bit more in the source code and added a ESP_LOGI here and there and came to the same conclusion a couple of minutes ago. It also works now for 8 channels, which gives me 1920 byte buffers every 1.36ms. 

In this case may I suggest to rephrase the documentation to state the issue more explicitly. Having to set `I2S_SLOT_MODE_STEREO` and `I2S_TDM_SLOT0 | ... | I2S_TDM_SLOT7` is a bit counterintuitive. To me that sounds like 8 stereo channels = 16 mono channels. So in the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c5/api-reference/peripherals/i2s.html#_CPPv425i2s_channel_init_tdm_mode17i2s_chan_handle_tPK16i2s_tdm_config_t)  you may want to add a phrase like,

Note: 
If you want to configure TDM mode for n slots and want them all processed, you need to set `I2S_SLOT_MODE_STEREO` and `I2S_TDM_SLOT0 | ... | I2S_TDM_SLOTn`.

L-KAYA, 2025-01-03:
@h-milz Thanks for you suggestion! From what I learnt `Stereo` has already means `multiple sound channels` (see [this article](https://www.lifewire.com/how-many-channels-3135062)). Two-channel stereo is actually 2.0/2.1 channel system, there are also 5.1/6.1/7.1 channel systems which are also called stereo.

Anyway, I'll add more information about the DMA buffer size as well as `STEREO`. Thank you!

h-milz, 2025-01-03:
@L-KAYA Thank you! Let me ask one more question: I'm thinking about using an external oscillator to avoid any possible clock jitter from the fractional divider. BCLK would be 11.2896 MHz at 44.1 kHz sample rate for TDM256 but the driver wants to have I2S_MCLK_MULTIPLE_384 for 4 channels and _768 for 8 channels. I assume this is to minimize the effects of the fractional divider when using the internal clock, which is good. But if I use an external 11.2896 MHz clock (which I just tried, and it works nicely for 2 channels), the driver also wants _384 for 4 channels and _768 for 8, although in this case the divisor will always be integer for 44.1 kHz sample rate. In this case, MCLK could actually be equal to BCLK, which is fine for the AK5538 ADC that I intend to use. Is there a way to avoid having to purchase a higher frequency oscillator, which would impose stricter constraints as far as PCB layout? 

and why is that in the first place? I'm not even using I2S_DATA_BIT_WIDTH_24BIT. I assume the driver code as soon as the `bclk_div` looks too small automatically requests a x3 factor even if this is technically not required when using an external clock. OK, I dug a bit more.  

The warning 

`W (308) i2s_tdm: the current mclk multiple is too small, adjust the mclk multiple to 768
`

is generated in i2s_tdm.c: i2s_tdm_calculate_clock(), which apparently does not check if `.clk_src = I2S_CLK_SRC_EXTERNAL` is set.  There is a comment saying 

`        /* While RECEIVING multiple slots, the data will go wrong if the bclk_div is equal or smaller than 2 */
`

The following additional code does the trick: 

```
        /* While RECEIVING multiple slots, the data will go wrong if the bclk_div is equal or smaller than 2 */
        if (clk_cfg->clk_src == I2S_CLK_SRC_EXTERNAL) {
            clk_info->mclk = clk_cfg->ext_clk_freq_hz;
            clk_info->bclk_div = clk_info->mclk / clk_info->bclk;
            ESP_LOGI(TAG, "mclk = %lu, bclk_div = %d", clk_info->mclk, clk_info->bclk_div );
        } else if (clk_info->bclk_div <= 2) {
            clk_info->bclk_div = 3;
            clk_info->mclk = clk_info->bclk * clk_info->bclk_div;
            ESP_LOGW(TAG, "the current mclk multiple is too small, adjust the mclk multiple to %"PRIu32, clk_info->mclk / rate);
        }

```

Or is this due to a hardware constraint? When clocking in the data bits, as long as you sample the bits with the rising edge of BCLK, everything should be fine, and MCLK plays no role at all. 

Should I create a PR? 

Interestingly, Roland use in their GR-55 guitar synthesizer (which I'm trying to interface to) also an external clock reference of 16.9344 MHz for their ADCs and DACs, which is 384 * 44.1 kHz. So they avoid any non-integer divisors as well. 

(Too bad the C6 and C5 don't have an APLL... )

h-milz, 2025-01-03:
(A similar trick could be done for slave mode.)

h-milz, 2025-01-05:
continued in  #15142.

closed 2025-01-05 by h-milz