espressif/esp-idf #15127: Unable to burn 2 HMAC keys in eFuse  back to back for DS peripheral

msivanesancq, 2025-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.1

### Espressif SoC revision.

ESP32-C3 QFN32(Rev v0.3)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-MINI-1

### Power Supply used.

USB

### What is the expected behavior?

I am generating 2 RSA keys and efuse must be burnt with HMAC for having DS peripheral support.
The 1st HMAC key is burnt successfully. But it is failing on 2nd one inside the function `ets_efuse_write_key()`
I am facing kernel panics and watchdog timer errors if I retry with the same board.
Finally after few tries system is failing with efuse slots are unavailable.
This error did not happened on esp-idf v 5.1.1 

### What is the actual behavior?

Both HMAC keys burning must be succesfull.

### Steps to reproduce.

Generate 1st and 2nd keys back to back and burn the efuse with `ets_efuse_write_key()`
I also tried to increase the timer CONFIG_SOC_DS_KEY_CHECK_MAX_WAIT_US=1100.
But did not help as I think this is a constant.

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (Stack protection fault).

Detected in task "main" at 0x403801de
Stack pointer: 0x3fc9c790
Stack bounds: 0x3fc9c190 - 0x3fc9c790

Core  0 register dump:
MEPC    : 0x40380262  RA      : 0x40380228  SP      : 0x3fc9c790  GP      : 0x3fc93800
TP      : 0x3fca3860  T0      : 0x600c2194  T1      : 0x600c2128  T2      : 0x00000002
S0/FP   : 0x00000037  S1      : 0x80000005  A0      : 0x00000000  A1      : 0x00000300
A2      : 0x600ce000  A3      : 0x4876f42f  A4      : 0x000001c8  A5      : 0x7b26c7f8
A6      : 0xe4000000  A7      : 0x72eea082  S2      : 0x00001881  S3      : 0x00000001
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x3fca2c28  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x4ffeeff6  T4      : 0xa79c519f  T5      : 0x0bec8ab4  T6      : 0x37f6ffea
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x0000001b  MTVAL   : 0x0000850a
MHARTID : 0x00000000

Stack memory:
3fc9c790: 0x60023000 0x40385d14 0x40385d20 0x3c0d9a80 0x00000003 0x3c0d0128 0x0000000b 0x00000000
3fc9c7b0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9c7d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
```

### More Information.

None

AdityaHPatwardhan, 2025-01-07:
Hi @msivanesancq Thanks for the issue.

1) Can you please share the test code with which you are able to reproduce the issue?

2) I would also like to know what was the eFuse burning status?  Were the eFuse keys burned in the eFuse in incomplete manner( Only few bits are burned) or nothing was written in the eFuse? Could you please share the eFuse summary as well? 

Thanks,
Aditya

AdityaHPatwardhan, 2025-01-17:
Hi @msivanesancq I think the issue is more related to the software settings such as stack size, other configurations for your case. I am closing the issue since it doesn't seem anything is wrong with the API.
Feel free to re-open if you are still facing the issue.
Thanks,
Aditya.

closed 2025-01-17 by AdityaHPatwardhan