espressif/esp-idf #15128: v5.1.5 - Bluetooth controller crash in btdm_sleep_check_duration

expresspotato, 2025-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.5

### Espressif SoC revision.

ESP32-Pico-V3-02

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

LyraT & Custom, same result

### Power Supply used.

USB

### What is the expected behavior?

The CPU doesn't crash, disconnect our customers, and hard reboot

### What is the actual behavior?

The CPU is unstable and crashes with:

```
ASSERT_PARAM(1 0), in ld_acl.c at line 6145
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400982b8: f01d092d 00004136 f01d0000
--- 0x400982b8: btdm_sleep_check_duration at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/controller/esp32/bt.c:934
```

### Steps to reproduce.

We're running BLE GATTS along side BT Classic for HFP and A2DP, but in this instance classic wasn't connected yet... 

After connection, some data exchange takes place via indication and the bluetooth controller crashes.

### Debug Logs.

```plain
ASSERT_PARAM(1 0), in ld_acl.c at line 6145
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400982b8: f01d092d 00004136 f01d0000
--- 0x400982b8: btdm_sleep_check_duration at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/controller/esp32/bt.c:934

Core  0 register dump:
PC      : 0x400982bf  PS      : 0x00060034  A0      : 0x800909a6  A1      : 0x3ffe13f0  
--- 0x400982bf: r_assert at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/controller/esp32/bt.c:1944

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ff9cd42  
A6      : 0x00001801  A7      : 0xfffffffb  A8      : 0x8000814b  A9      : 0x3ffe1360  
A10     : 0x00000000  A11     : 0x3ffe1384  A12     : 0x3ffe132f  A13     : 0x00000035  
A14     : 0x00000000  A15     : 0x3ffe1335  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffffe  
--- 0x400012c5: strcmp in ROM
0x400012d5: strcmp in ROM

Backtrace: 0x400982bc:0x3ffe13f0 0x400909a3:0x3ffe1410 0x40036c5f:0x3ffe1430 0x4023b336:0x3ffe1460 0x4023a61e:0x3ffe1480 0x400918d6:0x3ffe14a0 0x40019d11:0x3ffe14d0 0x40055b4d:0x3ffe14f0 0x402338ff:0x3ffe1510 0x40233f61:0x3ffe1530 0x4009e7bd:0x3ffe1560
--- 0x400982bc: r_assert at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/controller/esp32/bt.c:1943
0x400909a3: r_assert_param at ??:?
0x40036c5f: r_ld_acl_afh_set in ROM
0x4023b336: lmp_set_afh_handler at ??:?
0x4023a61e: lc_lmp_rx_handler at ??:?
0x400918d6: ke_task_schedule at ke_task.c:?
0x40019d11: r_ke_event_schedule in ROM
0x40055b4d: r_rwip_schedule in ROM
0x402338ff: r_rw_schedule at ??:?
0x40233f61: btdm_controller_task at ??:?
0x4009e7bd: vPortTaskWrapper at /Users/kevin/esp/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### More Information.

EXCVADDR of 00000000 is a NULL so a null pointer exception...

Is *slot_cnt NULL?

ESP-YTGerd, 2025-01-06:
Hi, @expresspotato,
Thanks for your report. Could you please apply the following patch and test again? The crash issue should be resolved and  let us to see if any other issues arise. If there are no further problems, we will merge this fix into the release branch.
[patch.zip](https://github.com/user-attachments/files/18317241/patch.zip)

expresspotato, 2025-01-07:
Hey @ESP-YTGerd, I've applied the patch and will report back regarding if the bug re-occurs within a few days time.

Honestly a more pressing issue is bug #13605 as it happens constantly across all our devices and has a reproducible example attached in the ticket.

ESP-YTGerd, 2025-01-09:
Hi, @expresspotato Looking forward to your feedback, but the issue https://github.com/espressif/esp-idf/issues/13605 is being tracked by @esp-zhp. 
@esp-zhp Could you please help with this issues(https://github.com/espressif/esp-idf/issues/13605)?

esp-zhp, 2025-01-09:
@ESP-YTGerd yeah, it's in progress.

expresspotato, 2025-01-14:
@ESP-YTGerd, this seems fixed, as of today there have been no crashes in the past 24 hours. I'll report back if they do occur again.

ESP-YTGerd, 2025-01-15:
> @ESP-YTGerd, this seems fixed, as of today there have been no crashes in the past 24 hours. I'll report back if they do occur again.

@expresspotato If the issue no longer reproduces, please also let us know so we can proceed with merging the fix.

Alvin1Zhang, 2025-03-12:
Thanks for reporting, fix on master branch is available at https://github.com/espressif/esp-idf/commit/94cfe394fe94af5e53da8398af5b886016e7fc11.

We are back-porting the fix onto release branches, feel free to reopen.

closed 2025-03-12 by Alvin1Zhang