espressif/esp-idf #15131: [BUG] ESP32-C3: Clang compiler compatibility issue when handling assembly and link scripts

ErEbusE, 2025-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3.2-dirty

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

1. Assembly code's .size directive should be compiled correctly
2. The link script should correctly identify the rtc_data_seg region definition
3. Clang compiler should behave consistently with GCC compiler Actual Behavior

### What is the actual behavior?

There are two compilation errors:

1. Assembly syntax error:
```
ESP-IDF/components/freertos/FreeRTOS-Kernel/portable/riscv/portasm.S:444:10: error: expected identifier
    .size, .-rtos_current_tcb
         ^
```
[portasm.S.txt](https://github.com/user-attachments/files/18303021/portasm.S.txt)@444g
2. Link script error:There are two compilation errors in memory.ld:
```
ld.lld: error: memory region not defined: rtc_data_seg
>>> _data_seg_org = ORIGIN(rtc_data_seg);
```
[memory.ld.txt](https://github.com/user-attachments/files/18303030/memory.ld.txt)@125g

### Steps to reproduce.

[sdkconfig.txt](https://github.com/user-attachments/files/18303045/sdkconfig.txt)
1.Create a new ESP32-C3 project
2.Set the compiler to esp-clang
3.Execute the compilation command

### More Information.

1.Fix assembly syntax:
```
    ret
    .size rtos_current_tcb, .-rtos_current_tcb
```
2.Fix link script
```
MEMORY 
{
    // ...existing code...
}

// Can the declaration order be swapped?
REGION_ALIAS("rtc_data_seg", rtc_iram_seg);
REGION_ALIAS("rtc_slow_seg", rtc_iram_seg);
REGION_ALIAS("rtc_data_location", rtc_iram_seg);

_data_seg_org = ORIGIN(rtc_data_seg);
```

ErEbusE, 2025-01-04:
> 2.Fix link script
> 
> ```
> MEMORY 
> {
>     // ...existing code...
> }
> 
>  Can the declaration order be swapped?
> REGION_ALIAS("rtc_data_seg", rtc_iram_seg);
> REGION_ALIAS("rtc_slow_seg", rtc_iram_seg);
> REGION_ALIAS("rtc_data_location", rtc_iram_seg);
> 
> _data_seg_org = ORIGIN(rtc_data_seg);
> ```

This doesn't work, there are errors

Lapshin, 2025-01-06:
@ErEbusE , the issue regarding wrong asm syntax was fixed in this commit https://github.com/espressif/esp-idf/commit/9ca231e76ed24801355a88410c82d3ed4b1e0a43

I can not reproduce your issue using clean IDF v5.3.2 and using clang + your sdkconfig file. For some reason in your logs appeared clang version `esp-18.1.2_20240912` which is wrong for IDF v5.3.2 (should be `16.0.1-fe4f10a809`). Also, you should not get assembler errors because in v5.3.2 for clang passed `-no-integrated-as` option (GNU as should be used instead).

After these mismatches it's difficult to say what else could be wrong with your environment... Could you please retry your test with fresh IDF and IDF's `export.bat`/`export.ps1` environment?

closed 2025-03-28 by Alvin1Zhang