espressif/esp-idf #15133: L2CAP_CONFIGURATION_RSP packets lost when establishing HID control channel between ESP32 & BK3513 based ShanWan PS3 controller

darthcloud, 2025-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-698-g030c9957db

### Espressif SoC revision.

ESP32 (ESP32-D0WD-V3 (revision v3.0))

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC-32E & ESP-WROVER-KIT

### Power Supply used.

USB

### What is the expected behavior?

When using an ESP32 as the BT BR/EDR Central device the two L2CAP_CONFIGURATION_RSP send by either the ESP32 or BK3513 should be received by the other device and allow the L2CAP config step to complete sucessfully.

This should result in the ShanWan PS3 controller (BK3513) to start sending HID report to the ESP32.

### What is the actual behavior?

1. L2CAP_CONNECTION step between the two device for the HID Control channel is Sucessful.
2. Each device send L2CAP_CONFIGURATION_REQ.
3. Each device receive the other device L2CAP_CONFIGURATION_REQ.
4. Each device send L2CAP_CONFIGURATION_RSP.
5. Neither device receive the other device L2CAP_CONFIGURATION_RSP.

The ShanWan PS3 controller (BK3513) then timeout and shutdown.

### Steps to reproduce.

I can reproduce 100% so I think the easiest way to debug this would be to send me a debug build of libbtdm_app.a .
I did debug another issue this way a couple year back with BetterJincheng (#10814).

Here the step I used to use the ESP32 with Linux BlueZ stack:

0. Pair ShanWan PS3 controller to ESP32 BDADDR using the sixpair linux USB utility.
1. Build and flash controller_hci_uart_esp_wrover_kit on a ESP-WROVER-KIT.
   This special version of the esp-idf example change the pins to use IOs 12,13,14,15.
```
git clone https://github.com/darthcloud/controller_hci_uart_esp_wrover_kit
cd controller_hci_uart_esp_wrover_kit
idf.py build
idf.py flash
```
2. Set the jumper on JP2 as if using JTAG (First UART of FTDI2232 used for ESP UART1 HCI)
   (TMS, TDO, TDI & TCK)
3. Connect ESP-WROVER-KIT to a Linux PC with USB cable.
4. Attach the ESP32 to BlueZ:
   `sudo hciattach /dev/ttyUSB0 any 921600 flow`
5. Enter `sudo bluetoothctl` and type the following commands:
```
list
select <ESP32 BDADDR FROM LIST>
discoverable on
```
6. Connect ShanWan PS3 controller to ESP32 by pressing P3 button of the controller.

### Debug Logs.

```plain
See below in more information section.
```

### More Information.

# sdkconfig:
[sdkconfig.txt](https://github.com/user-attachments/files/18304691/sdkconfig.txt)

# Btsnoop traces on both the central & peripheral side HCI interface:
[shanwan_btsnoop_traces.zip](https://github.com/user-attachments/files/18304692/shanwan_btsnoop_traces.zip)

This zip archive contain 6 btsnoop trace you can open in wireshark:

## Test with Bluez + ESP32 (Fail)
Done with Ubuntu 24.04 & BlueZ 5.79
* Bluez_ESP32_with_ShanWan_PS3_BK3513_central_fail.log
  Trace of the HCI packets between the ESp32 and BlueZ
* Bluez_ESP32_with_ShanWan_PS3_BK3513_peripheral_fail.log
  Trace of the HCI packets between the ShanWan controller MCU & BK3513

You can see in this trace that both device send a L2CAP_CONFIGURATION_RSP but neither of them receive one.

## Test with BlueZ + Intel AX201 (Sucess)
Done with Ubuntu 24.04 & BlueZ 5.79
* Bluez_Intel_AX201_with_ShanWan_PS3_BK3513_central_success.log
  Trace of the HCI packets between Intel BT and BlueZ
* Bluez_Intel_AX201_with_ShanWan_PS3_BK3513_peripheral_success.log
  Trace of the HCI packets between the ShanWan controller MCU & BK3513

## Test with a 8bitdo receiver (Sucess)
* 8bitdo_Realtek_with_ShanWan_PS3_BK3513_central_success.log
  Trace of the HCI packets between 8bitdo MCU and Realtek radio
* 8bitdo_Realtek_with_ShanWan_PS3_BK3513_peripheral_success.log
  Trace of the HCI packets between the ShanWan controller MCU & BK3513

NOTES: All the peripheral traces were sniffed using 2 independant UART interface. So due to buffering sometimes the request and responces packets are out of order due to the way my script merged the TX & RX packet in the same trace.

# ESP32 boot logs:
[esp32_boot_logs.txt](https://github.com/user-attachments/files/18304693/esp32_boot_logs.txt)

darthcloud, 2025-01-10:
I realize your are probably very busy but let me know if you need any extra information @BetterJincheng.
Thank you

jcbetter, 2025-01-13:
I will start dealing with this issue within this week.

BetterJincheng, 2025-01-13:
@darthcloud Sorry for late reply.
I checked the log file "Bluez_ESP32_with_ShanWan_PS3_BK3513_peripheral_fail.log" and found this:
[image]

Here we see that the host sent two packets, but only received one number_of_complete_packet event. I guess that the controller never received the second package.

Try turning off the flow control on the ESP32 side and reducing the baud rate and try again.

(Top) → Component config → Bluetooth → Controller Options → HCI UART(H4) Options

BetterJincheng, 2025-01-14:
@darthcloud Could you please attach the HCI log using BlueRetro and I want to find some information about flow control.

darthcloud, 2025-01-14:
@BetterJincheng see both BlueRetro traces and Bluez one w/ 115200 noflow in the following zip:
[shawan_ps3_btsnoop_traces_2024_01_13.zip](https://github.com/user-attachments/files/18404599/shawan_ps3_btsnoop_traces_2024_01_13.zip)

Bluez w/ 115200 noflow:
Trace of Bluez <-> ESP32: bluez_shanwan_central_115200_noflow.log
Trace of Shanwan PS3 controller H4 link: bluez_shanwan_peripheral.log

Blueretro files:
Trace of ESP32 internal VHCI: blueretro_shanwan_central_VHCI.log
Trace of Shanwan PS3 controller H4 link: blueretro_shanwan_peripheral.log

Note the peripheral trace got the RX and TX packet a bit out of order due to the way I'm sniffing with 2 indepandant UART.

BetterJincheng, 2025-01-14:
@darthcloud I added some weak functions at the TX and RX of baseband in this lib.([libbtdm_app.zip](https://github.com/user-attachments/files/18405067/libbtdm_app.zip))

[image]

You can implement them in your project.

```
void IRAM_ATTR bt_bb_rx_start(void)
{
    gpio_set_level(x, 1);
}

void IRAM_ATTR bt_bb_rx_end(void)
{
    gpio_set_level(x, 0);
}

void IRAM_ATTR bt_bb_rx_start(void)
{
    gpio_set_level(y, 1);
}

void IRAM_ATTR bt_bb_rx_end(void)
{
    gpio_set_level(y, 0);
}

```
And, I added some status trance. So please attach the log file of ESP32. Reproduce with BlueRetro, please.

darthcloud, 2025-01-14:
@BetterJincheng I made a build with the lib, here the logs:
[blueretro_debug_btlib.zip](https://github.com/user-attachments/files/18405455/blueretro_debug_btlib.zip)
[blueretro_log_debug_lib.txt](https://github.com/user-attachments/files/18405457/blueretro_log_debug_lib.txt)

Regarding the weak function for tx/rx do you want me to make a logic analyzer trace of the 4 signals?

darthcloud, 2025-01-15:
@BetterJincheng 

I redid another test but this time adding the logic analyzer.

Upon sending the L2CAP config responce I triggered my logic analyzer to save the output of the bt_bb rx & tx signals:

I used the following modified code (gpio_set_level was crashing the esp32):
```
void IRAM_ATTR bt_bb_rx_start(void) {
    GPIO.out_w1ts = 1ULL << 13;
}

void IRAM_ATTR bt_bb_rx_end(void) {
    GPIO.out_w1tc = 1ULL << 13;
}

void IRAM_ATTR bt_bb_tx_start(void) {
    GPIO.out_w1ts = 1ULL << 14;
}

void IRAM_ATTR bt_bb_tx_end(void) {
    GPIO.out_w1tc = 1ULL << 14;
}
```

See zip files with Logic trace, btsnoop trace, esp32 log:

[blueretro_shanwan_esp32_logic_trace.zip](https://github.com/user-attachments/files/18417314/blueretro_shanwan_esp32_logic_trace.zip)

[image]

I included the original logic trace file "bt_bb_rx_tx_esp32.lpf"
You can open it with the logicport application:
https://www.pctestinstruments.com/downloads/logicport_2371.exe

BetterJincheng, 2025-01-15:
@darthcloud From the results of the logic analyzer, it can be seen that the packet of L2CAP_CFG_RSP has been continuously retransmitted and has not received an ACK from peer device. But ESP32 has always been able to receive packets from the other end, which may be due to incorrect received packets or the ARQN in the packet being NAK. In this way, we will continue to retransmit L2CAP_CFG_RSP.
Therefore, this is not an issue with HCI. It's more like, for some reason, the peer device keeps making NAK to our L2CAP_CFG_SSP.

darthcloud, 2025-01-15:
BLE disabled tests:
I've set:
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=y

&

esp_bt_controller_enable(ESP_BT_MODE_CLASSIC_BT)

I still see the issue.

See all logs (logic trace, bt trace, esp log) here:
[blueretro_shanwan_esp32_nole.zip](https://github.com/user-attachments/files/18418267/blueretro_shanwan_esp32_nole.zip)

darthcloud, 2025-01-15:
Would an actual wireless trace helps figuring why the device NAK the CFG packet?

Like one trace from the working Intel radio, and another one from the ESP32?

I don't have access to modern wideband sniffer like EllisysExplorer or Frontline x500 unfortunately :(

I got a very old Frontline BPA600 which is kind of a pain to use.
But I could try to set it up if you think that would help.

BetterJincheng, 2025-01-15:
@darthcloud It will be very useful if you can provide data for air interface interaction.

darthcloud, 2025-01-15:
@BetterJincheng 

He the trace from the BPA600 Comprobe
[frontline_comprobe_trace_shanwan_ps3_esp32.zip](https://github.com/user-attachments/files/18419157/frontline_comprobe_trace_shanwan_ps3_esp32.zip)

Do you have the software to read frontline cfa traces?

BetterJincheng, 2025-01-15:
@darthcloud I can read it.

darthcloud, 2025-01-15:
[screenshot_comprobe_traces.zip](https://github.com/user-attachments/files/18419247/screenshot_comprobe_traces.zip)
Here some screenshot of the traces

BetterJincheng, 2025-01-15:
@darthcloud I guessI have found some clues, and I will provide you with a new debug library as soon as possible.

BetterJincheng, 2025-01-15:
[libbtdm_app.zip](https://github.com/user-attachments/files/18420153/libbtdm_app.zip)
@darthcloud Please try with this lib and attach the esp32 log.

darthcloud, 2025-01-15:
@BetterJincheng That one actualy worked once!!!
But all my other attempt failled :(

Here logs of 3 attemps the one that worked and 2 fail one
[blueretro_dbg_lib2_tests.zip](https://github.com/user-attachments/files/18431400/blueretro_dbg_lib2_tests.zip)

BetterJincheng, 2025-01-16:
@darthcloud This lib may work well. 
[libbtdm_app.zip](https://github.com/user-attachments/files/18432900/libbtdm_app.zip)

BetterJincheng, 2025-01-16:
@darthcloud 

The peer device sent a pdu of `lmp_preferred_rate` to ESP32 which indicates that its preferred RX package type is DM1.
[image]
Upon receiving this PDU, ESP32 did modified the allowed TX packet types to DM1.
However, later ESP32 sent another pdu of `lmp_max_slot_req` indicating that it wanted to use packets with a maximum length of 5 slots and the peer device accepted it.
[image]
Therefore, ESP32 modified the allowed TX packet types again and no longer only supports DM1.
Due to the length of L2CAP_CFG_RSP exceeding the maximum allowable length of DM1, ESP32 choses another packet 2-DH1 to send the data.
[image]
I guess the peer device only accepts DM1 packets at this time, so it NAKed our 2-DH1 packets.
[image]
Receive NAK, retransmit 2-DH1(L2CAP_CFG_RSP), Receive NAK, retransmit 2-DH1(L2CAP_CFG_SSP), ......
The packets marked with yellow shadows are retransmission packets.
[image]
According to the core specification, `lmp_preferred_rate` is only a preference and not a mandatory requirement.
Specification says "Device A is not required to change to the packet type specified by this parameter".
Moreover, the peer device also accepted the `lmp_max_slot_req` from ESP32 so it should not reject packets other than DM1.
[image]
The library above is that we have disabled the `lmp_auto_rate` request, so the peer device should no longer send `lmp_perferred_rate`. Of course, this is a workaround specific to your device, and we will not merge this change to the release branches.

darthcloud, 2025-01-16:
@BetterJincheng That last library do allow the device to work!!

But it's not really useful to me if it's not going to be merged, do you have an alternate fix in mind?

ESP32 is the only BT radio I saw that didn't work with that controller.
Competing product to mine using other radio do not have problem with those controllers.

In the Intel trace packet # 4127 lmp_prefered_rate the shanwan does request 2mbps packets right away, what is the sequence before that which make this request different between the ESP32 and the Intel radio? This result in the device L2CAP connection request to be already at 2-DH1 type and avoid the packet type change in the middle of the L2CAP cfg.

While it might within the spec, I find it weird for the ESP32 to change packet type in the middle of the L2CAP handshake.

BetterJincheng, 2025-01-16:
@darthcloud Packet types can be changed anytime after connection established. It is not weird this happens during L2CAP handshake. L2CAP packets is just ACL packets and the controller does not think it is anything special. Packet type changing is driven by many logics.
At the very least, the peer device should not reject 2-DH1. Preferred rate is just a preference. The peer device can select any packet types listed in `lmp_packet_type_table_req`.

darthcloud, 2025-01-16:
Ok, but what about the other aspects I mentioned, on others radio the shanwan use use 2-DH1 for all ACL packet. What would trigger it to use basic rate with ESP32 only?

BetterJincheng, 2025-01-16:
@darthcloud This is from core spec.
[image]

I don't know why it wants to use basic rate in communication with ESP32.

BetterJincheng, 2025-01-16:
@darthcloud  Maybe we can provide a vendor-specific HCI command to mask the LMP feature bit of CQDDR ?
[image]

darthcloud, 2025-01-16:
I don't mind having to use a vendor-specific HCI command for masking LMP feature bits.

I think it could be handy if it allowed any of the bits to be set/clear for debugging purpose as a bonus.

But I still think you should have some logic to fallback to basic rates on repeated transmission failure with EDR rates.

BetterJincheng, 2025-02-07:
@darthcloud Vendor specific HCI commands are available, which will not take long to access on GitHub.

closed 2025-02-28 by espressif-bot

darthcloud, 2025-02-28:
@BetterJincheng Do you some documentation on how to use the "added a VSC to control whether to initiate lmp_auto_rate"

jcbetter, 2025-03-03:
@darthcloud 

The VSC has already merged into the main branch, and we are looking for a better way to disclose to developers the VSC and VSE we support. If you need to use this feature immediately, you can refer to the following temporarily.

```
Write Auto Rate Enable command
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
﻿
Description
^^^^^^^^^^^
﻿
This command is used to control whether the local Bluetooth controller has the ability to actively initiate to send LMP_AUTO_DATE to enable Channel quality driven data rate.
﻿
OPCODE
^^^^^^
﻿
0xFD8B
﻿
Command parameters
^^^^^^^^^^^^^^^^^^
﻿
- Enable: 1 octet
- 0x00: The local Bluetooth controller will not send LMP_AUTO_DATE
- 0x01: The local Bluetooth controller may send LMP_AUTO_DATE
﻿
Return parameters
^^^^^^^^^^^^^^^^^
﻿
None
﻿
Event(s) generated
^^^^^^^^^^^^^^^^^^
﻿
When the HCI_Vendor_BT_Write_Auto_Rate_Enable command has completed, an HCI_Command_Complete event shall be generated.
```

darthcloud, 2025-03-05:
@BetterJincheng 

I have updated to the following version of esp-idf (latest as of today):

```
commit a6c3a9cbbb5e0a99e0d30e58271f27a0d4e9ae90 (HEAD -> master, origin/master, origin/HEAD)
Merge: 83e8f70ee4 6858b37f51
Author: Sudeep Mohanty <sudeep.mohanty@espressif.com>
Date:   Thu Feb 27 22:03:12 2025 +0800

    Merge branch 'bugfix/adc_work_mode_for_ulp_riscv' into 'master'

    fix(ulp-riscv): Fixed ADC oneshot initialization for ULP RISC-V

    See merge request espressif/esp-idf!37245
```

I also double checked the bt lib version:

```
commit fbbb054cbc5c8b5aa466208dc8d12ccc10d7e08c (HEAD, origin/master, origin/HEAD, master)
Author: Jin Cheng <JinCheng@espressif.com>
Date:   Fri Feb 21 11:03:06 2025 +0800

    fix(bt): Update libbtdm_app.a to 94054498

        1. added a VSC to control whether to initiate lmp_auto_rate
        2. fixed EA resource cleanup error after SNIFF negotiation failure
        3. removed an assertion in SCO data TX handler after disconnection
```

I sent the vendor cmd opcode: 0xFD8B with parmeter 0x00.

See packet #79 in `shanwan_verdor_cmd.log`

Command is ack sucessfully in packet #80

However the L2CAP config still fail as before for the shanwan controller.

For reference see trace `shanwan_dbg_bt_lib.log` were it worked with your previous debug lib.

jcbetter, 2025-03-06:
@darthcloud From the packet capture screenshot of the previous reply, we can see that lmp_auto_rate was already interactive at a relatively early stage. So, 0xfd8b is recommended to be issued at a very early stage, such as immediately after hci_reset.

darthcloud, 2025-03-06:
I moved it after HCI reset and it didn't help:
[shanwan_vendor_cmd_after_rst.zip](https://github.com/user-attachments/files/19100422/shanwan_vendor_cmd_after_rst.zip)

jcbetter, 2025-03-06:
@darthcloud I added some trace log，please help to reproduce with this lib. Thanks~

[libbtdm_app.zip](https://github.com/user-attachments/files/19100637/libbtdm_app.zip)

jcbetter, 2025-03-06:
> [@darthcloud](https://github.com/darthcloud) I added some trace log，please help to reproduce with this lib. Thanks~
> 
> [libbtdm_app.zip](https://github.com/user-attachments/files/19100637/libbtdm_app.zip)

@darthcloud No need to try anymore, I have made some mistakes and will provide a new lib later.

jcbetter, 2025-03-06:
@darthcloud This lib should work well.

[libbtdm_app.zip](https://github.com/user-attachments/files/19100854/libbtdm_app.zip)

darthcloud, 2025-03-06:
@BetterJincheng 

Here the serial logs:
[debug_log.txt](https://github.com/user-attachments/files/19101087/debug_log.txt)

[shanwan_vendor_cmd_after_dbg_lib.zip](https://github.com/user-attachments/files/19101074/shanwan_vendor_cmd_after_dbg_lib.zip)

It still didn't work however.

jcbetter, 2025-03-06:
@darthcloud “get from 0” is a check performed by ESP32 before sending lmp_auto_rate. If it gets "0", it will not send the PDU.
If this still doesn't solve the problem, you may need to provide the captured file again for analysis.

darthcloud, 2025-03-06:
Here the OTA trace of both the latest lib and the one that worked in january:
[capture_march_frontline.zip](https://github.com/user-attachments/files/19108136/capture_march_frontline.zip)

jcbetter, 2025-03-07:
@darthcloud 
From the OTA trace we can see that ESP32 did not send any lmp_auto_rate but peer device sent one. However, ESP32 did not provide any response by lmp_preferred_rate and it is in compliance with the core specification. The L2CAP_CONFIGURATION_RSP sent by esp32 has been NAK by the peer device.

[image]

[image]

The previously provided working library doesn't seem to have much difference, it's just that the peer device ACKed the packet sent by esp32.

[image]

darthcloud, 2025-03-09:
Was threre others change in the debug library you sent me in January that are missing in the VSC implementation?

January lib work with 100% reliability. It always work.

I think there is something in there that must have fixed the issue and the VSC cmd does not produce an equivalent behavior to the january build.

I think the issue should be reopen, since the VSC cms does not fix the issue.

jcbetter, 2025-03-10:
@darthcloud The only difference between these two libraries is whether ESP32 has initiated a request for a packet type table. In our SDK, lmp_auto_rate and lmp_packet_type_table_deq are placed together and executed sequentially. At that time, in order to verify whether the issue was caused by lmp_auto_rate, I simply commented out these two lmps as a whole. Now it seems that the problem may not be with lmp_auto_rate, but rather with some behaviors starting to become incorrect after the peer device receives the lmp_acker_type_table_req from ESP32.
Now I am ignoring the VSC of 0XFD8B and only forcing ESP32 not to initiate lmp_packet_type_table_deq. If I guess correctl it will work well.

[libbtdm_app.zip](https://github.com/user-attachments/files/19158356/libbtdm_app.zip)