espressif/esp-idf #15137: The ESP32-P4 encountered an error while running the example MIPI DSI ILI9881C screen driver. The `esp_lcd_panel_io_tx_param` function got stuck during the transmission of the `vendor_specific_init_default` initialization code, continuously triggering the watchdog timer.

Llgok, 2025-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

I have tried versions v5.3.2, v5.4.0, and the master branch v5.5.0.

### Espressif SoC revision.

ESP32P4 

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

The ESP32-P4-Function-EV-Board-v1.4 board and the ESP32P4 board that I designed myself.

### Power Supply used.

USB

### What is the expected behavior?

Screen initialization -> Execute LVGL task

### What is the actual behavior?

The screen failed to initialize successfully; the code is stuck at the position of the `esp_lcd_panel_io_tx_param` function initialization. It seems that `esp_lcd_panel_io_tx_param` can only send initialization command codes consecutively up to 17 times. Any attempts beyond 17 times will cause the system to get stuck and continuously trigger a watchdog.

### Steps to reproduce.

1.  Create an example of mipi_dsi
2.  Change the board to ESP32P4
3.  Modify the `Select MIPI LCD model` screen type in the SDK configuration editor to ILI9881C
4.  Compile and flash
5.  Check the program for a deadlock triggering the watchdog via serial port

### Debug Logs.

```plain
ESP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
rst:0x1 (POWERON),boot:0x30f (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x4ff33ce0,len:0x1190
load:0x4ff2abd0,len:0xaf4
load:0x4ff2cbd0,len:0x30b0
entry 0x4ff2abd0
I (130) hex_psram: vendor id    : 0x0d (AP)
I (131) hex_psram: Latency      : 0x01 (Fixed)
I (131) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (132) hex_psram: dev id       : 0x03 (generation 4)
I (136) hex_psram: density      : 0x07 (256 Mbit)
I (141) hex_psram: good-die     : 0x06 (Pass)
I (145) hex_psram: SRF          : 0x02 (Slow Refresh)
I (150) hex_psram: BurstType    : 0x00 ( Wrap)
I (154) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (158) hex_psram: BitMode      : 0x01 (X16 Mode)
I (163) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (168) hex_psram: DriveStrength: 0x00 (1/1)
I (172) MSPI DQS: tuning success, best phase id is 2
I (355) MSPI DQS: tuning success, best delayline id is 12
I (355) esp_psram: Found 32MB PSRAM device
I (355) esp_psram: Speed: 200MHz
I (356) hex_psram: psram CS IO is dedicated
I (359) cpu_start: Multicore app
I (1312) esp_psram: SPI SRAM memory test OK
I (1322) cpu_start: Pro cpu start user code
I (1322) cpu_start: cpu freq: 360000000 Hz
I (1322) app_init: Application information:
I (1322) app_init: Project name:     mipi_dsi_panel
I (1327) app_init: App version:      1
I (1330) app_init: Compile time:     Jan  4 2025 14:21:37
I (1335) app_init: ELF file SHA256:  b8b8e9592...
I (1340) app_init: ESP-IDF:          v5.5-dev-1050-gb5ac4fbdf9
I (1345) efuse_init: Min chip rev:     v0.1
I (1349) efuse_init: Max chip rev:     v1.99 
I (1353) efuse_init: Chip rev:         v1.0
I (1357) heap_init: Initializing. RAM available for dynamic allocation:
I (1364) heap_init: At 4FF254A0 len 00015B20 (86 KiB): RAM
I (1369) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (1374) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (1380) heap_init: At 50108080 len 00007F80 (31 KiB): RTCRAM
I (1385) heap_init: At 30100068 len 00001F98 (7 KiB): TCM
I (1390) esp_psram: Adding pool of 32768K of PSRAM memory to heap allocator
I (1397) spi_flash: detected chip: generic
I (1400) spi_flash: flash io: dio
W (1403) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (1416) main_task: Started on CPU0
I (1426) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1426) main_task: Calling app_main()
I (1426) gpio: GPIO[20]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1436) example: MIPI DSI PHY Powered on
I (1446) example: Install MIPI DSI LCD control IO
I (1446) example: Install MIPI DSI LCD data panel
E (6496) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (6496) task_wdt:  - IDLE0 (CPU 0)
E (6496) task_wdt: Tasks currently running:
E (6496) task_wdt: CPU 0: main
E (6496) task_wdt: CPU 1: IDLE1
E (6496) task_wdt: Print CPU 0 (current core) backtrace
esp_backtrace_print: Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x40046a18  RA      : 0x40046a12  SP      : 0x4ff28540  GP      : 0x4ff11800  
TP      : 0x4ff28710  T0      : 0x000004e2  T1      : 0x48000a08  T2      : 0x00000000  
S0/FP   : 0x00000001  S1      : 0x48000a0c  A0      : 0x48000a0c  A1      : 0x00000000  
A2      : 0x00000006  A3      : 0x00000000  A4      : 0x500a0000  A5      : 0x00000801  
A6      : 0x00000000  A7      : 0x00000008  S2      : 0x4ff2859b  S3      : 0x00000000  
S4      : 0x00000006  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000004  T4      : 0x00000003  T5      : 0x000000da  T6      : 0x0000003e  
MSTATUS : 0x00011888  MTVEC   : 0x4ff00003  MCAUSE  : 0xdeadc0de  MTVAL   : 0xdeadc0de  
MHARTID : 0x00000000  
E (11496) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (11496) task_wdt:  - IDLE0 (CPU 0)
E (11496) task_wdt: Tasks currently running:
E (11496) task_wdt: CPU 0: main
E (11496) task_wdt: CPU 1: IDLE1
E (11496) task_wdt: Print CPU 0 (current core) backtrace
esp_backtrace_print: Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x40046a18  RA      : 0x40046a12  SP      : 0x4ff28540  GP      : 0x4ff11800  
TP      : 0x4ff28710  T0      : 0x000004e2  T1      : 0x48000a08  T2      : 0x00000000  
S0/FP   : 0x00000001  S1      : 0x48000a0c  A0      : 0x48000a0c  A1      : 0x00000000  
A2      : 0x00000006  A3      : 0x00000000  A4      : 0x500a0000  A5      : 0x00000801  
A6      : 0x00000000  A7      : 0x00000008  S2      : 0x4ff2859b  S3      : 0x00000000  
S4      : 0x00000006  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000004  T4      : 0x00000003  T5      : 0x000000da  T6      : 0x0000003e  
MSTATUS : 0x00011888  MTVEC   : 0x4ff00003  MCAUSE  : 0xdeadc0de  MTVAL   : 0xdeadc0de  
MHARTID : 0x00000000  
E (16496) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (16496) task_wdt:  - IDLE0 (CPU 0)
E (16496) task_wdt: Tasks currently running:
E (16496) task_wdt: CPU 0: main
E (16496) task_wdt: CPU 1: IDLE1
E (16496) task_wdt: Print CPU 0 (current core) backtrace
esp_backtrace_print: Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x40046a18  RA      : 0x40046a12  SP      : 0x4ff28540  GP      : 0x4ff11800  
TP      : 0x4ff28710  T0      : 0x000004e2  T1      : 0x48000a08  T2      : 0x00000000  
S0/FP   : 0x00000001  S1      : 0x48000a0c  A0      : 0x48000a0c  A1      : 0x00000000  
A2      : 0x00000006  A3      : 0x00000000  A4      : 0x500a0000  A5      : 0x00000801  
A6      : 0x00000000  A7      : 0x00000008  S2      : 0x4ff2859b  S3      : 0x00000000  
S4      : 0x00000006  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000004  T4      : 0x00000003  T5      : 0x000000da  T6      : 0x0000003e  
MSTATUS : 0x00011888  MTVEC   : 0x4ff00003  MCAUSE  : 0xdeadc0de  MTVAL   : 0xdeadc0de  
MHARTID : 0x00000000
```

### More Information.

In the `esp_lcd_ili9881c.c` file, if you comment out the code related to the `esp_lcd_panel_io_tx_param` function in the `static esp_err_t panel_ili9881c_init(esp_lcd_panel_t *panel)` function on line 366, the program will no longer get stuck and trigger the watchdog. After testing, it was found that the program gets stuck on the 17th use of the `esp_lcd_panel_io_tx_param` function. Upon opening the internal implementation of the function, it seems that commenting out the line `mipi_dsi_host_ll_gen_set_packet_header(hal->host, vc, dt, pkt_hdr_msb, pkt_hdr_lsb);` on line 160 in the `mipi_dsi_hal.c` file also prevents the program from getting stuck.

ginkgm, 2025-01-06:
Hi @Llgok  

Thanks for your report, we will look into this issue

ginkgm, 2025-01-06:
Shorten the title. Orignal title: `The ESP32-P4 encountered an error while running the example MIPI DSI ILI9881C screen driver. The `esp_lcd_panel_io_tx_param` function got stuck during the transmission of the `vendor_specific_init_default` initialization code, continuously triggering the watchdog timer.`

suda-morris, 2025-01-06:
Hi @Llgok  I connected my ILI9881C LCD but didn't reproduce your issue. 

```
I (1264) main_task: Calling app_main()
I (1264) gpio: GPIO[20]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1264) example: MIPI DSI PHY Powered on
I (1264) example: Install MIPI DSI LCD control IO
I (1264) example: Install MIPI DSI LCD data panel
I (1264) ili9881c: ID1: 0x98, ID2: 0x81, ID3: 0x5c
I (1444) example: Initialize LVGL library
```

I'm on the IDF master branch, test with the `lcd/mipi_dsi` example, only select the ILI9881C LCD model in the menuconfig, no other code change. 

Maybe you can try to upgrade the ili9881c component to e.g. `  esp_lcd_ili9881c: "^1.0.0"`?

Llgok, 2025-01-07:
@suda-morris 
May I ask if there will be a watchdog timeout error reported when the ILI9881C LCD is not connected?

suda-morris, 2025-01-07:
@Llgok yes, if you didn't connect any MIPI LCD during initialization, then the error you met is as expected. Every command sent from MIPI host will wait for an ACK from the device.

Creeper666, 2025-01-07:
@suda-morris 
我在使用这个例程[https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/device/usb_extend_screen](url)的时候，也遇到了类似问题。
在`managed_components\espressif__esp_lcd_ek79007\esp_lcd_ek79007.c`中第107行开始，EK79007默认只有9条初始化指令。但我使用的屏幕需要66行LCD初始化命令，就会触发看门狗，程序也无法正常运行。报错如下：
```
E (6250) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (6250) task_wdt:  - IDLE0 (CPU 0)
E (6250) task_wdt: Tasks currently running:
E (6250) task_wdt: CPU 0: main
E (6250) task_wdt: CPU 1: IDLE1
E (6250) task_wdt: Print CPU 0 (current core) backtrace
esp_backtrace_print: Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x400200a0  RA      : 0x400196b2  SP      : 0x4ff1c9f0  GP      : 0x4ff13b80
--- 0x400200a0: mipi_dsi_host_ll_gen_is_cmd_fifo_full at ***/v5.4/esp-idf/components/hal/esp32p4/include/hal/mipi_dsi_host_ll.h:723     
 (inlined by) mipi_dsi_hal_host_gen_write_dcs_command at ***/v5.4/esp-idf/components/hal/mipi_dsi_hal.c:152
0x400196b2: panel_io_dbi_tx_param at ***/v5.4/esp-idf/components/esp_lcd/dsi/esp_lcd_panel_io_dbi.c:84

TP      : 0x4ff1cb60  T0      : 0xffff8000  T1      : 0x4ff13459  T2      : 0x00000855
S0/FP   : 0x400516b8  S1      : 0x00000000  A0      : 0x00880000  A1      : 0x00000000
A2      : 0x00000015  A3      : 0x500a0000  A4      : 0x00008300  A5      : 0x00060016
A6      : 0x00000002  A7      : 0x00000010  S2      : 0x00000013  S3      : 0x10624dd3
S4      : 0x480e623c  S5      : 0x480e6264  S6      : 0x4ff2aee0  S7      : 0x00000019
S8      : 0x000003e8  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000002  T4      : 0x00000001  T5      : 0x0000014d  T6      : 0x00000869
MSTATUS : 0x00011888  MTVEC   : 0x4ff00003  MCAUSE  : 0xdeadc0de  MTVAL   : 0xdeadc0de
--- 0x4ff00003: _vector_table at ??:?

MHARTID : 0x00000000
```
但是使用`lvgl_demo_v9`时，则没有这种情况发生。
IDF版本是5.4.0。

Llgok, 2025-01-08:
I have identified the cause of the issue. It is necessary to add several timeout judgment code segments under the function `mipi_dsi_host_ll_gen_set_packet_header(hal->host, vc, dt, pkt_hdr_msb, pkt_hdr_lsb);`.

closed 2025-01-08 by Llgok

misiek1994, 2025-02-06:
So what's the official solution? My code is also stuck in mipi_dsi_host_ll_gen_is_cmd_fifo_full and triggers watchdog. Do we have any idea why Hal is not working correctly in this case?