espressif/esp-idf #15138: Ethernet not NTP server time is not updated if WiFi interface is enabled

rmshub, 2025-01-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

To enabled NTP server time updated from both Ethernet and wifi network any specific required. If wifi soft ap or station enabled, ethernet is not getting NTP server time if cable is connected to router

rmshub, 2025-01-05:
When WiFi SoftAP is enabled, it is sending request through SoftAP (192.168.4.1) interface not sending through ethernet
[image]

How to enable NTP request through ethernet if wifi softap is enabled.

david-cermak, 2025-01-06:
You need to make sure that Ethernet `route_prio` is greater than the WiFi's `route_prio` (and Ethernet must be in the "link-up" state).
You can also use `esp_netif_set_default_netif()` to override these priorities and use the selected netif as the default one.