espressif/esp-idf #15143: Python 3.9.0, "python3" has been detected, error: Activation script failed

mihirk98, 2025-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Successful export.

### What is the actual behavior?

Issue on `source export.sh,`

`Checking "python3" ...
Python 3.9.0
"python3" has been detected
error: Activation script failed`

### Steps to reproduce.

Using [this](https://github.com/shorepine/tulipcc/blob/main/docs/tulip_flashing.md) to compile and flash tulip.

Instructions,

`cd ~
mkdir esp
cd esp
git clone https://github.com/espressif/esp-idf.git
cd esp-idf
git checkout 70f222e5d29df8ffe5da25057601708c8097bcd1
git submodule update --init --recursive
./install.sh esp32s3
source export.sh`

fhrbata, 2025-01-05:
Hello @mihirk98,

thank you for bringing this issue to our attention. It seems like it might be a similar problem to the one mentioned in https://github.com/espressif/esp-idf/issues/15006. Please try running `install.sh` and `export.sh` in a new terminal where the IDF environment hasn't been activated yet. This issue should hopefully be resolved by commit https://github.com/espressif/esp-idf/commit/c0954cf0b280332ef4d0c1f17a68196d17da5e05. However, the commit you are using, `70f222e5d29df8ffe5da25057601708c8097bcd1`, does not include this fix. Alternatively, you can work around this by using the legacy export scripts: set `export ESP_IDF_LEGACY_EXPORT=1` and run `source export.sh` again. It appears that the issue is `IDF_PYTHON_ENV_PATH` being set to an unexpected location.

Thank you.

mihirk98, 2025-01-05:
> Hello @mihirk98,
> 
> thank you for bringing this issue to our attention. It seems like it might be a similar problem to the one mentioned in #15006. Please try running `install.sh` and `export.sh` in a new terminal where the IDF environment hasn't been activated yet. This issue should hopefully be resolved by commit [c0954cf](https://github.com/espressif/esp-idf/commit/c0954cf0b280332ef4d0c1f17a68196d17da5e05). However, the commit you are using, `70f222e5d29df8ffe5da25057601708c8097bcd1`, does not include this fix. Alternatively, you can work around this by using the legacy export scripts: set `export ESP_IDF_LEGACY_EXPORT=1` and run `source export.sh` again. It appears that the issue is `IDF_PYTHON_ENV_PATH` being set to an unexpected location.
> 
> Thank you.

Thank you for your time and help, the second solution definitely works. Haven't tried the first.

closed 2025-01-05 by mihirk98