espressif/esp-idf #15145: How to properly read the local IP address of the device

pavel808, 2025-01-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have an esp32-s3 device and I am trying to read its assigned local Wi-Fi IP address as below using `esp_netif_get_default_netif() `

This does not work unfortunately. `intf` is not NULL, but doesn't get the actual IP address etc. 

What is strange however is that this actually worked on a previous network, but not on the current one. 

I need to obtain the IP address automatically because I need to delegate an mDNS hostname using `mdns_delegate_hostname_add() `. 

Hard-coding the device IP address works fine and the mDNS server works as expected then, but of course the address needs to be read from the network interface somehow. 
  
    esp_netif_t *intf = esp_netif_get_default_netif();;
	while (intf == NULL)
	{
	    ESP_LOGE(TAG, "ERROR ! netif is NULL");
        sleep(2);
	    intf = esp_netif_get_default_netif();
	}

	mdns_ip_addr_t addr4, addr6;
	addr4.addr.type = ESP_IPADDR_TYPE_V4;
	esp_netif_ip_info_t info;
	esp_netif_get_ip_info(intf, &info);

    // IP address is 0.0.0.0 when printed
	ESP_LOGI(TAG, "IP Address of device is : " IPSTR "\n", IP2STR(&info.ip));

nopnop2002, 2025-01-05:
```
    esp_netif_ip_info_t ip_info;
    esp_netif_get_ip_info(esp_netif_get_handle_from_ifkey("WIFI_STA_DEF"), &ip_info);
    ESP_LOGI(TAG, "ip_info.ip=%s", ip4addr_ntoa((const ip4_addr_t *)&ip_info.ip));
    ESP_LOGI(TAG, "ip_info.ip=" IPSTR, IP2STR(&ip_info.ip));
```

david-cermak, 2025-01-06:
The issue arises because the DHCP client hasn't yet assigned an IP address to the interface when you're calling `esp_netif_get_ip_info()`, so it returns 0.0.0.0. This is expected behavior during the network setup phase. Instead of polling for the network interface, use the ESP-IDF event system to listen for the `IP_EVENT_STA_GOT_IP` event, which indicates that the interface has acquired a valid IP address. This ensures you only attempt to retrieve the IP once it’s ready, avoiding timing issues. After receiving the event, you can safely use the IP address for tasks like delegating the mDNS hostname.

pavel808, 2025-01-10:
@nopnop2002 @david-cermak Thanks. I have solved it now. Assigning a callback to assign the hostname when `IP_EVENT_STA_GOT_IP `occurs works the best for me.