espressif/esp-idf #15147: wifi 802.11 tx esp_wifi_set_channel fail with station example

mintisan, 2025-01-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2-1-g7c4d58e215

### Espressif SoC revision.

ESP326

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitM-1

### Power Supply used.

USB

### What is the expected behavior?

After esp_wifi_start, you should be able to call esp_wifi_set_channel and esp_wifi_80211_tx, returning a result of err=0, and send 802.11 broadcast data frames.

### What is the actual behavior?

I referred to the modifications in this [wifi 802.11 tx on channel 1 only (IDFGH-14111) #14919 issue](https://github.com/espressif/esp-idf/issues/14919) and made changes to the esp-idf/examples/wifi/getting_started/station routine, adding the following code. 

```c
    if (bits & WIFI_CONNECTED_BIT) {
        ESP_LOGI(TAG, "connected to ap SSID:%s password:%s",
                 EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);

    //////////////////////// add here start
    sleep(1);
    esp_err_t err_set_channel = esp_wifi_set_channel(6, WIFI_SECOND_CHAN_NONE);
    ESP_LOGI(TAG, "set channel 6:%x",err_set_channel);

    static const char ds2ds_pdu[] = {
        0x48, 0x03, 0x00, 0x00, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
        0xE8, 0x65, 0xD4, 0xCB, 0x74, 0x19, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
        0x60, 0x94, 0xE8, 0x65, 0xD4, 0xCB, 0x74, 0x1C, 0x26, 0xB9,
        0x0D, 0x02, 0x7D, 0x13, 0x00, 0x00, 0x01, 0xE8, 0x65, 0xD4, 0xCB, 0x74,
        0x1C, 0x00, 0x00, 0x26, 0xB9, 0x00, 0x00, 0x00, 0x00
    };
    esp_err_t err = esp_wifi_80211_tx(WIFI_IF_STA, ds2ds_pdu, sizeof(ds2ds_pdu), true);
    ESP_ERROR_CHECK_WITHOUT_ABORT(err);

    ESP_LOGI(TAG, "set channel 6:%x",err);
```

However, the log output is `W (3510) wifi:STA is scanning or connecting, or AP has connected with external STAs, cannot set channel`. I'm not sure why. Below are the completed code and logs.

Code:

```
[... 247 lines cut ...]
I (1420) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (1430) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (1440) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (1450) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (1470) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1470) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (1480) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (1490) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
W (1510) wifi:<ba-add>idx:0, ifx:0, tid:7, TAHI:0x1008561, TALO:0x4691297a, (ssn:1, win:64, cur_ssn:1), CONF:0xc0007005
W (1510) wifi:<ba-add>idx:1, ifx:0, tid:0, TAHI:0x1008561, TALO:0x4691297a, (ssn:2, win:64, cur_ssn:2), CONF:0xc0000005
I (2500) esp_netif_handlers: sta ip: 192.168.8.236, mask: 255.255.255.0, gw: 192.168.8.1
I (2500) wifi station: got ip:192.168.8.236
I (2500) wifi station: connected to ap SSID:GL-MT3000-2A2 password:LIn19881108
W (3510) wifi:STA is scanning or connecting, or AP has connected with external STAs, cannot set channel

I (3510) wifi station: set channel 6:ffffffff
I (3510) wifi station: set channel 6:0
I (3510) main_task: Returned from app_main()
```

### Steps to reproduce.

1. modify the `station_example_main.c` example
2. idf.py build flash monitor  
3. see the log

mintisan, 2025-01-06:
@zhangyanjiaoesp  Can you help me figure out the reason? Is it an issue caused by the new version, or is there a problem with my code? Thank you.

mintisan, 2025-01-06:
@MaxwellAlan If you need any additional information, feel free to ask, and I'll provide it for you. Thank you.

mintisan, 2025-01-06:
The following is the 802.11 data frame captured after setting on channel 11. I didn't see the same packets as in the [issue](https://github.com/espressif/esp-idf/issues/14919), so it should not have been sent successfully. This corresponds with the results in the logs.

[image]

zhangyanjiaoesp, 2025-01-06:
@mintisan 
Please take a look at this line when using the `esp_wifi_set_channel()`:
https://github.com/espressif/esp-idf/blob/b5ac4fbdf9e9fb320bb0a98ee4fbaa18f8566f37/components/esp_wifi/include/esp_wifi.h#L764

In the https://github.com/espressif/esp-idf/issues/14919#issuecomment-2499831250,  when I call `esp_wifi_set_channel()`, I have commented out the `esp_wifi_connect()` function and the code for waiting connection bit. Therefore, setting the channel can be successful. 

If you need the connection function, please call the `esp_wifi_set_channel()` before `esp_wifi_connect()`.

mintisan, 2025-01-06:
@zhangyanjiaoesp Hello, thank you very much for your reply. 

Question 1: Moving `esp_wifi_set_channel(11, WIFI_SECOND_CHAN_NONE);` to after `esp_wifi_connect();` does indeed eliminate the error with `esp_wifi_set_channel`. 
However, I found that after calling `esp_wifi_connect` and then checking with `esp_wifi_get_channel`, the main channel is reset to 1. Therefore, in STA mode, it is not possible to send data on a specified channel using 802.11 `esp_wifi_80211_tx` after `esp_wifi_connect`,right? 

Question 2: This is my inquiry; if you can answer it, I would be very grateful. My actual requirement is to send small amounts of periodic sensor data locally to a server with as low power consumption as possible[**<4mA @1Hz with 10Bytes**]. So I used our [ITWT routine](https://github.com/espressif/esp-idf/tree/master/examples/wifi/itwt) for wake-up[**～2.5mA @1Hz**] and discovered that TCP or UDP consumes quite a bit of power. Therefore, I plan to use `esp_wifi_80211_tx` for transmission and have a service running within the local network capture this data information but encountered the current issue.The above is what I believe to be the lowest power consumption solution for transmitting small amounts of data with ESP32C6, right?

Below is the complete code and log for question 1, FYI.

```c
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via project configuration menu

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_ESP_MAXIMUM_RETRY  CONFIG_ESP_MAXIMUM_RETRY

#if CONFIG_ESP_WPA3_SAE_PWE_HUNT_AND_PECK
#define ESP_WIFI_SAE_MODE WPA3_SAE_PWE_HUNT_AND_PECK
#define EXAMPLE_H2E_IDENTIFIER ""
#elif CONFIG_ESP_WPA3_SAE_PWE_HASH_TO_ELEMENT
#define ESP_WIFI_SAE_MODE WPA3_SAE_PWE_HASH_TO_ELEMENT
#define EXAMPLE_H2E_IDENTIFIER CONFIG_ESP_WIFI_PW_ID
#elif CONFIG_ESP_WPA3_SAE_PWE_BOTH
#define ESP_WIFI_SAE_MODE WPA3_SAE_PWE_BOTH
#define EXAMPLE_H2E_IDENTIFIER CONFIG_ESP_WIFI_PW_ID
#endif
#if CONFIG_ESP_WIFI_AUTH_OPEN
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_OPEN
#elif CONFIG_ESP_WIFI_AUTH_WEP
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WEP
#elif CONFIG_ESP_WIFI_AUTH_WPA_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WPA_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA2_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WPA2_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA_WPA2_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WPA_WPA2_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA3_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WPA3_PSK
#elif CONFIG_ESP_WIFI_AUTH_WPA2_WPA3_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WPA2_WPA3_PSK
#elif CONFIG_ESP_WIFI_AUTH_WAPI_PSK
#define ESP_WIFI_SCAN_AUTH_MODE_THRESHOLD WIFI_AUTH_WAPI_PSK
#endif

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t s_wifi_event_group;
[... 290 lines cut ...]
I (1440) wifi:ifidx:0, rssi:-33, nf:-97, phytype(0x3, CBW20-SGI), phymode(0x7, 11ax), max_rate:172, he:1, vht:0, ht:1
I (1450) wifi:(ht)max.RxAMPDULenExponent:3(65535 bytes), MMSS:4(2 us)
I (1470) wifi:(extcap)mbssid:0, enhanced_mbssid_advertise:0, complete_nontxbssid_profiles:0, twt_responder: 1
I (1470) wifi:connected with GL-MT3000-2A2, aid = 4, channel 1, BW20, bssid = 7a:29:91:46:61:85
I (1470) wifi:security: WPA2-PSK, phy:11ax, rssi:-33, cipher(pairwise:0x3, group:0x3), pmf:0, 
I (1490) wifi:pm start, type: 1, twt_start:0

I (1490) wifi:pm start, type:1, aid:0x4, trans-BSSID:7a:29:91:46:61:85, BSSID[5]:0x85, mbssid(max-indicator:0, index:0), he:1
I (1500) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (1500) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (1510) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (1520) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (1530) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (1550) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (1550) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (1560) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1570) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
W (1590) wifi:<ba-add>idx:0, ifx:0, tid:7, TAHI:0x1008561, TALO:0x4691297a, (ssn:2, win:64, cur_ssn:2), CONF:0xc0007005
W (1590) wifi:<ba-add>idx:1, ifx:0, tid:0, TAHI:0x1008561, TALO:0x4691297a, (ssn:2, win:64, cur_ssn:2), CONF:0xc0000005
I (2580) esp_netif_handlers: sta ip: 192.168.8.236, mask: 255.255.255.0, gw: 192.168.8.1
I (2580) wifi station: got ip:192.168.8.236
I (2580) wifi station: connected to ap SSID:GL-MT3000-2A2 password:LIn19881108
I (2590) main_task: Returned from app_main()

```

zhangyanjiaoesp, 2025-01-07:
@mintisan 
Answer1: After the STA connects to AP, the channel will follow the AP's channel. 

> Therefore, in STA mode, it is not possible to send data on a specified channel using 802.11 `esp_wifi_80211_tx` after `esp_wifi_connect`,right?

Yes.  If you want to send data on channel 11, how about change the AP's channel to 11?

Answer2: yes， 80211 tx will be a good choice.

mintisan, 2025-01-10:
@zhangyanjiaoesp Thanks for your reply.
According to your request, it is indeed possible to set the router's channel so that the channel can be fixed to a specified number(like 11). However, based on the current itwt routine, I am unable to capture the custom-sent `ds2ds_pdu` array data.

The code I modified and added based on main_itwt.c is as follows:[no esp_wifi_set_channel called]
[image]

```c
static void itwt_setup_handler(void *arg, esp_event_base_t event_base,
                               int32_t event_id, void *event_data)
{
    static const char ds2ds_pdu[] = {
        0x48, 0x03, 0x00, 0x00, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
        0xE8, 0x65, 0xD4, 0xCB, 0x74, 0x19, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF,
        0x60, 0x94, 0xE8, 0x65, 0xD4, 0xCB, 0x74, 0x1C, 0x26, 0xB9,
        0x0D, 0x02, 0x7D, 0x13, 0x00, 0x00, 0x01, 0xE8, 0x65, 0xD4, 0xCB, 0x74,
        0x1C, 0x00, 0x00, 0x26, 0xB9, 0x00, 0x00, 0x00, 0x00
    };
    wifi_event_sta_itwt_setup_t *setup = (wifi_event_sta_itwt_setup_t *) event_data;
    if (setup->status == ITWT_SETUP_SUCCESS) {
        /* TWT Wake Interval = TWT Wake Interval Mantissa * (2 ^ TWT Wake Interval Exponent) */
        ESP_LOGI(TAG, "<WIFI_EVENT_ITWT_SETUP>twt_id:%d, flow_id:%d, %s, %s, wake_dura:%d, wake_dura_unit:%d, wake_invl_e:%d, wake_invl_m:%d", setup->config.twt_id,
                setup->config.flow_id, setup->config.trigger ? "trigger-enabled" : "non-trigger-enabled", setup->config.flow_type ? "unannounced" : "announced",
                setup->config.min_wake_dura, setup->config.wake_duration_unit, setup->config.wake_invl_expn, setup->config.wake_invl_mant);
        ESP_LOGI(TAG, "<WIFI_EVENT_ITWT_SETUP>target wake time:%lld, wake duration:%d us, service period:%d us", setup->target_wake_time, setup->config.min_wake_dura << (setup->config.wake_duration_unit == 1 ? 10 : 8),
                setup->config.wake_invl_mant << setup->config.wake_invl_expn);

        esp_err_t err = esp_wifi_80211_tx(WIFI_IF_STA, ds2ds_pdu, sizeof(ds2ds_pdu), true);
        ESP_ERROR_CHECK_WITHOUT_ABORT(err);
        ESP_LOGI(TAG, "[%d]send 802.11 data :%x",__LINE__,err);

    } else {
        if (setup->status == ESP_ERR_WIFI_TWT_SETUP_TIMEOUT) {
            ESP_LOGE(TAG, "<WIFI_EVENT_ITWT_SETUP>twt_id:%d, timeout of receiving twt setup response frame", setup->config.twt_id);
        } else if (setup->status == ESP_ERR_WIFI_TWT_SETUP_TXFAIL) {
            ESP_LOGE(TAG, "<WIFI_EVENT_ITWT_SETUP>twt_id:%d, twt setup frame tx failed, reason: %d", setup->config.twt_id, setup->reason);
        } else if (setup->status == ESP_ERR_WIFI_TWT_SETUP_REJECT) {
            ESP_LOGE(TAG, "<WIFI_EVENT_ITWT_SETUP>twt_id:%d, twt setup request was rejected, setup cmd: %d", setup->config.twt_id, setup->config.setup_cmd);
        } else {
            ESP_LOGE(TAG, "<WIFI_EVENT_ITWT_SETUP>twt_id:%d, twt setup failed, status: %d", setup->config.twt_id, setup->status);
        }
    }
}

```
The list of 802.11 data frames I captured with Wireshark on channel 11 is as follows: [No custom data broadcast type with a length of 93]

[image]

Here are my current router settings, all the features should be enabled.
[image]

The following are the logs printed after powering on.
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x3f (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x18ec
load:0x4086c110,len:0xe5c
load:0x4086e610,len:0x2f50
entry 0x4086c11a
I (23) boot: ESP-IDF v5.3.2-1-g7c4d58e215-dirty 2nd stage bootloader
I (24) boot: compile time Jan 10 2025 11:42:48
I (25) boot: chip revision: v0.0
I (28) boot: efuse block revision: v0.1
I (33) boot.esp32c6: SPI Speed      : 80MHz
I (38) boot.esp32c6: SPI Mode       : DIO
I (42) boot.esp32c6: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
[... 136 lines cut ...]

I (2750) wifi:pm start, type:1, aid:0x11, trans-BSSID:fc:34:97:5c:0c:08, BSSID[5]:0x8, mbssid(max-indicator:0, index:0), he:1
I (2761) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (2769) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (2777) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (2788) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (2799) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:32, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (2810) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2816) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (2825) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (2834) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:32, batimeout:0, txa_wnd:32
I (2844) esp_netif_handlers: sta ip: 192.168.50.10, mask: 255.255.255.0, gw: 192.168.50.1
I (2878) itwt: <WIFI_EVENT_ITWT_SETUP>twt_id:0, flow_id:0, trigger-enabled, announced, wake_dura:255, wake_dura_unit:0, wake_invl_e:10, wake_invl_m:512
I (2881) itwt: <WIFI_EVENT_ITWT_SETUP>target wake time:114311168, wake duration:65280 us, service period:524288 us
I (2892) itwt: [189]send 802.11 data :0
W (18647) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x100080c, TALO:0x5c9734fc, (ssn:0, win:64, cur_ssn:0), CONF:0xc0000005
I (18698) wifi:<ba-del>idx:0, tid:0
W (19701) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x100080c, TALO:0x5c9734fc, (ssn:1, win:64, cur_ssn:1), CONF:0xc0000005
I (19754) wifi:<ba-del>idx:0, tid:0
W (20755) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x100080c, TALO:0x5c9734fc, (ssn:1, win:64, cur_ssn:1), CONF:0xc0000005

Done
(base) ➜  itwt git:(7c4d58e215) ✗ 

```

zhangyanjiaoesp, 2025-01-10:
@mintisan 
1. According to the mac header "0x48 0x03", the 80211 tx packet should be a "Null data" packet, not the "probe request" packet. 
2. According to the itwt log: 
```
I (2878) itwt: <WIFI_EVENT_ITWT_SETUP>twt_id:0, flow_id:0, trigger-enabled, announced, wake_dura:255, wake_dura_unit:0, wake_invl_e:10, wake_invl_m:512
I (2881) itwt: <WIFI_EVENT_ITWT_SETUP>target wake time:114311168, wake duration:65280 us, service period:524288 us
```
The STA should wake up once every 524288 us to transmit a packet, and the duration of each wake-up period is 65280 us.

zhangyanjiaoesp, 2025-02-25:
@mintisan 
Do you still have this issue? If no, can we close it?

Alvin1Zhang, 2025-03-27:
Thanks for reporting, will close due to inactivity, feel free to reopen if any updates.

closed 2025-03-27 by Alvin1Zhang