espressif/esp-idf #15149: where is "CONFIG_OPENTHREAD_BR_START_SERVER " in thread ot-br example menuconfig?

githubworksonline, 2025-01-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

as “https://docs.espressif.com/projects/esp-thread-br/en/latest/codelab/web-gui.html”，i used esp-idf 5.3.2, but "https://github.com/espressif/esp-thread-br.git", i can not find CONFIG_OPENTHREAD_BR_START_SERVER , "esp-thread-br" is necessary？

githubworksonline, 2025-01-11:
> Hi @githubworksonline Please enable `OPENTHREAD_BR_START_WEB`, sorry, we update the name of the config but did not change it in docs.

can you llist the path? i cannot found it.

zwx1995esp, 2025-01-13:
> > Hi @githubworksonline Please enable `OPENTHREAD_BR_START_WEB`, sorry, we update the name of the config but did not change it in docs.
> 
> can you llist the path? i cannot found it.

Hi @githubworksonline  Sure, you can find this configuration here:
[image]
Also a tip for using `menuconfig`: you can use the `/` to jump to a specific configuration:
[image]

githubworksonline, 2025-01-18:
> Hi [@githubworksonline](https://github.com/githubworksonline) Sure, you can find this configuration here: [image] Also a tip for using `menuconfig`: you can use the `/` to jump to a specific [...]

May I ask which branch you are using？i cannot found it in release v5.4.

zwx1995esp, 2025-01-20:
Oh, Hi, please see the repo: https://github.com/espressif/esp-thread-br , the web server feature are supported in this repo.

closed 2025-02-26 by chshu