espressif/esp-idf #15151: Load address changed and now ESP is restarting indefinitely

yanmartins, 2025-01-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.4

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Hello,

Some of our devices started to have a problem with ESP initialization.

### What is the actual behavior?

## Device log with problems:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0xff0034ff,len:356415

## Device log working:

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:1392

## We noticed that there is an offset in the last load of the devices:

- load:`0x3fff0034`,len:1392 -- Log Ok
- load:`0xff0034ff`,len:356415 -- Log with problems, there is an offset in the load address

When we read the entire flash and flash the same image again, the device works. Is there any idea what might be causing this?

Thank you!
 

### Steps to reproduce.

We can't reproduce

### Debug Logs.

```plain
Here I have two logs from two different devices:

## Device 1

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:1392
ho 0 tail 12 room 4
load:0x40078000,len:11064
ho 0 tail 12 room 4
load:0x40080400,len:3820
entry 0x40080650
ets Jul 29 2019 12:21:46
```

## Device 2

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0xff0034ff,len:356415
1150 mmu set 00010000, pos 00010000
1150 mmu set 00020000, pos 00020000
1150 mmu set 00030000, pos 00030000
1150 mmu set 00040000, pos 00040000
1150 mmu set 00050000, pos 00050000
ho 0 tail 11 room 5
load:0x27ee3639,len:284230490
1150 mmu set 00060000, pos 00060000
1150 mmu set 00070000, pos 00070000
1150 mmu set 00080000, pos 00080000
1150 mmu set 00090000, pos 00090000
ets Jul 29 2019 12:21:46
```

# Both keep restarting indefinitely
```

ginkgm, 2025-01-07:
Hi @yanmartins ,
I doubt this is not an SW issue... Looks like to be caused by hardware design or the flash you are using. Do you see the same problem with an official module?

yanmartins, 2025-01-08:
> Hi @yanmartins , I doubt this is not an SW issue... Looks like to be caused by hardware design or the flash you are using. Do you see the same problem with an official module?

Hello @ginkgm,

We produce hundreds of devices. Some of these devices are experiencing this problem. A minority of the total number, but at the same time a relevant number.

Since I can't reproduce the problem on other devices, it is still too abstract what could be the root cause.

It works on official modules. But it also works on the vast majority of our devices. So I believe that this path does not help much in solving the problem.

I wanted to know if there is any knowledge about what could be causing this load shift, and what it means.

Then we can try to identify if there is a problem with our recording process or hardware.

Thank you.

igrr, 2025-01-08:
Could you please try dumping the entire flash contents of the "broken" device (using `esptool.py read_flash`) and compare it with that of a working device? This could help identify which part of flash gets corrupted or overwritten. The specific difference (single bit errors? complete 4kB sectors overwritten? something else?) will also help point to the possible cause of this issue.

ginkgm, 2025-01-09:
And here is one possible reason. ESP MSPI have a buffer size of 64 bytes. It means that when writing for flash, it needs to finishes the writing of 64 bytes before the next 64 bytes, not aligned to the 256 boundary.

When writng every 64 bytes, the address is sent. It looks like your data is shifted by byte, instead of bit. So it's likely to be not a noise in the clock but in the address. If the address is changed from 0 -> 1 when sending the address, this issue can happen.

If I'm correct, you will see the data is corrupted from a 64*N + 1 address.

ginkgm, 2025-02-26:
Long time no response. Issue closed. Feel free to reopen if you still have problems.

closed 2025-02-26 by ginkgm

yanmartins, 2025-02-26:
Sorry for the lack of feedback. The problem was resolved by updating the bootloader version of the devices.