espressif/esp-idf #15156: Philosophically - Is there any really compelling reason to impose watchdog timers on a task?

SolidStateLEDLighting, 2025-01-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I consider this posting to be a lack of documentation.

I have been working with the IDF for 5 years.   I do accept the default WDT timer settings in menuConfig for any projects that I build and I typically don't implement extra WDT calls in my code.    What am I missing out on by not making use of extra watch dog timers?   I don't see the value -- and philosophically, Espressif does not explain in documentation how the WDT functions help produce a better result.

I'm asking this set of questions because I just reviewed a project where a developer imposed watchdog timers on his task and then sprinkled esp_task_wdt_reset() calls throughout the code.   It looks like a massive waste of clock cycles.

I can understand the idea of knowing if IDLE tasks are not getting serviced -- but I don't see the values of imposing this burden on developer constructed tasks.   Can anyone at Espressif provide some understanding on this concern?   Thanks.

ginkgm, 2025-01-07:
I think it's about making sure a task is still real-time.. (serviced in time) 

wdt for idle task is setting a bottom line for all tasks. And if you want higher realtime performance for a specific task, you can give it a shorter WDT

SolidStateLEDLighting, 2025-01-07:
Couldn't a developer just measure the microseconds elapsed in a routine and not risk a crash?

KonstantinKondrashov, 2025-01-07:
Hi!
There is [the Watchdog doc ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#watchdogs)
1. Interrupt Watchdog Timer (**IWDT**): The purpose of the IWDT is to ensure that interrupt service routines (ISRs) are not blocked from running for a prolonged period of time. Preventing ISRs from running in a timely manner is undesirable as it can increase ISR latency, and also prevent task switching.

2. The Task Watchdog Timer (**TWDT**) is used to monitor particular tasks, ensuring that they are able to execute within a given timeout period. The TWDT primarily watches the Idle Tasks of each CPU, however any task can subscribe to be watched by the TWDT. By watching the Idle Tasks of each CPU, the TWDT can detect instances of tasks running for a prolonged period of time without yielding. This can be an indicator of poorly written code that spinloops on a peripheral, or a task that is stuck in an infinite loop. 

> I'm asking this set of questions because I just reviewed a project where a developer imposed watchdog timers on his task and then sprinkled esp_task_wdt_reset() calls throughout the code. It looks like a massive waste of clock cycles.

If a task is subscribed (via [esp_task_wdt_add](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#_CPPv416esp_task_wdt_add12TaskHandle_t)) to the TWDT. Each subscribed task must periodically call esp_task_wdt_reset().

Appropriate Kconfig timeout values for them can be adjusted via menuconfig.

> Couldn't a developer just measure the microseconds elapsed in a routine and not risk a crash?

It depends on the application—if the app is thoroughly debugged, you can disable WDTs. However, it is not recommended because WDTs provide a critical safety mechanism:
- Even in well-tested applications, unexpected conditions such as hardware faults, edge cases in logic, or power fluctuations can cause hangs or system instability.
- WDTs automatically detect situations where tasks, ISRs, or the system become unresponsive and reset the system
- WDTs monitor critical sections, ISRs, and tasks to ensure they don’t monopolize resources, block other operations, or create deadlocks.
- In applications with real-time requirements, WDTs help ensure timely task execution, avoiding timing violations that could compromise functionality.

SolidStateLEDLighting, 2025-01-07:
On a practical level, I submit to you that instead of showing how a developer can beat a watchdog timers with calls to esp_task_wdt_reset() , that Espressif endorses more techniques in their examples of how to frequently yield to RTOS.

I don't disable any of the default WDT dependencies -- but I certainly don't add more tasks to those timers.

ESP-Marius, 2025-01-08:
> On a practical level, I submit to you that instead of showing how a developer can beat a watchdog timers with calls to esp_task_wdt_reset() , that Espressif endorses more techniques in their examples of how to frequently yield to RTOS.

I think maybe you are mixing the two purposes a bit. Yielding to the RTOS more frequently will only ensure that the IDLE task feeds the watchdog. (which I think you already stated that you understand/see the purpose of)

Adding a developer task, and then calling `esp_task_wdt_reset` periodically` from it is to ensure that this task in question does indeed run periodically. Just yielding to the RTOS frequently (even if all tasks does this) do not ensure this, as the task could be blocking on a resource or a semaphore that it never gets. Calling `esp_task_wdt_reset()` from the task being watched is the intended purpose to signal that the task did indeed do what we intended it to do, inside the time-frame requested (the WDT timeout). So this is not "beating" the WDT, but using it as intended.

Not sure if this made it a bit clearer.

SolidStateLEDLighting, 2025-01-08:
Going back to the word "Philosophically", a WDT is an indicator of a problem.   It is my impression that it's primary purpose in the IDF is to make sure IDLE processing takes place as there is important freeRTOS housekeeping happening there.    A secondary (more traditional) reason would be to indicate that an app is blocked.

In my opinion, the call to esp_task_wdt_reset() doesn't achieve the goal of making sure IDLE processing it taking place.   So, I consider it a possible nasty way of injecting a problem into your project if IDLE processing is being unknowingly starved of resources.

I do understand that under limited conditions (large mathematical operations as an example), perhaps there might be a need to traverse a WDT timer period.    So, a way to "kick the dog" is necessary.

My post here was primarily a question of "Did I miss something"?    I believe the answer to that question is No.

So, thank you for all your thoughtful comments.

SolidStateLEDLighting, 2025-01-08:
I just reread your posting again.... so, another valid use might be also that a task blocks on a resource.   And this might be unknown without a WDT alert inside the running task...  (dual core processors might not pick up on this as the other CPU would continue to be available).

boarchuz, 2025-01-08:
> In my opinion, the call to esp_task_wdt_reset() doesn't achieve the goal of making sure IDLE processing it taking place.

Understand that `esp_task_wdt_reset` is task-specific.

Calling esp_task_wdt_reset from a task other than the idle task will _not_ feed the task watchdog timer specific to the idle task. The goal of `esp_task_wdt_reset` is to feed the watchdog timer only for the active task from which it is called. If the idle task is being starved, the system will still panic even if `esp_task_wdt_reset` has been called recently from a _different_ task.

SolidStateLEDLighting, 2025-01-08:
Ah, ok.   So, this call inside a task has no negative effects.   Thank you.

SolidStateLEDLighting, 2025-01-08:
I can see this being very important for dual core processors -- but I tend to think not much extra benefit for single cores.

igrr, 2025-01-08:
> so, another valid use might be also that a task blocks on a resource. And this might be unknown without a WDT alert inside the running task... (dual core processors might not pick up on this as the other CPU would continue to be available).

> I can see this being very important for dual core processors -- but I tend to think not much extra benefit for single cores as the IDLE task WDT routines would cover for a single core.

I think it doesn't matter that much if there are two processors or one in this case.

A task blocked on a resource (e.g. reading a value from a queue) will not prevent lower-priority tasks, including the idle tasks, from running. Even on a single-core system, the idle task can still run and feed its own watchdog.

SolidStateLEDLighting, 2025-01-08:
If a task is waiting on a queue in a normally expected fashion, the CPU not held up.    No need for any kind of WDT intervention there.

I never use the task assigned WDT calls but I'm also very keen to employ freeRTOS constructs (taskNotfication / queues) in every running task that I have.    I almost never allow one task to touch another tasks variables so freeRTOS mechanisms are mandatory.   And as you know these mechanisms have yielding conditions.

I've never seen any sample code that shows a compelling use case for the task assigned WDT functions.   And I suspect there is a substantial opportunity for a user to misuse these WDT functions.

ESP-Marius, 2025-01-08:
> If a task is waiting on a queue in a normally expected fashion, the CPU not held up. No need for any kind of WDT intervention there.

Yes a task waiting on e.g. `xQueueReceive()` do not hold up the CPU, and will not trigger the IDLE task WDT. But some users might want additional assurances that the code in the task waiting for `xQueueReceive` do run occasionally. Adding extra tasks to be watched by the WDT is intended for these users.

igrr, 2025-01-08:
> If a task is waiting on a queue in a normally expected fashion, the CPU not held up. No need for any kind of WDT intervention there.

Task watchdog is a tool, and it's the application developer's choice whether to apply it or not.

Of course, when waiting on some synchronization primitive or a resource, you can guard against lock-up by providing a specific timeout, for example:

```c++
res = xQueueReceive(queue, &item, pdMS_TO_TICKS(timeout));
if (res != pdPASS) {
   // perform corrective action
}
```

This is completely fine and is sufficient in most cases. However, if it is really essential that the task keeps performing the action it needs to perform, an application developer might _also_ enable a task WDT for this task — so that if the timeouts somehow fail to catch the lockup, the device can recover.

BitsForPeople, 2025-01-15:
A watchdog is a tool to detect and react to an application/system not working as intended. It is a somewhat crude tool, but for this reason it is also very simple; so simple that it can be easily implemented in hardware and be very robust.
The WDT can potentially even catch failures which cannot be caught/detected in software, like corruption of (some) critical memory or other erratic execution of code, be it caused by a bug, a Vcc glitch, by cosmic rays, or other interference.

If you see a WDT reset as an unnecessary "crash" you may indeed not need to use it. If you're controlling physical devices, like the motors of a robotic actuator, you may be glad the system automatically detected that your control task broke and reset itself into a safe state instead of swinging the robot around until someone manages to pull the plug.

closed 2025-02-05 by KonstantinKondrashov