espressif/esp-idf #15157: nimble ble_gap_master_set_timer compile fail if CONFIG_BT_NIMBLE_ROLE_CENTRAL is not set

AxelLin, 2025-01-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.3-783-g5eaa63af7d

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

compile was fine before updating esp-idf with BT_NIMBLE_ROLE_PERIPHERAL=y && BT_NIMBLE_ROLE_CENTRAL=n.
Now I got build failure.

### What is the actual behavior?

build error.

### Steps to reproduce.

Use examples/bluetooth/nimble/bleprph as example
Unselect CONFIG_BT_NIMBLE_ROLE_CENTRAL and build 
Then you will hit the build error

### Build or installation Logs.

```plain
/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c: In function 'ble_gap_disc':
/home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:5684:9: error: implicit declaration of function 'ble_gap_master_set_timer'; did you mean 'ble_gap_master_timer'? [-Werror=implicit-function-declaration]
 5684 |         ble_gap_master_set_timer(duration_ticks);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~
      |         ble_gap_master_timer
cc1: some warnings being treated as errors
```

rahult-github, 2025-01-07:
Hi @AxelLin , 
 This has been fixed [here](https://github.com/espressif/esp-idf/commit/905a83ecbf187bc25be5dbbb1383fa688805bce6).

AxelLin, 2025-01-08:
> Hi @AxelLin , This has been fixed [here](https://github.com/espressif/esp-idf/commit/905a83ecbf187bc25be5dbbb1383fa688805bce6).

No, it's not true.
As I mention the build issue with IDF v5.2.3-783-g5eaa63af7d which already include above fix.
But I still got the build error.

rahult-github, 2025-01-08:
Hi @AxelLin , 

I rechecked. I think you have OBSERVER role flag enabled, but disabled only CENTRAL role flag.   Please find attached patch that should help in this case.  

Request to share your sdkconfig, if this patch is not helping fix the issue. 

[observer_flag_fix.txt](https://github.com/user-attachments/files/18348741/observer_flag_fix.txt)

AxelLin, 2025-01-09:
> Hi @AxelLin ,
> 
> I rechecked. I think you have OBSERVER role flag enabled, but disabled only CENTRAL role flag. 

Yes.

> Please find attached patch that should help in this case.

Yes, the patch fixed it.

AxelLin, 2025-01-21:
> > Please find attached patch that should help in this case.
> 
> Yes, the patch fixed it.

@rahult-github 
Seems this is not yet fix in esp-nimble.

rahult-github, 2025-01-21:
Hello, change has been merged . It should show up once the GitHub sync happens

rahult-github, 2025-02-03:
Commit is available on [github](https://github.com/espressif/esp-idf/commit/65aaf217d8eaab6d50bff3c034d69892eb98f7d2).

closed 2025-02-03 by rahult-github