espressif/esp-idf #15158: Support for LAN8651 MAC & PHY drivers (Single Pair Ethernet)

amirdahal, 2025-01-07:
### Is your feature request related to a problem?

With Single Pair Ethernet being popular, I was hoping to find an easy way to integrate the LAN8651 SPE module with my espressif microcontrollers, which have not been developed yet.

### Describe the solution you'd like.

In the ESP-IDF components library, add support for additional SPE modules like the LAN8651 creating easily accessible APIs such as:
1. esp_eth_mac_new_lan8651
2. esp_eth_phy_new_lan8651

lolimpol, 2025-03-31:
+1

Will this be done?

kostaond, 2025-04-01:
Yes, it's on my plan.

kostaond, 2025-06-26:
Good news, everyone! LAN8651 driver is in the review! https://github.com/espressif/esp-eth-drivers/pull/70

closed 2025-06-26 by kostaond