espressif/esp-idf #15162: http_server does not close session on "Connection: close" header

DCSBL, 2025-01-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-S3, ESP32

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I expect that when a http client sends a request with the 'Connection: close' header, that the server closes the connection after replying.

### What is the actual behavior?

The header is ignored

### Steps to reproduce.

1. Set up a HTTP server from the examples.
2. Send a request with a "Connection: close" header
3. Connection is not closed

### More Information.

We can implement this manually, but I think this should be default behaviour, or at least configurable via `menuconfig`.

hrushikesh430, 2025-03-06:
Hi @DCSBL, To close the connection return ESP_FAIL from the handler itself. We intentionally kept the connection closing capability on the application side and we will go with this method only. So if you want to close the connection after receiving `Connection:close` header, then you can handle it in you handler.

For example - 

```
    buf_len = httpd_req_get_hdr_value_len(req, "Connection") + 1;
    if (buf_len > 1) {
        buf = malloc(buf_len);
        ESP_RETURN_ON_FALSE(buf, ESP_ERR_NO_MEM, TAG, "buffer alloc failed");
        if (httpd_req_get_hdr_value_str(req, "Connection", buf, buf_len) == ESP_OK) {
            ESP_LOGI(TAG, "Found header => Connection: %s", buf);
            if (strcasecmp(buf, "close") == 0) {
                ESP_LOGI(TAG, "Connection: close header received, returning ESP_FAIL");
                free(buf);
                return ESP_FAIL;
            }
        }
        free(buf);
    }
```

You can refer this [docs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/esp_http_server.html) for more understanding of http_server.

DCSBL, 2025-03-06:
That is fine with me, thank you for your reply