espressif/esp-idf #15164: IDF v5.1.5 - AVRC Event 0x04 (Album) meta data isn't null terminated

expresspotato, 2025-01-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.5

### Espressif SoC revision.

ESP32-Pico-V3-02

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

LyraT & Custom, same result

### Power Supply used.

External 5V

### What is the expected behavior?

The AVRC metadata should be null terminated for this field

### What is the actual behavior?

The field value is not null terminated like the Artist (0x02) and Track (0x01) fields are

### Steps to reproduce.

Use the A2D example and wait for AVRC metadata

### Debug Logs.

```plain
I (28174) LIBERTY: AVRC metadata rsp: attribute id 0x1, When Doves Cry, length: 14
I (28174) LIBERTY: --> Received AVRC Event: 2
I (28179) LIBERTY: AVRC metadata rsp: attribute id 0x2, Prince, length: 6
I (28189) LIBERTY: --> Received AVRC Event: 2
I (28194) LIBERTY: AVRC metadata rsp: attribute id 0x4, Purple RainxV���, length: 11

...

I (21741) LIBERTY: --> Received AVRC Event: 2
I (21741) LIBERTY: --> Sending all set values to the server...
I (21746) LIBERTY: AVRC metadata rsp: attribute id 0x1, When Doves Cry, length: 14
I (21761) LIBERTY: --> Received AVRC Event: 2
I (21771) LIBERTY: AVRC metadata rsp: attribute id 0x2, Prince, length: 6
E (21776) LIBERTY: --> Finished, sending back item... is threaded (1)
I (21786) LIBERTY: --> Received AVRC Event: 2
I (21796) LIBERTY: AVRC metadata rsp: attribute id 0x4, Purple RainxV��y, length: 11
CORRUPT HEAP: Bad tail at 0x3f84cd85. Expected 0xbaad5678 got 0xbaad5600

assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)

Backtrace: 0x40082ab2:0x3ffe2e90 0x4009c5f1:0x3ffe2eb0 0x400a3db5:0x3ffe2ed0 0x400a28b5:0x3ffe3000 0x40083603:0x3ffe3020 0x400a3e85:0x3ffe3040 0x40086225:0x3ffe3060 0x400a3afd:0x3ffe3090 0x4008798d:0x3ffe30f0 0x4012f061:0x3ffe3130 0x4012e57e:0x3ffe3210 0x4017a72f:0x3ffe3250 0x4012e971:0x3ffe3270 0x4016f4e9:0x3ffe3290 0x40171d57:0x3ffe32b0 0x4009f371:0x3ffe32e0
--- 0x40082ab2: panic_abort at /Users/kevin/esp/esp-idf-v5.1.5/components/esp_system/panic.c:466
0x4009c5f1: esp_system_abort at /Users/kevin/esp/esp-idf-v5.1.5/components/esp_system/port/esp_system_chip.c:84
0x400a3db5: __assert_func at /Users/kevin/esp/esp-idf-v5.1.5/components/newlib/assert.c:81
0x400a28b5: multi_heap_free at /Users/kevin/esp/esp-idf-v5.1.5/components/heap/multi_heap_poisoning.c:276 (discriminator 1)
0x40083603: heap_caps_free at /Users/kevin/esp/esp-idf-v5.1.5/components/heap/heap_caps_base.c:64
0x400a3e85: free at /Users/kevin/esp/esp-idf-v5.1.5/components/newlib/heap.c:39
0x40086225: fs_logging_vprintf at /Users/kevin/workspace/liberty_device/main/main.c:198
0x400a3afd: esp_log_writev at /Users/kevin/esp/esp-idf-v5.1.5/components/log/log.c:197
 (inlined by) esp_log_write at /Users/kevin/esp/esp-idf-v5.1.5/components/log/log.c:207
0x4008798d: app_bt_sys_callback_avrc at /Users/kevin/workspace/liberty_device/main/app_bt.c:1602 (discriminator 13)
0x4012f061: btc_avrc_ct_cb_to_app at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c:262
 (inlined by) btc_avrc_ct_cb_to_app at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c:254
 (inlined by) handle_rc_attributes_rsp at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c:556
 (inlined by) handle_rc_metamsg_rsp at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c:804
 (inlined by) handle_rc_metamsg_cmd at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c:664
 (inlined by) btc_rc_handler at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c:961
0x4012e57e: btc_av_state_opened_handler at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/host/bluedroid/btc/profile/std/a2dp/btc_av.c:810
0x4017a72f: btc_sm_dispatch at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/host/bluedroid/btc/core/btc_sm.c:145
0x4012e971: btc_a2dp_cb_handler at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/host/bluedroid/btc/profile/std/a2dp/btc_av.c:1638
0x4016f4e9: btc_thread_handler at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/common/btc/core/btc_task.c:207
0x40171d57: osi_thread_run at /Users/kevin/esp/esp-idf-v5.1.5/components/bt/common/osi/thread.c:165
0x4009f371: vPortTaskWrapper at /Users/kevin/esp/esp-idf-v5.1.5/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### More Information.

None

ESP-YTGerd, 2025-01-13:
Hi, @expresspotato,
Is this issue consistently reproducible? I ran the example `a2dp_sink` locally for over an hour but couldn't reproduce the problem. Additionally, based on the logs, there seems to be a heap corruption. 
`CORRUPT HEAP: Bad tail at 0x3f84cd85. Expected 0xbaad5678 got 0xbaad5600`
So could you confirm if you have modified the example code? If so, could you share the changes you made?

expresspotato, 2025-02-10:
So this was caused by a missing call to bt_app_alloc_meta_buffer on each ESP_AVRC_CT_METADATA_RSP_EVT event. The corresponding free was also added.

closed 2025-02-10 by expresspotato