espressif/esp-idf #15165: ot_rcp example could use more documentation and debug steps

stevenlybeck, 2025-01-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm attempting to use [openthread daemon](https://github.com/openthread/openthread/tree/main/src/posix) for the Host portion of an [openthread co-processor design](https://openthread.io/platforms/co-processor) - along with an esp32-c6 flashed with the [ot_rcp firmware](https://github.com/espressif/esp-idf/tree/b5ac4fbd/examples/openthread/ot_rcp).

There aren't really any comprehensive instructions about this, but I'm finding that ot-daemon bails as it attempts to connect to my esp32 devboard flashed with ot_rcp:

```
>> sudo ./build/posix/src/posix/ot-daemon --verbose 'spinel+hdlc+uart:///dev/cu.usbmodem101?uart-baudrate=460800&no-reset'
Jan  8 00:11:00  ./build/posix/src/posix/ot-daemon[49362] <Info>: Running OPENTHREAD/thread-reference-20230706-1115-gdc4acf034-dirty; POSIX; Jan  7 2025 23:50:16
Jan  8 00:11:00  ./build/posix/src/posix/ot-daemon[49362] <Info>: Thread version: 5
Jan  8 00:11:04  ./build/posix/src/posix/ot-daemon[49362] <Critical>: 49d.21:57:21.396 [C] Platform------: Init() at spinel_driver.cpp:83: Failure
```

I realize this is a kind of cross-project issue - but my hope is that espressif can provide more comprehensive guidance on using ot_rcp specifically for Thread (and Zigbee) Sniffing.

zwx1995esp, 2025-03-07:
Hi, @stevenlybeck  could you please share a picture how you connect the C6 with the host? And did you modify any configurations in the `ot_rcp` example? Our example use 460800 as the default UART baudrate, so you need to change the configurations for ot-br-posix daemon: './build/posix/src/posix/ot-daemon 'spinel+hdlc+uart:///dev/ttyUSB0?uart-baudrate=460800'

Alvin1Zhang, 2025-03-27:
@stevenlybeck Thanks for reporting, would you please help share if any updates? Thanks.

chshu, 2025-04-25:
@stevenlybeck Please refer to this doc for using esp ot_rcp with OTBR: https://docs.espressif.com/projects/esp-thread-br/en/latest/qa.html#using-ot-rcp-with-otbr-ot-br-posix

Closing the issue, feel free to reopen if any follow up questions.

closed 2025-04-25 by chshu