espressif/esp-idf #15166: Sdk 5.4 xtensa crashes and halt without proper reboot

filzek, 2025-01-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-1-g6b7b8ea3d2

### Espressif SoC revision.

Esp32 Wrover

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Esp32 Wrover module

### Power Supply used.

External 5V

### What is the expected behavior?

Doesnt crash, but as it crashes the system should REBOOT, but it just halt and locked.

### What is the actual behavior?

Guru Meditation Error: Core  1 panic'ed (Double exception).

Core  1 register dump:
PC      : 0x4009d882  PS      : 0x00040136  A0      : 0x801d3620  A1      : 0x3fff8710
--- 0x4009d882: _xt_context_save at C:/Espressif/frameworks/esp-idf-v5.4/components/xtensa/xtensa_context.S:203

A2      : 0xd5d83b00  A3      : 0x00062b23  A4      : 0x00000000  A5      : 0x00060b23
A6      : 0x00000001  A7      : 0x3ffc3628  A8      : 0x00000000  A9      : 0x00000000
A10     : 0xc4f40983  A11     : 0x0000000c  A12     : 0x00000001  A13     : 0x3f4408b0
A14     : 0x00000003  A15     : 0x00000000  SAR     : 0x00000001  EXCCAUSE: 0x00000002
EXCVADDR: 0x3f4408a0  LBEG    : 0x400838c9  LEND    : 0x400838d1  LCOUNT  : 0x00000026
--- 0x400838c9: esp_timer_impl_get_counter_reg at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_timer/src/esp_timer_impl_lac.c:119
0x400838d1: esp_timer_impl_get_counter_reg at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_timer/src/esp_timer_impl_lac.c:129

Backt�ace: 0x4009d87f:0x3fff8710 |<-CORRUPTED
--- 0x4009d87f: _xt_context_save at C:/Espressif/frameworks/esp-idf-v5.4/components/xtensa/xtensa_context.S:202

get locked and halted!

### Steps to reproduce.

timers, gptimer, adc continous read, some isrs.

### Debug Logs.

```plain
Guru Meditation Error: Core  1 panic'ed (Double exception).

Core  1 register dump:
PC      : 0x4009d882  PS      : 0x00040136  A0      : 0x801d3620  A1      : 0x3fff8710
--- 0x4009d882: _xt_context_save at C:/Espressif/frameworks/esp-idf-v5.4/components/xtensa/xtensa_context.S:203

A2      : 0xd5d83b00  A3      : 0x00062b23  A4      : 0x00000000  A5      : 0x00060b23
A6      : 0x00000001  A7      : 0x3ffc3628  A8      : 0x00000000  A9      : 0x00000000
A10     : 0xc4f40983  A11     : 0x0000000c  A12     : 0x00000001  A13     : 0x3f4408b0
A14     : 0x00000003  A15     : 0x00000000  SAR     : 0x00000001  EXCCAUSE: 0x00000002
EXCVADDR: 0x3f4408a0  LBEG    : 0x400838c9  LEND    : 0x400838d1  LCOUNT  : 0x00000026
--- 0x400838c9: esp_timer_impl_get_counter_reg at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_timer/src/esp_timer_impl_lac.c:119
0x400838d1: esp_timer_impl_get_counter_reg at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_timer/src/esp_timer_impl_lac.c:129

Backt�ace: 0x4009d87f:0x3fff8710 |<-CORRUPTED
--- 0x4009d87f: _xt_context_save at C:/Espressif/frameworks/esp-idf-v5.4/components/xtensa/xtensa_context.S:202
```

### More Information.

To cause the problem esp32 are receivend a HTTP post, receving an ISR of 433mhz radio on gpio, and the iot buttom are being triggered. All those acting are enable and disable gpios while acting.

sudeep-mohanty, 2025-01-08:
Hi @filzek,
Could you make sure that you have either `CONFIG_ESP32_PANIC_PRINT_REBOOT` or `CONFIG_ESP32_PANIC_SILENT_REBOOT` enabled in your `sdkconfig` and not `CONFIG_ESP32_PANIC_PRINT_HALT`. Secondly, if you have a debugger attached then the default behavior would be to not reboot the chip on panic. You could change the behavior if you turn off `CONFIG_ESP_DEBUG_OCDAWARE`.

filzek, 2025-01-08:
CONFIG_ESP32_PANIC_PRINT_REBOOT=y
# CONFIG_ESP32_PANIC_PRINT_HALT is not set
CONFIG_ESP32_PANIC_PRINT_REBOOT=y

the 
CONFIG_ESP_DEBUG_OCDAWARE=y
so need to change to  is not set?

sudeep-mohanty, 2025-01-09:
Hi @filzek,
If you have a debugger connected to your board then the board will not reboot when a panic happens. This behavior is in place to allow you to inspect the cause of the panic. The feature could be turned off by disabling `CONFIG_ESP_DEBUG_OCDAWARE`. However, if you do not have a debugger attached then this option would not have any effect. Could you confirm if this is the case?

filzek, 2025-01-13:
> Hi @filzek, If you have a debugger connected to your board then the board will not reboot when a panic happens. This behavior is in place to allow you to inspect the cause of the panic. The feature [...]

no debug attached, just reading tx/rx from serial.

sudeep-mohanty, 2025-01-13:
Hi @filzek,
This likely points to a memory corruption which could be altering the panic handler behavior. The fact the system entered a `Double exception` signifies that there was a panic while handling a previous exception. The system getting stuck could be explained by the fact that either the panic handler is stuck in a panic loop or there is something that has caused the watchdog timer to not be configured correctly to reboot the system.

We do have an internal task to make the panic handler more robust and try to avoid such problems in the future.

sudeep-mohanty, 2025-01-21:
Re-opening this ticket to track improvements to the panic handling infrastructure.

closed 2025-02-23 by espressif-bot

filzek, 2025-02-25:
If wanna that us do any advanced test we can do as you wish