espressif/esp-idf #15167: Undefined reference to `__atomic_test_and_set'

aiquie, 2025-01-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Successful build.

### What is the actual behavior?

Linking error: `undefined reference to '__atomic_test_and_set'`

### Steps to reproduce.

1. create project using function `atomic_flag_test_and_set`:
a. `idf.py create-project`
b. `main/test.c`
```
#include <stdatomic.h>

void app_main(void)
{
    atomic_flag_test_and_set(0);
}
```
2. set esp32c2 target **OR** enable PSRAM:
a. `idf.py set-target esp32c2`
b. `CONFIG_SPIRAM=y`
3. `idf.py build`

### Build or installation Logs.

```plain
a. with esp32c2 target

~/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(test.c.obj):(.literal.app_main+0x0): undefined reference to `__atomic_test_and_set'
~/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/main/libmain.a(test.c.obj): in function `app_main':
~/test/main/test.c:5:(.text.app_main+0x7): undefined reference to `__atomic_test_and_set'
```

b. enabled PSRAM:
```
~/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(test.c.obj): in function `app_main':
~/test/main/test.c:5:(.text.app_main+0x8): undefined reference to `__atomic_test_and_set'
```
```

### More Information.

- build on version 5.3 and earlier
- build with default settings

nopnop2002, 2025-01-09:
atomic_flag_test_and_set() is a C++ function.
This is my main.cpp.
```
#include <stdatomic.h>

extern "C" void app_main(void)
{
    std::atomic_flag x = ATOMIC_FLAG_INIT;
    atomic_flag_test_and_set(&x);
}
```

nopnop2002, 2025-01-09:
This is miy compile logging.
```
$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/nop/rtos/cpp-example/build
Executing "ninja all"...
[4/882] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[322/882] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.25.1")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/nop/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/nop/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/nop/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c2
-- Project sdkconfig file /home/nop/rtos/cpp-example/sdkconfig
-- Adding linker script /home/nop/esp-idf/components/riscv/ld/rom.api.ld
-- Adding linker script /home/nop/esp-idf/components/soc/esp32c2/ld/esp32c2.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script /home/nop/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.ld
-- Adding linker script /home/nop/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.api.ld
-- Adding linker script /home/nop/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.rvfp.ld
-- Adding linker script /home/nop/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.wdt.ld
-- Adding linker script /home/nop/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.systimer.ld
-- Adding linker script /home/nop/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.version.ld
-- Adding linker script /home/nop/esp-idf/components/esp_rom/esp32c2/ld/esp32c2.rom.newlib.ld
-- Adding linker script /home/nop/esp-idf/components/bootloader/subproject/main/ld/esp32c2/bootloader.ld
-- Adding linker script /home/nop/esp-idf/components/bootloader/subproject/main/ld/esp32c2/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table riscv soc spi_flash
-- Component paths: /home/nop/esp-idf/components/bootloader /home/nop/esp-idf/components/bootloader_support /home/nop/esp-idf/components/efuse /home/nop/esp-idf/components/esp_app_format /home/nop/esp-idf/components/esp_bootloader_format /home/nop/esp-idf/components/esp_common /home/nop/esp-idf/components/esp_hw_support /home/nop/esp-idf/components/esp_rom /home/nop/esp-idf/components/esp_system /home/nop/esp-idf/components/esptool_py /home/nop/esp-idf/components/freertos /home/nop/esp-idf/components/hal /home/nop/esp-idf/components/log /home/nop/esp-idf/components/bootloader/subproject/main /home/nop/esp-idf/components/bootloader/subproject/components/micro-ecc /home/nop/esp-idf/components/newlib /home/nop/esp-idf/components/partition_table /home/nop/esp-idf/components/riscv /home/nop/esp-idf/components/soc /home/nop/esp-idf/components/spi_flash
-- Configuring done (5.6s)
-- Generating done (0.1s)
-- Build files have been written to: /home/nop/rtos/cpp-example/build/bootloader
[93/94] Generating binary image from built executable
esptool.py v4.8.1
Creating esp32c2 image...
Merged 2 ELF sections
Successfully created esp32c2 image.
Generated /home/nop/rtos/cpp-example/build/bootloader/bootloader.bin
[94/94] cd /home/nop/rtos/cpp-example/build/bootloa...op/rtos/cpp-example/build/bootloader/bootloader.bin
Bootloader binary size 0x4d10 bytes. 0x32f0 bytes (40%) free.
[881/882] Generating binary image from built executable
esptool.py v4.8.1
Creating esp32c2 image...
Merged 2 ELF sections
Successfully created esp32c2 image.
Generated /home/nop/rtos/cpp-example/build/cpp_exceptions_example.bin
[882/882] cd /home/nop/rtos/cpp-example/build/esp-i...p/rtos/cpp-example/build/cpp_exceptions_example.bin
cpp_exceptions_example.bin binary size 0x241a0 bytes. Smallest app partition is 0x100000 bytes. 0xdbe60 bytes (86%) free.

Project build complete. To flash, run:
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32c2 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 60m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/cpp_exceptions_example.bin
or from the "/home/nop/rtos/cpp-example/build" directory
 python -m esptool --chip esp32c2 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
```

closed 2025-01-23 by espressif-bot

albkharisov, 2025-02-11:
Dobriy den, @Lapshin
When to expect next release with this fix?

Lapshin, 2025-02-11:
@albkharisov , 👋

According to the [roadmap](https://github.com/espressif/esp-idf/blob/master/ROADMAP.md):

- 5.5.0 release date is 2025/08/04
- 5.4.1 release date is 2025/03/27

stockwell, 2025-05-05:
@Lapshin unless I'm missing something, this fix doesn't seem to have made its way into the 5.4 branch and isn't in the 5.4.1 release. I can only see it in the master branch.