espressif/esp-idf #15169: Example code NimBLE_GATT_Server & esp32c6

tjmcwiz, 2025-01-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

The example code should advertise services that can be seen and accessed from the the esp32-c6. The code works properly for a esp32 dev chip. 

### What is the actual behavior?

See screenshot from nRF connect of the services not being displayed 

[image]

### Steps to reproduce.

1. Follow the steps provided in the example: esp/esp-idf/examples/bluetooth/ble_get_started/nimble/NimBLE_GATT_Server
2. Check for proper advertisements (services) in nRF connect
...

### Debug Logs.

```plain
--- esp-idf-monitor 1.5.0 on /dev/ttyACM0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
I (188) esp_image: segment 3: paddr=0009a49c vaddr=408051cc size000 - 0x4ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400294a6
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
--- 0x400294a6: uart_hal_get_txfifo_count in ROM

load:0x4086c110,len:0xe84
load:0x4086e610,len:0x304c
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4 2nd stage bootloader
I (23) boot: compile time Jan  8 2025 15:28:24
I (24) boot: chip revision: v0.0
[... 182 lines cut ...]
I (9480) NimBLE_GATT_Server: connection handle: 0
I (9480) NimBLE_GATT_Server: device id address: type=0, value=82:6D:40:CA:4C:40
I (9490) NimBLE_GATT_Server: peer id address: type=1, value=DB:40:D6:84:C0:5B
I (9490) NimBLE_GATT_Server: conn_itvl=24, conn_latency=0, supervision_timeout=72, encrypted=0, authenticated=0, bonded=0

I (9500) NimBLE: GAP procedure initiated: 
I (9510) NimBLE: connection parameter update; conn_handle=0 itvl_min=24 itvl_max=24 latency=3 supervision_timeout=72 min_ce_len=0 max_ce_len=0
I (9520) NimBLE: 

I (9850) NimBLE_GATT_Server: mtu update event; conn_handle=0 cid=4 mtu=256
I (10150) NimBLE_GATT_Server: connection updated; status=0
I (10150) NimBLE_GATT_Server: connection handle: 0
I (10150) NimBLE_GATT_Server: device id address: type=0, value=82:6D:40:CA:4C:40
I (10160) NimBLE_GATT_Server: peer id address: type=1, value=DB:40:D6:84:C0:5B
I (10160) NimBLE_GATT_Server: conn_itvl=36, conn_latency=3, supervision_timeout=72, encrypted=0, authenticated=0, bonded=0

I (10260) NimBLE_GATT_Server: heart rate updated to 61
I (11260) NimBLE_GATT_Server: heart rate updated to 75
I (12260) NimBLE_GATT_Server: heart rate updated to 64
```

### More Information.

When using a ESP32 dev chip --- it works correctly .... 
[sdkconfig.txt](https://github.com/user-attachments/files/18353046/sdkconfig.txt)

rahult-github, 2025-01-09:
Hi @tjmcwiz , 

 I gave a quick try for mentioned chip and release and i do see it working ok for c6.    Able to see the heart rate and automation IO service seen. 
 
The console log of my run matches exactly to one you shared .  
 
[console_log.txt](https://github.com/user-attachments/files/18356452/console_log.txt)

Can you please help  check once the verbose log in nrf connect app and see if service discovery has happened or any error is seen on nrf side ? 

 
[image]

tjmcwiz, 2025-01-23:
I think we can close this. I tried it on a newer esp32c6 dev-kit board and it is working correctly. Not sure what is wrong with the other board but it appears to be defective in this case.  

Thanks for reviewing.

closed 2025-01-23 by tjmcwiz

tomasmcguinness, 2025-12-11:
I know this is a very old post, but I've found that on my iPhone at least, I need to turn Bluetooth off and back on. 

I think the iOS Bluetooth holds onto the GATT details discovered, so if you're adding new services/characteristics, you might need to cycle Bluetooth.