espressif/esp-idf #15170: led_strip example is broken on ESP32-S3

zackees, 2025-01-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Led strip 3.0.0

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Xaio Esp32S3

### Power Supply used.

USB

### What is the expected behavior?

I expect the demo to work with 4 or more strips.

### What is the actual behavior?

It only works with two strips. Third strip causes a runtime crash.

### Steps to reproduce.

Keep in mind I'm using a lightly modified version of the led_strip component library to enable async drawing so that parallel rmt drivers can be used.

I may issue an additional PR to enable parallel async for rmt led_strip drawing. You can find my fork here:

https://github.com/zackees/esp-rmt-led-strip-component-idf-5-1-cpp

```C++
/*
 * SPDX-FileCopyrightText: 2023-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "led_strip.h"
#include "esp_log.h"
#include "esp_err.h"
#include "esp_check.h"

#ifndef LED_STRIP_GPIO_PIN
// GPIO assignment
#define LED_STRIP_GPIO_PIN  2
#endif

#ifndef LED_STRIP_LED_COUNT
[... 283 lines cut ...]
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Jan  8 2025 15:11:25
  ESP-IDF Version   : v5.1.4-586-gb6b4727c58-dirty
  Arduino Version   : 3.0.4
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
  Arduino Variant   : esp32s3
  Core Debug Level  : 5
  Arduino Runs Core : 1
  Arduino Events on : 1
  Arduino USB Mode  : 1
  CDC On Boot       : 0
============ Before Setup End ============
```

### More Information.

build type is platformio.

Here is the pull request to fix it.
https://github.com/espressif/idf-extra-components/pull/465

suda-morris, 2025-01-09:
Yes, this is a hardware change in the RMT. On esp32 and esp32s2, the FIFO size for each channel is 64, but on other targets, the size is chrunked to 48. Thanks for your PR, we will take a look there. Using zero is a clever solution.

suda-morris, 2025-01-09:
PR is merged. :tada:

closed 2025-01-09 by suda-morris