espressif/esp-idf #15172: Is it possible for the ESP to simulatneously have both encrypted and unencrypted app partitions?

Dazza0, 2025-01-09:
### Is your feature request related to a problem?

Does the ESP support having both encrypted and unencrypted app partitions at the same time?

According to the [Flash Encryption](https://docs.espressif.com/projects/esp-idf/en/v5.3.2/esp32s3/security/flash-encryption.html) documentation, it is possible to have both encrypted and non-encrypted partitions in flash at the same time via the use of the `encrypted` flash in `partitions.csv`.

However, the docs also says that "all flash content accessed via the MMU's flash cache is transparently decrypted". Therefore, does this mean that it is not possible to XIP from flash when there are two or more app partitions in flash, where one is encrypted and the other isn't? In other words, does the MMU flash cache have a setting where it only transparently decrypts certain flash address ranges.

### Describe the solution you'd like.

Would be handy if there was some sort of hardware setting to let the MMU flash cache know to only transparently decrypt certain flash address ranges.

### Describe alternatives you've considered.

If the MMU flash cache cannot decrypted only specific address ranges, the other possibility is just to read the unencrypted app partition using `esp_partition_read()`, and writing it into IRAM for execution.

### Additional context.

The use case I have in mind is having a factory app which is encrypted and acts as a fallback/default app, and another "user app" which is unencrypted, thus can be easily reprogrammed/OTA'd by the user without needing to share the flash encryption key.

igrr, 2025-01-09:
> Therefore, does this mean that it is not possible to XIP from flash when there are two or more app partitions in flash, where one is encrypted and the other isn't?

That's correct, on the existing SoCs this is not possible.

> The use case I have in mind is having a factory app which is encrypted and acts as a fallback/default app, and another "user app" which is unencrypted, thus can be easily reprogrammed/OTA'd by the user without needing to share the flash encryption key.

I would recommend adding some sort of "OTA" functionality to the fallback/default app. It doesn't have to be over the air, for example you could support uploading the user application from an SD card or over UART — depending on your hardware. The fallback/default app will then write the user application into an app partition, transparently encrypting it.

Dazza0, 2025-01-14:
@igrr On second thought, having a user programmed unencrypted app and an encrypted app on the same device is probably a bad idea anyways, since the user program can just be programmed to read the encrypted app. I guess what I'm trying to achieve isn't feasible.

Thanks for the response anyways.

closed 2025-01-14 by Dazza0