espressif/esp-idf #15174: idf component esp_driver_ledc CMakeLists requires issue

sfxfs, 2025-01-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The header file "driver/ledc.h" includes "driver/gpio.h"
[image]
but the CMakeLists for this component categorizes esp_driver_gpio as PRIV_REQUIRES. Is this correct?
[image]
When my project depends on esp_driver_ledc, I have to manually add esp_driver_gpio, instead of it being automatically included by CMake.

sfxfs, 2025-01-10:
Thank you for your reply. I am glad to have identified an issue and look forward to your fix.

KaeLL, 2025-01-13:
@ginkgm 
> IDF v6.0

Any ETA for it?

closed 2025-01-15 by espressif-bot

ginkgm, 2025-01-16:
> @ginkgm
> 
> > IDF v6.0
> 
> Any ETA for it?

We will release a roadmap for this year later. Please keep an eye on related news.